betaflight/betaflight #11977: Pull request 11771 not working since build 2323 from 31.10.2022

OlegSyr, 2022-11-10:
### Describe the bug

#11771 allowed us to get warnings and LQ in 'Craft_name' field when using DJI googles (many thanks to @howels).

Since build 2323 from 31.10.2022 this feature is not working any more. 

Warnings in 2322 are working OK.

Build 2323 renamed 'display_name' to "pilot _name' and 'name' to 'craft_name' among another changes. 
  

### To Reproduce

Use any nightly build starting from 2323 and configure use of 'craft_name' field for warnings in GDI googles

### Expected behavior

I expect merged pull request #11771 continue to work 

### Diff configuration

```txt
# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Nov  8 2022 / 23:10:51 (b44129f38) MSP API: 1.45

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAF722_2022A
manufacturer_id DIAT
mcu_id 002d00253431510a37393330
signature 

# name: LINK QUALITY

# feature
feature GPS

# serial
serial 0 2 115200 57600 0 115200
serial 2 1 115200 57600 0 115200
serial 3 64 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1700 2100 0 0
aux 1 46 1 1700 2100 0 0
aux 2 13 2 1700 2100 0 0
aux 3 26 0 1700 2100 0 0
aux 4 35 3 1700 2100 0 0

# rxfail
rxfail 5 s 1500

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 0
set yaw_spin_recovery = ON
set dyn_notch_count = 4
set dyn_notch_q = 600
set dyn_notch_min_hz = 125
set dyn_notch_max_hz = 700
set gyro_lpf1_dyn_min_hz = 500
set gyro_lpf1_dyn_max_hz = 1000
set acc_calibration = 24,22,19,1
set min_check = 1010
set rc_smoothing_auto_factor = 45
set fpv_mix_degrees = 30
set serialrx_provider = CRSF
set blackbox_sample_rate = 1/1
set blackbox_disable_gps = ON
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 20
set failsafe_switch_mode = STAGE2
set failsafe_procedure = GPS-RESCUE
set failsafe_recovery_delay = 20
set align_board_yaw = 180
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_ascend_rate = 800
set gps_rescue_return_alt = 20
set gps_rescue_ground_speed = 800
set gps_rescue_throttle_hover = 1265
set gps_rescue_throttle_p = 13
set gps_rescue_throttle_d = 17
set deadband = 2
set simplified_gyro_filter_multiplier = 200
set osd_warn_bitmask = 24573
set osd_rssi_alarm = 0
set osd_cap_alarm = 256
set osd_alt_alarm = 0
set osd_tim1 = 5121
set osd_tim2 = 1
set osd_vbat_pos = 480
set osd_rssi_pos = 491
set osd_link_quality_pos = 2496
set osd_link_tx_power_pos = 288
set osd_rssi_dbm_pos = 456
set osd_tim_1_pos = 491
set osd_flymode_pos = 1
set osd_vtx_channel_pos = 1
set osd_crosshairs_pos = 270
set osd_ah_pos = 142
set osd_current_pos = 481
set osd_mah_drawn_pos = 4
set osd_craft_name_pos = 2536
set osd_gps_speed_pos = 2272
set osd_gps_lon_pos = 2080
set osd_gps_lat_pos = 2048
set osd_gps_sats_pos = 2304
set osd_home_dir_pos = 2511
set osd_home_dist_pos = 2240
set osd_flight_dist_pos = 2208
set osd_compass_bar_pos = 46
set osd_altitude_pos = 14624
set osd_warnings_pos = 14441
set osd_avg_cell_voltage_pos = 2545
set osd_battery_usage_pos = 49386
set osd_disarmed_pos = 2251
set osd_flip_arrow_pos = 142
set osd_log_status_pos = 2282
set osd_stat_bitmask = 14116
set osd_gps_sats_show_hdop = ON
set osd_craftname_msgs = ON
set debug_mode = GYRO_SCALED
set rpm_filter_min_hz = 95
set craft_name = LINK QUALITY

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 78
set dterm_lpf1_dyn_max_hz = 157
set dterm_lpf1_dyn_expo = 6
set dterm_lpf1_type = BIQUAD
set dterm_lpf1_static_hz = 0
set dterm_lpf2_static_hz = 0
set iterm_rotation = ON
set iterm_relax = RPY
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set throttle_boost = 7
set p_pitch = 84
set i_pitch = 250
set d_pitch = 79
set f_pitch = 314
set p_roll = 67
set i_roll = 204
set d_roll = 62
set f_roll = 251
set p_yaw = 63
set i_yaw = 0
set d_yaw = 47
set f_yaw = 201
set horizon_transition = 0
set abs_control_gain = 10
set use_integrated_yaw = ON
set d_min_roll = 47
set d_min_pitch = 58
set d_min_yaw = 44
set auto_profile_cell_count = 6
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 50
set feedforward_jitter_factor = 8
set feedforward_boost = 16
set dyn_idle_min_rpm = 30
set simplified_pids_mode = RP
set simplified_master_multiplier = 150
set simplified_i_gain = 170
set simplified_d_gain = 105
set simplified_feedforward_gain = 140
set simplified_pitch_d_gain = 110
set simplified_pitch_pi_gain = 120
set simplified_dterm_filter_multiplier = 105
set tpa_breakpoint = 1450

profile 1

profile 2

profile 3

# restore original profile selection
profile 1

rateprofile 0

# rateprofile 0
set thr_mid = 23
set thr_expo = 80
set rates_type = BETAFLIGHT
set roll_rc_rate = 65
set pitch_rc_rate = 65
set yaw_rc_rate = 53
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 12
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

### Resources configuration

```txt
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: FREE
B01: FREE
B02: BEEPER
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: SERIAL_TX 1
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: PINIO 2
C01: ADC_BATT
C02: PINIO 1
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 1
C09: MOTOR 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 3
    CH4 : DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 1
```

### Flight controller

Mamba mk4 f722 mini and HJLRC F722 mini

### Other components

DJI Vista and Caddx Nebula Pro 

### How are the different components wired up

Standard connection between FC and DJI Vista using MSP on one of the port.

haslinghuis, 2022-11-10:
@howels can you please look at this

howels, 2022-11-10:
@OlegSyr I see what is happening here. DJI's OSD routine is actually still retrieving the NAME parameter but we have renamed to CRAFT_NAME.

@haslinghuis will we support `pilotConfigMutable()->name` going forward? Because DJI's original OSD implementation will continue to look in that location.

howels, 2022-11-10:
I can see the data we need written to screen when I enable FPV.WTF's OSD but the original DJI OSD simply renders "CRAFT_NAME" as the value.

howels, 2022-11-11:
Please test the fix using this @OlegSyr https://github.com/betaflight/betaflight/suites/9246443007/artifacts/432455722

OlegSyr, 2022-11-11:
@howels - provided solution works like a charm. I see LQ value now instead of "craft name" label.
Thanks a lot!

howels, 2022-11-11:
You can thank @haslinghuis for that, glad it's working.

OlegSyr, 2022-11-11:
I stand corrected:)
@haslinghuis - many thanks for resolving this issue and for relentless efforts in Betaflight development!

PR #11980 (2022-11-14): Fix compatibility for name field
Fixes: #11977
closed 2022-11-14 by blckmn