betaflight/betaflight #11981: ExpressLRS commands missing from 4.3.0 and 4.3.1 for BETAFPVF4SX1280 target

cynfewl, 2022-11-11:
### Describe the bug

Trying to configure the binding phrase on my BetaFPV F411 SPI board but I am not able to find any of the expresslrs options in the CLI.

board_name
board_name BETAFPVF4SX1280

# get rx_spi_protocol
rx_spi_protocol = EXPRESSLRS
Allowed values: V202_250K, V202_1M, SYMA_X, SYMA_X5C, CX10, CX10A, H8_3D, INAV, FRSKY_D, FRSKY_X, FLYSKY, FLYSKY_2A, KN, SFHSS, SPEKTRUM, FRSKY_X_LBT, REDPINE, FRSKY_X_V2, FRSKY_X_LBT_V2, EXPRESSLRS
Default value: V202_250K

# get expresslrs
###ERROR IN get: INVALID NAME###

# set expresslrs_uid
###ERROR IN set: INVALID NAME###

# set expresslrs_switch_mode
###ERROR IN set: INVALID NAME###

Has the ExpressLRS binding phrase been renamed in 4.3.0/4.3.1 or is there another method of setting this?

### To Reproduce

Connect to CLI and try and get/set any of the expresslrs options, including expresslrs_uid and expresslrs_switch_mode.

### Expected behavior

I should be able to set expresslrs_uid to my binding phrase and the switch mode to WIDE as I can on other targets such as CRAZYBEEF4SX1280.

### Diff configuration

```txt
# 

# diff all

# version
# Betaflight / STM32F411 (S411) 4.3.1 Jul 13 2022 / 03:32:52 (8d4f005) MSP API: 1.44
# config: manufacturer_id: BEFH, board_name: BETAFPVF4SX1280, version: b7a1de2d, date: 2022-11-06T06:14:40Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name BETAFPVF4SX1280
manufacturer_id BEFH
mcu_id 006400594e32500620333730
signature 

# name: 65ProHDzero

# feature
feature -SOFTSERIAL
feature -RX_PARALLEL_PWM

# serial
serial 0 1 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200

# beacon
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1700 2100 0 0
[... 329 lines cut ...]
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 84

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

### Flight controller

BetaFPV F4 5A 1S ELRS (SPI) https://betafpv.com/collections/brushless-flight-controller/products/f4-1s-5a-aio-brushless-flight-controller

haslinghuis, 2022-11-11:
Looks like an issue with ELRS 2.x

Still works on 4.4 with 10.9 master and ELRS 3.x:

```
# get expresslrs
expresslrs_uid = 0,0,203,105,237,216
Array length: 6
Default value: 0,0,0,0,0,0

expresslrs_domain = CE2400
Allowed values: ISM2400, CE2400
Default value: ISM2400

expresslrs_rate_index = 0
Allowed range: 0 - 3

expresslrs_switch_mode = HYBRID
Allowed values: WIDE, HYBRID
Default value: WIDE

expresslrs_model_id = 255
Allowed range: 0 - 255
```

cynfewl, 2022-11-11:
Is 4.4 ELRS 3.x only?

I did think of trying 4.4, but I wanted to upgrade all at once after 4.4 went final release.

David-OConnor, 2022-11-11:
Is this related? https://github.com/betaflight/betaflight/pull/11966

I don't think the ELRS version should make a diff here.

Is this due to the SX1280 unified targets being removed in the latest dev versions in favor of cloud builds?

cynfewl, 2022-11-12:
So it seems like using the **13-07-2022 13:13 - 4.3.1 (Legacy)** build rather than **13-07-2022 13:13 - 4.3.1** fixes this issue.

What is the difference between these two firmware versions?

damianqx, 2022-12-02:
Where did you download "13-07-2022 13:13 - 4.3.1 (Legacy) "? I have same problem. I can't use elrs

tomich, 2022-12-28:
Same thing happening on latest stable 4.3

ELRS is broken on target BETAFPVF4SX1280

Using 4.4-zulu is not an option as it uses ELRS 4 but also, an more important, it does not recognize gyro nor ESCs, so it is unusable :S

haslinghuis, 2022-12-28:
| Firmware | SPI ELRS | TX |
| --- | --- | --- |
| 4.3.2 | 2.x | 2.x |
| 4.4.x | 3.x | 3.x |

Using BETAFPVSX1280 with ICM20689 without any problems on latest BF 4.4 version with bidirectional dshot on BlueJay 18.x
You will have to upgrade the radio's TX to ELRS 3.x.

tomich, 2022-12-28:
Upgrading is not viable as there is no gyro, accel nor motors on 4.4.x on my BETAFPVF4SX1280

Are you using custom defines when building/flashing 4.4.x?

I'm using a DarwinFPV ELRS AIO F4 board (They call it AIO002 i think).

tomich, 2022-12-28:
I believe this may be related to:

https://github.com/betaflight/betaflight/issues/11375

https://github.com/betaflight/betaflight/issues/12049

Will have to test removing different commits or testing the builds mentioned on those threads

haslinghuis, 2022-12-28:
@tomich need `4.3.3` as it's not official yet.

This is the only commit: https://github.com/betaflight/betaflight/pull/12057

tomich, 2022-12-28:
Yes. 

This build:
https://github.com/betaflight/betaflight/pull/12057#issuecomment-1356225190

works fine for me and solves the issue for me. So it appears that the gyro/elrs issue is also related to:
https://github.com/betaflight/betaflight/issues/12033

@cynfewl please try this build and check if it also works for you. 
https://github.com/betaflight/betaflight/pull/12057#issuecomment-1356225190

That build works, but all stable releases, RCs, pre-releases and development releases (available in the configurator) have issues for me with this boards. 

DarwinFPV AIO 15A ELRS2

Target: BEFH/BETAFPVF4SX1280(STM32F411SX1280)

closed 2023-02-05 by github-actions[bot]

andrey2ag, 2023-06-08:
Do you have any updates on this? I'm having the same issue with the BETAFPVF411(STM32F411)