betaflight/betaflight #11985: Latest master not flyable

TheIsotopes, 2022-11-12:
### Describe the bug

using latest master on matekh743 as hexcopter. quad flip immediately after take off.

### To Reproduce

using latest master

### Expected behavior

good flying firmware

### Diff configuration

```txt
# Betaflight / STM32H743 (SH74) 4.4.0 Nov 12 2022 / 09:40:30 (931c982) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name MATEKH743
manufacturer_id MTKS

# name: HEXAMOVR

# feature
feature -OSD
feature LED_STRIP

# mixer
mixer HEX6X

# beacon
beacon RX_LOST
beacon RX_SET

# led
led 0 6,2::AW:0
led 1 6,3::AW:0
led 2 6,4::AW:0
led 3 6,5::AW:0
led 4 9,5::AW:0
led 5 9,4::AW:0
led 6 9,3::AW:0
led 7 9,2::AW:0

# mode_color
mode_color 6 1 3

# aux
aux 0 0 0 1900 2100 0 0
aux 1 13 1 900 1100 0 0
aux 2 36 1 1900 2100 0 0

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 125
set dyn_notch_count = 2
set dyn_notch_q = 350
set gyro_lpf1_dyn_min_hz = 0
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set rc_smoothing_auto_factor = 25
set rc_smoothing_auto_factor_throttle = 25
set serialrx_provider = GHST
set blackbox_sample_rate = 1/8
set dshot_idle_value = 400
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1800
set use_cbat_alerts = ON
set cbat_alert_percent = 20
set vbat_duration_for_warning = 50
set vbat_duration_for_critical = 20
set ibata_scale = 200
set beeper_dshot_beacon_tone = 4
set simplified_gyro_filter = OFF
set rpm_filter_q = 350
set rpm_filter_min_hz = 80
set craft_name = HEXAMOVR

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 60
set dterm_lpf1_dyn_max_hz = 120
set dterm_lpf2_static_hz = 120
set p_pitch = 32
set i_pitch = 58
set d_pitch = 32
set f_pitch = 87
set p_roll = 31
set i_roll = 56
set d_roll = 28
set f_roll = 84
set p_yaw = 31
set i_yaw = 56
set f_yaw = 84
set d_min_roll = 21
set d_min_pitch = 23
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 35
set feedforward_jitter_factor = 4
set feedforward_boost = 18
set simplified_d_gain = 70
set simplified_pi_gain = 70
set simplified_feedforward_gain = 70
set simplified_dterm_filter = OFF

rateprofile 0

# rateprofile 0
set roll_rc_rate = 20
set pitch_rc_rate = 20
set yaw_rc_rate = 20
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50
set roll_srate = 99
set pitch_srate = 99
set yaw_srate = 72
set throttle_limit_type = SCALE
set throttle_limit_percent = 80

# end the command batch
batch end
```

### Resources configuration

```txt
# Betaflight / STM32H743 (SH74) 4.4.0 Nov 12 2022 / 09:40:30 (931c982) MSP API: 1.45
```

### Flight controller

MATEKH743

KarateBrot, 2022-11-13:
Sure, I will immediately have a look in the evening. Weird, because there was no issue on my quad when testing but let's see what's up. Thanks for flagging it.

TheIsotopes, 2022-11-13:
@KarateBrot no problem, if you find something i can test it.

KarateBrot, 2022-11-13:
@TheIsotopes Thanks!

So the suspect is [**this change**](https://github.com/betaflight/betaflight/compare/cffa1845686647a20d361cb2d263ec942527b511..96a0e6098c18c61838f19b28cd659d7e4128d849). I reverted this in the compiled test files below.

## How to test

Before you do anything please enable blackbox logging and then go to the CLI to type:
```
set debug_mode = GYRO_SAMPLE
set gyro_debug_axis = PITCH
save
```
This will enable a debug mode where I can see what the RPM filter does before and after filtering on the pitch axis. The code change from above only gets applied to pitch and yaw axes, so it's  important the debug axis is NOT roll.

Now you can test carefully if the quad flips out again.
- If it flips out, please provide me with the log
- If it doesn't flip out, we are halfway there :)

## Test files

### [📁 **BF_4_4_RpmFilterFix_00.zip**](https://github.com/betaflight/betaflight/files/9997132/BF_4_4_RpmFilterFix_00.zip)

TheIsotopes, 2022-11-13:
@KarateBrot Quickly tested again a hover flight in the garden before the Munich football game begins. #GoBucs
Everything seems to be working as expected again.

KarateBrot, 2022-11-13:
@TheIsotopes Nice, thanks. Now I made another change (made a "custom memcpy"). Can you test this again? If it works we are golden and I'm going to push a fix immediately.

### [📁 **BF_4_4_RpmFilterFix_01.zip**](https://github.com/betaflight/betaflight/files/9997731/BF_4_4_RpmFilterFix_01.zip)

closed 2022-11-14 by blckmn