betaflight/betaflight #11991: HW Xrotor F7 missing onboard flash

mituritsyn, 2022-11-14:
### Describe the bug

No dataflash chip found

### To Reproduce

Update HW Xrotor F7 to latest BF 4.4 

### Expected behavior

16Mb of flash available

### Diff configuration

```txt
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Nov 14 2022 / 12:45:07 (2216b42db) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name HOBBYWING_XROTORF7CON
manufacturer_id HOWI

# master
set blackbox_device = NONE

profile 0

rateprofile 0

# end the command batch
batch end
```

### Resources configuration

```txt
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: MOTOR 4
B00: PINIO 1
B01: MOTOR 1
B02: PREINIT 2
B03: MOTOR 3
B04: MOTOR 2
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: CAMERA_CONTROL
B09: PINIO 2
B10: FREE
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_CURR
C01: ADC_BATT
C02: FREE
C03: GYRO_EXTI
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: BEEPER
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1 : MOTOR 4
    CH2 : MOTOR 3
TIM3:
    CH1 : MOTOR 2
    CH4 : MOTOR 1
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10:
    CH1 : CAMERA_CONTROL
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: MOTOR 4
B00: PINIO 1
B01: MOTOR 1
B02: PREINIT 2
B03: MOTOR 3
B04: MOTOR 2
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: CAMERA_CONTROL
B09: PINIO 2
B10: FREE
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_CURR
C01: ADC_BATT
C02: FREE
C03: GYRO_EXTI
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: BEEPER
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1 : MOTOR 4
    CH2 : MOTOR 3
TIM3:
    CH1 : MOTOR 2
    CH4 : MOTOR 1
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10:
    CH1 : CAMERA_CONTROL
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

HobbyWing xrotor f7

### Add any other context about the problem that you think might be relevant here

I've tryed to add FLASH_W25Q128FV to custom defines field with no success. Build log link:
https://build.betaflight.com/api/builds/4.4.0/HOBBYWING_XROTORF7CONV/16318da2-9609-d2a0-d22c-bd4bd4fbf5e1/log

mituritsyn, 2022-11-14:
@ASDosjani in discord helped to find workaround:
need to flash via 10.8 configurator to bypass new online build system

haslinghuis, 2022-11-15:
@mituritsyn we rather like to solve the problem here.

Besides `-DUSE_FLASH_W25Q128FV` (which could be wrong) we might need a combination with one of these:
```
#define USE_FLASH_M25P16           // 16MB Micron M25P16
#define USE_FLASH_W25N01G          // 1Gb NAND flash support
#define USE_FLASH_W25M             // 16, 32, 64 or 128MB Winbond stacked die support
#define USE_FLASH_W25M512          // 512Kb (256Kb x 2 stacked) NOR flash support
#define USE_FLASH_W25M02G          // 2Gb (1Gb x 2 stacked) NAND flash support
```

ASDosjani, 2022-11-15:
> @mituritsyn we rather like to solve the problem here.
>
> Besides `-DUSE_FLASH_W25Q128FV` (which could be wrong) we might need a combination with one of these:
>
> ```
> #define USE_FLASH_M25P16 [...]

I will make a PR tomorrow. The W25Q128FV is good, it's on the FC. Why would we need to combine this flag with the others? Those are ment for different chips.

haslinghuis, 2022-12-15:
@mituritsyn @ASDosjani assume issue is resolved?

Please also use:

    set blackbox_device = SPIFLASH

closed 2022-12-15 by haslinghuis

ASDosjani, 2022-12-15:
Yeah, this is resolved, I just forgot to mark this issue in my PR.