betaflight/betaflight #12003: After successfully modifying the power of VTX in the 4.3.0/4.3.1 firmware, it will not take effect after restarting.

yanwenjun, 2022-11-17:


yanwenjun, 2022-11-18:
> If you change the `vtx_power` in CLI and it does not get saved, then something is wrong in the firmware and not in the Configurator app. I move this to the firmware repo.

Thanks for the correction.

closed 2022-12-25 by github-actions[bot]

fam891, 2023-02-21:
Was there any update on this? The problem persists in 4.4.0 with the Speedybee TX800. After finally getting the VTX power set to 800mw, a power cycle has it drop back to 25mw.

haslinghuis, 2023-02-21:
Correct. Labels are wrong but it had to do with internal workings, so the json should be correct, although labels seems off.
Please consider joining our discord server and contacting Speedybee.

fam891, 2023-02-22:
After contacting SpeedyBee, there is a workaround that allows you to set the VTX power and have it stay after a power cycle. The workaround requires purposely introducing an error after setting the desired power level, so the Low Power Disarm feature cannot be used. The procedure that was given is below. However, this wasn't an issue in BF 4.2.11; it started with BF 4.3. Did something change about how the power level is sent to the VTX from 4.2.11 to 4.3? Is there a fix in Betaflight coming for this?

Issue: After the FC is flashed with 4.3 firmware, or later, every time it is powered on, the power becomes 25mW on the VTX (TX800).
Workaround:
1. Set the number of power groups to 5, and set the fifth group to an invalid value (888): The following 1-5 groups correspond to 25 milliwatts, 200 milliwatts, 400 milliwatts, 800 milliwatts, and invalid milliwatts. Click Save. (Note: 600 is the correct Value for the corresponding 800mw Label).

Power Level: 1, 2, 3, 4, 5
Value:  25, 200, 400, 600, 888
Label: 25, 200, 400, 800, 888

2. Set the power value to 800, and click save, if you find that the orange light of the VTX is on solid, it means that the VTX has turned on 800 milliwatts.

3. Then set the power to invalid 888, and click save, if you find that the orange light of VTX is still on solid, it means that the power of VTX is still 800 milliwatts. (This is set to prevent the FC from setting the power of VTX back to 25 milliwatts when the FC is restarted).

At this time, if you turn on the power again, the power of VTX will remain at 800 milliwatts.

haslinghuis, 2023-02-22:
If this issue is related to one product no - there is a workaround, but if it turns out to be a common problem we should have a look.

dmak, 2024-07-30:
>     1. Set the number of power groups to 5, and set the fifth group to an invalid value (888): The following 1-5 groups correspond to 25 milliwatts, 200 milliwatts, 400 milliwatts, 800 milliwatts, and invalid milliwatts. Click Save. (Note: 600 is the correct Value for the corresponding 800mw Label).
> 
>     2. Set the power value to 800, and click save, if you find that the orange light of the VTX is on solid, it means that the VTX has turned on 800 milliwatts.
> 
>     3. Then set the power to invalid 888, and click save, if you find that the orange light of VTX is still on solid, it means that the power of VTX is still 800 milliwatts. (This is set to prevent the FC from setting the power of VTX back to 25 milliwatts when the FC is restarted).
> 
> 
> At this time, if you turn on the power again, the power of VTX will remain at 800 milliwatts.

Any idea how to unlock SpeedyBee TX1600? I have used VTX table [from here](https://github.com/betaflight/firmware-presets/pull/456) and added one more level 888/888 as suggested above:

Power Level: 1, 2, 3, 4, 5
Value: 25, 100, 200, 600, 888
Label: 25, 200, 800, 1.6, 888

When I power on, no LED is flashing (OSD displays 25mV, Betaflight is set to 1.6, pit mode is off). When I try switching power level in Betaflight, nothing is changed in OSD.

When I press & hold the button on VTX for > 10 seconds, orange LED starts blinking once per 2 seconds (200mW?). When I press & hold the button on VTX for > 5 seconds, orange LED is permanently on. I have tried to set the power value to 888, but the value is not persisted after battery power off.

BlastAction, 2024-08-01:
Ich hab leider das gleiche Problem mit dem tx 1600. Ich komme auch nicht weiter. Hat einer schon eine Lösung gefunden? Ich wäre für jede Hilfe dankbar

dmak, 2025-04-18:
I have probably related issue with VTX 1600: I am not able to make Betaflight to control that VTX, in particular, to switch VTX to maximum power (orange LED is constantly ON). After power up, orange LED blinks once per second (is probably at 800mW?).

Betaflight v4.5.1.

[image]

kedeng, 2025-04-19:
> After power up, orange LED blinks once per second (is probably at 800mW?).

From https://store-fhxxhuiq8q.mybigcommerce.com/product_images/img_SpeedyBee_TX_ULTRA/SpeedyBee-TX-ULTR-EN.pdf

[image]

Orange LED blinks once per second means max power.

kedeng, 2025-04-19:
> I have probably related issue with VTX 1600: I am not able to make Betaflight to control that VTX, in particular, to switch VTX to maximum power (orange LED is constantly ON).

@dmak can you try this config?
[SpeedyBee-TX+ULTRA（CE）.json](https://github.com/user-attachments/files/19819899/SpeedyBee-TX%2BULTRA.CE.json)

dmak, 2025-04-21:
> [@dmak](https://github.com/dmak) can you try this config? [SpeedyBee-TX+ULTRA（CE）.json](https://github.com/user-attachments/files/19819899/SpeedyBee-TX%2BULTRA.CE.json)

I have applied that VTX table and I was able to get the orange LED permanently ON 🎉 Thanks! On that power level the range is really increased and VTX gets sensitively hotter. As to my observations the behaviour of orange led LED when applying the corresponding power level is:

  * OFF = 25mw
  * Blinking once in 3 seconds = 200mW
  * Blinking once in 2 seconds = 800mW
  * Blinking once in 1 second = 1000mW
  * Constant ON = 1600mW
  * Blinking 2 times in 1 second = PIT mode

That does not correspond to screenshot above, but probably I have "new revision" of SpeedyBee VTX?

kedeng, 2025-04-22:
> That does not correspond to screenshot above, but probably I have "new revision" of SpeedyBee VTX?

Not revision, different countries have different power limits.

kedeng, 2025-04-22:
> After restarting the plane was able to keep the 800mW vtx setting instead of reverting to 25mW.

From what I know, the revision TX800 has fixed this issue

dmak, 2025-04-22:
> Not revision, different countries have different power limits.

How SpeedyBee VTX "knows" in which country it is running? 😁I can only think of an option when VTx arrives in a locked state (red LED is blinking) and one needs to unlock it by pressing a button for 10s (check [this video](https://www.youtube.com/watch?v=WJXMTNHr2Cw)).

Anyway there are small nuances the way orange LED reacts: if one sees a different behaviour than I have posted, please share yours 🙏