betaflight/betaflight #12031: GPS not locking fast to sats if there is microsd in OpenLager

deeph46, 2022-12-01:
### Describe the bug

Hi,

I have installed GPS Beitian BN-880 module and OpenLager. But I have issue locking sats. I can't lock sats if microsd in OpenLager inserted. If I don't use microsd in OpenLager, I can get sats lock in couple of second.

Thank you

### To Reproduce

* Memory card inserted in OpenLager :
1. Memory card is inserted in OpenLager.
2. Connected a lipo.
3. I have to wait about 6 minutes to lock some sats. 

*Memory card NOT inserted in Open Lager : 
1. With no meory card inserted in OpenLager.
2. Connected Lipo
3. GPS will lock in 5s.

### Expected behavior

Can't lock GPS for short periode of time in cold or hot start.

### Diff configuration

```txt
# diff

# version
# Betaflight / STM32H743 (SH74) 4.3.2 Nov 28 2022 / 07:31:53 (60c9521) MSP API: 1.44
# config: manufacturer_id: JHEF, board_name: JHEH743_AIO, version: bff306ae, date: 2022-11-15T12:36:53Z

# start the command batch
batch start

board_name JHEH743_AIO
manufacturer_id JHEF

# feature
feature -RX_PARALLEL_PWM
feature SERVO_TILT
feature GPS
feature TELEMETRY

# serial
serial 0 2 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 5 1 115200 57600 0 115200
serial 7 128 115200 57600 0 2000000

# servo
servo 0 1250 1600 1500 -100 5

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 8 4 1700 2100 0 0
aux 2 13 2 1700 2100 0 0
aux 3 35 5 1700 2100 0 0
aux 4 40 3 1700 2100 0 0

# rxfail
rxfail 11 s 1000

# master
set acc_calibration = 32,-7,-73,1
set mag_align_pitch = 1800
set mag_align_yaw = 1800
set mag_calibration = -36,-68,34
set baro_hardware = BMP280
set rssi_channel = 12
set serialrx_provider = CRSF
set blackbox_device = SERIAL
set dshot_idle_value = 500
set dshot_bidir = ON
set motor_output_reordering = 1,0,3,2,4,5,6,7
set align_board_yaw = 45
set bat_capacity = 1300
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_set_home_point_once = ON
set osd_tim1 = 2561
set osd_vbat_pos = 2125
set osd_rssi_pos = 2083
set osd_link_quality_pos = 134
set osd_rssi_dbm_pos = 134
set osd_tim_1_pos = 2541
set osd_throttle_pos = 2382
set osd_gps_speed_pos = 2414
set osd_gps_sats_pos = 2105
set osd_home_dir_pos = 2159
set osd_home_dist_pos = 2446
set osd_warnings_pos = 2475
set osd_avg_cell_voltage_pos = 2093
set osd_disarmed_pos = 2509
set osd_displayport_device = MSP
set displayport_msp_serial = 5
set pinio_box = 40,255,255,255
set gyro_1_sensor_align = CW90FLIP
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 900

profile 0

# profile 0
set p_pitch = 51
set i_pitch = 101
set d_pitch = 50
set p_roll = 49
set i_roll = 96
set d_roll = 44
set p_yaw = 49
set i_yaw = 96
set d_min_roll = 33
set d_min_pitch = 37
set auto_profile_cell_count = 6
set feedforward_smooth_factor = 45
set feedforward_jitter_factor = 10
set dyn_idle_min_rpm = 10
set simplified_i_gain = 110
set simplified_d_gain = 110
set simplified_pi_gain = 110

rateprofile 0

# rateprofile 0
set thr_expo = 14
set rates_type = BETAFLIGHT
set roll_rc_rate = 137
set pitch_rc_rate = 137
set yaw_rc_rate = 142
set roll_expo = 15
set pitch_expo = 15
set yaw_expo = 12
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70

# end the command batch
batch end
```

### Resources configuration

```txt
# resource
resource BEEPER 1 D15
resource MOTOR 1 E09
resource MOTOR 2 B00
resource MOTOR 3 E11
resource MOTOR 4 B01
resource PPM 1 A03
resource LED_STRIP 1 D12
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 E08
resource SERIAL_TX 8 E01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 E07
resource SERIAL_RX 8 E00
resource I2C_SCL 1 B08
resource I2C_SCL 2 B10
resource I2C_SDA 1 B09
resource I2C_SDA 2 B11
resource LED 1 C13
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_SCK 4 E02
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MISO 4 E05
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource SPI_MOSI 4 E06
resource CAMERA_CONTROL 1 C08
resource ADC_BATT 1 C03
resource ADC_RSSI 1 C05
resource ADC_CURR 1 C02
resource ADC_EXT 1 C01
resource FLASH_CS 1 A15
resource OSD_CS 1 E04
resource GYRO_EXTI 1 D00
resource GYRO_EXTI 2 D08
resource GYRO_CS 1 A04
resource GYRO_CS 2 B12
```

### Flight controller

Jhemcu H743 AIO

### Other components

- Beitian BN-880
- OpenLager
- TBS Mambo
- Tracer RX
- DJI O3

pitts-mo, 2022-12-02:
@deeph46 I do not understand how this is a betaflight issue.

GPS signals are very weak and prone to interfearance from many sources. My first guesses here are the OpenLager mcu or it's wiring are eminating too much noise while in operation. So try to isolate the gps and it's wiring from everything else (espetially the OpenLager and it's wiring)

If that doesn't help try googling "Paweł Spychalski gps" i seem to remember he had a few videos that help explain gps lock time.

deeph46, 2022-12-03:
@pitss-mo I have tried giving a distance from OpenLager to GPS and yes I can get sats more faster (but not that fast like before I added OpenLager). It's 1 minute to lock the sats.

pitts-mo, 2022-12-03:
Sounds like good progress. :-) Personnaly i think you shpuld be happy if your first time lock averages ess than two minutes. But if ypu want other ideas to reduce GPS lock time..

 interferance effect is cumulative... use of vtx low power disarm helps.  And I have also observed DJI googles appear to add impedement to gps lock... so maybe leave dji goggles off or use a VTX power on switch until gps drift slows down in telemitry on your radio

You will find a lot more tricks people have found helpful to reduce gps lock time by via web search and FPV related discord servers.

i suggest closing this issue and join betaflight discord and ask additional questions in #support... while you are there you will see lots of other people discussing some pretty cool projects :-)

deeph46, 2022-12-06:
Thanks, pitts-mo, I have managed with the sattelite lock in seconds with 21 sats. 

Have to re-position OpenLager about 1 inch from GPS, also GPS cable is using shielded cable.

Thanks again...

closed 2022-12-06 by deeph46