betaflight/betaflight #12032: HGLRC XJB F428 GYRO not working after firmware update

knnadeera, 2022-12-02:
### Describe the bug

My HGLRC XJB F428 FC showing no GYRO after firmware updating to 4.3.
even downgrade the firmware to 4.1.1 which it came with still not working
I have two of them. one is defected and on is working.

### To Reproduce

doesn't show any gyro data
[image]

### Expected behavior

I just update the firmware and fc doesn't show any gyro data

### Diff configuration

```txt
# diff

# version
# Betaflight / OMNIBUSF4 (OBF4) 4.1.1 Nov 15 2019 / 12:41:27 (1e5e3d369) MSP API: 1.42

# start the command batch
batch start

board_name OMNIBUSF4
manufacturer_id AIRB

profile 0

rateprofile 0

# end the command batch
batch end
```

### Resources configuration

```txt
resource
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource PPM 1 B14
resource PWM 1 B14
resource PWM 2 B15
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource SONAR_TRIGGER 1 A01
resource SONAR_ECHO 1 A08
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 6 C07
resource INVERTER 1 C00
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B14
resource ADC_BATT 1 C02
resource ADC_CURR 1 C01
resource FLASH_CS 1 B03
resource OSD_CS 1 A15
resource RX_SPI_CS 1 D02
resource RX_SPI_EXTI 1 A00
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource USB_DETECT 1 C05

and

resource show
Currently active IO resource assignments:
(reboot to update)
A00: RX_SPI_EXTI
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: PREINIT 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B03: FLASH_CS
B04: BEEPER
B05: LED 1
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: FREE
B13: FREE
B14: FREE
B15: FREE
C00: INVERTER 1
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: FREE
C05: USB_DETECT
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
D02: RX_SPI_CS
```

### Flight controller

version Betaflight / OMNIBUSF4 (OBF4) 4.1.1 Nov 15 2019 / 12:41:27 (1e5e3d3) MSP API: 1.42

knnadeera, 2022-12-02:
@x4FF3 could you please help me?

pitts-mo, 2022-12-02:
I have these boards with MPU-6000 gyro and try to  keep them up to date with release builds. But I still have 4.3.1 on them.

When flashing did you try full chip erase and let it load custom defaults to see if gyro works prior to load of config?

If that doesn't work maybe check to see if hglrc used a different gyro on your board. Or check for hardware problems and disconnect everything from fc and reflash just to verify the gyro still doesn't work.

KarateBrot, 2022-12-02:
What gyro is on your FC? Seems like your FC version has a gyro on the board which is not yet defined in the unified target. If you don't know please contact the manufacturer to ask.

knnadeera, 2022-12-05:
> I have these boards with MPU-6000 gyro and try to keep them up to date with release builds. But I still have 4.3.1 on them.
>
> When flashing did you try full chip erase and let it load custom defaults [...]

yep. MPU-
[image]
6000

pitts-mo, 2022-12-05:
I currently believe the biggest problem is the contaminant and corrosion on the PCB shown. You can try to clean it with alcohol and old tooth brush but i do not think odds of recovery are good :-( ... prepare youself to replace this board and likely other boards from this build. :-( ... but worth a try to clean it first. Good luck. -p

KarateBrot, 2022-12-05:
After looking at your diff I suspect that there are no FC defaults applied. Did you click "Apply Defaults" on first connect to the configurator?

Anyway, copy paste this into the CLI and see what happens:

```
# Betaflight / STM32F405 (S405) 4.1.0 May 22 2019 / 02:24:46 (1541466da) MSP API: 1.42

#define USE_GYRO_SPI_MPU6500
#define USE_ACC_SPI_MPU6500
#define USE_GYRO_SPI_MPU6000
#define USE_ACC_SPI_MPU6000
#define USE_MAX7456

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name OMNIBUSF4
manufacturer_id AIRB

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource PPM 1 B14
resource PWM 1 B14
resource PWM 2 B15
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource SONAR_TRIGGER 1 A01
resource SONAR_ECHO 1 A08
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 6 C07
resource INVERTER 1 C00
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B14
resource ADC_BATT 1 C02
resource ADC_CURR 1 C01
resource FLASH_CS 1 B03
resource OSD_CS 1 A15
resource RX_SPI_CS 1 D02
resource RX_SPI_EXTI 1 A00
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource USB_DETECT 1 C05

# timer
timer B14 AF9
# pin B14: TIM12 CH1 (AF9)
timer B15 AF9
# pin B15: TIM12 CH2 (AF9)
timer C06 AF3
# pin C06: TIM8 CH1 (AF3)
timer C07 AF3
# pin C07: TIM8 CH2 (AF3)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer A03 AF1
# pin A03: TIM2 CH4 (AF1)
timer A02 AF1
# pin A02: TIM2 CH3 (AF1)
timer A01 AF2
# pin A01: TIM5 CH2 (AF2)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
timer A09 AF1
# pin A09: TIM1 CH2 (AF1)
timer A10 AF1
# pin A10: TIM1 CH3 (AF1)

# dma
dma ADC 2 1
# ADC 2: DMA2 Stream 3 Channel 1
dma pin C06 0
# pin C06: DMA2 Stream 2 Channel 0
dma pin C07 0
# pin C07: DMA2 Stream 2 Channel 0
dma pin C08 0
# pin C08: DMA2 Stream 2 Channel 0
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin A03 1
# pin A03: DMA1 Stream 6 Channel 3
dma pin A02 0
# pin A02: DMA1 Stream 1 Channel 3
dma pin A01 0
# pin A01: DMA1 Stream 4 Channel 6
dma pin A08 0
# pin A08: DMA2 Stream 6 Channel 0
dma pin A09 0
# pin A09: DMA2 Stream 6 Channel 0
dma pin A10 0
# pin A10: DMA2 Stream 6 Channel 0

# feature
feature OSD

# master
set mag_bustype = I2C
set mag_i2c_device = 2
set baro_bustype = I2C
set baro_i2c_device = 2
set rx_spi_bus = 3
set adc_device = 2
set blackbox_device = SPIFLASH
set current_meter = ADC
set battery_meter = ADC
set beeper_inversion = ON
set beeper_od = OFF
set system_hse_mhz = 8
set max7456_clock = DOUBLE
set max7456_spi_bus = 3
set dashboard_i2c_bus = 2
set flash_spi_bus = 3
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_sensor_align = CW180
set gyro_2_spibus = 1

# save configuration
save
```

knnadeera, 2022-12-07:
> # Betaflight / STM32F405 (S405) 4.1.0 May 22 2019 / 02:24:46 (1541466da) MSP API: 1.42
>
> #define USE_GYRO_SPI_MPU6500
> #define USE_ACC_SPI_MPU6500
> #define USE_GYRO_SPI_MPU6000
> #define [...]

I clicked "Apply Defaults" after flashing. and also I paste your text in CLI. but still not working.

KarateBrot, 2022-12-07:
Did you select the correct target for your board before flashing?

knnadeera, 2022-12-07:
> Did you select the correct target for your board before flashing?

yes I did.

closed 2023-01-15 by github-actions[bot]