betaflight/betaflight #12033: Darwin FPV BETAFPVF4SX1280 ELRS SPI will not bind in 4.3.2

joeyw007, 2022-12-03:
### Describe the bug

ELRS SPI RX will not bind to TX, works fine in 4.3.1.

### To Reproduce

- Fresh load 4.3.2
- go to receiver tab and enable SPI RX
- set provider to EXPRESSLRS
- Enable Telemetry
- Save and reboot
- return to receiver tab start binding procedure by putting TX into bind mode then click "Bind Receiver" button or go to CLI and enter rx_bind

### Expected behavior

expect bind

### Diff configuration

```txt
# version
# Betaflight / STM32F411 (S411) 4.3.2 Nov 28 2022 / 07:27:10 (60c9521) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKF411, version: bff306ae, date: 2022-11-15T12:36:53Z

# start the command batch
batch start

board_name MATEKF411
manufacturer_id MTKS

# name: Baby Ape Pro

# resources
resource BEEPER 1 B02
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B06
resource MOTOR 4 B07
resource MOTOR 5 B03
resource MOTOR 6 B10
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
[... 1220 lines cut ...]
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

Darwin FVP AIO Baby Ape FC BETAFPVF4SX1280

### Other components

Darwin FPV VTX006, Runcam Nano 4

### Add any other context about the problem that you think might be relevant here

Tried all Darwin FPV binding procedures, ELRS website binding procedures, ELRS troubleshooting, ELRS discord support troubleshooting. Tried converting passphrase to UID number and entering in CLI, tried resetting UID to 0 in CLI and then binding. Tried using stock CLI dump/diff and then binding. Tried everything I'm aware of and cannot get to bind.

joeyw007, 2022-12-03:
Wrong Diff

dlkj, 2022-12-03:
I'm having a similar problem with CRAZYBEEF4SX1280 with the Happymodel Mobula7 v2/1S - SPI ELRS/X12 5-IN-1 AIO.

Won't bind with ELRS passphrase or manual binding. Reverting back to 4.3.1 fixes the problem - binds with passphrase.

Let me know if I can provide any diagnostics. Tried 4.3.2 and 4.3.2 (legacy) via the configurator.

haslinghuis, 2022-12-03:
The good news is that on firmware 4.4 with SPI ELRS 3.0 binding works (on BETAFPVF4SX1280) even using Bind Receiver button here:
[image]

haslinghuis, 2022-12-03:
This is probably the commit which breaks it for SPI ELRS 2.x: https://github.com/betaflight/betaflight/commit/d0199eaa607b49bd5e598b782deec2a9def1c3ed

joeyw007, 2022-12-03:
Does it have anything to do with D8 working like a champ now in 4.3.2 on crazybeeF4FR and matekf411rx boards. Just curious, fixing one seems to break the other. I had to use RC1 after the 4.3 Final on many D8 quads to fly 4.3. Thanks you guys do a great job. I bet it's rough with all the new tech in the hobby.

haslinghuis, 2022-12-03:
Exactly 4.3.2 exclusive mitigates the issues found with Frsky SPI as we were obvious not aware this would break ELRS 2.x SPI.

nerdCopter, 2022-12-03:
temporary solution -- unofficial builds without said-breaking commit.
[betaflight_4.3.2_CRAZYBEEF4SX1280_8d3ae8f51.hex.zip](https://github.com/betaflight/betaflight/files/10147308/betaflight_4.3.2_CRAZYBEEF4SX1280_8d3ae8f51.hex.zip)
[betaflight_4.3.2_BETAFPVF4SX1280_8d3ae8f51.hex.zip](https://github.com/betaflight/betaflight/files/10147309/betaflight_4.3.2_BETAFPVF4SX1280_8d3ae8f51.hex.zip)

i.e. rebase dropping `d0199ea`.

git log 
```
8d3ae8f51 Fix read of DPS310 coef C11 (#12011)
6a8be3a60 4.3 maint SteveCEvans (#12009)
2b4118081 Raise upper limit of dyn_idle_min_rpm
1da3906eb allow dyn_notch_min_Hz as low as 20Hz
cdb782711 MSC - Fix MSC on H7.  Commit 9629f966760580f25171a3c2f483789689a3241c
d05c7c545 Incrementing version
8d4f00532 Update version.h
```

joeyw007, 2022-12-03:
Yup that fixed it for me, thanks a million. I wonder how the crazybee board came out. don't have one to test yet

nerdCopter, 2022-12-06:
@haslinghuis , this is basically a duplicate of https://github.com/betaflight/betaflight/issues/11842
i believe cherry pick of https://github.com/betaflight/betaflight/pull/11845/commits would also fix this, but i'm not around my hardware for testing. (or would such break FrSky SPI yet again??)

haslinghuis, 2022-12-06:
#11842 solves ELRS bind on 4.4 and breaks it on 4.3.2 and a fix will probably work against Frsky SPI.

**Advisory**:
SPI ELRS 2.x users are recommended to stay on firmware 4.3.1 or use 4.4 using ELRS 3.x as ELRS 2.x is no longer supported.

johnkerman, 2022-12-10:
@haslinghuis as you said, "ELRS 2.x is no longer supported".
Then, what will you suggest to betafpv literadio 2 (elrs) users, which have custom firmware?
I think, this will be a huge pain for all of them.

haslinghuis, 2022-12-17:
@joeyw007 please test [ new firmware:point_up: ](https://github.com/betaflight/betaflight/pull/12057#issuecomment-1356225190)

joeyw007, 2022-12-17:
@haslinghuis Get back to you. Thought I already tested but will test this links firmware in case different, NP. On another topic the Configurator release of 4.3.2 is still randomly failsafing on Happymodel CRAZYBEEF4FR board with BMI270 Gyro when using FrSky D8. Also reduced range and telemetry lost warnings constantly. Still gotta run RC1 on two of those. Working great on all other boards I've tried it on, mostly MatekF411RX targets. Get back with you soon.

joeyw007, 2022-12-17:
@haslinghuis First pack I flew around the yard. Second pack I flew a few miles to the interstate. All is good on my end!

haslinghuis, 2022-12-19:
Fixed in https://github.com/betaflight/betaflight/pull/12057

closed 2022-12-19 by haslinghuis