betaflight/betaflight #12034: Feedforward in angle mode does not respect angle rates and uses acro rates.

ChrisRosser, 2022-12-03:
### Describe the bug

In BF4.3 feedforward was added for ANGLE mode. The current implementation uses the ACRO rates set in BF when calculating the feedforward component. While it provides a useful inputs to push the quad into sharp moves the push is mishapen and causes overshoot and bounceback due to the very different scaling and expo of ACRO rates compared to ANGLE rates.

This causes the following undesirable behaviour:
1. Feedforward shows a double dip when crossing the center stick position (due to the expo on acro rates) when there is no corresponding change in rate of setpoint change (which is calculated using angle rates which don't have expo).
2. Normal feedforward values that give good setpoint tracking in acro mode (well matched to PI and D) don't work well for angle mode because in angle mode PI and D are using totally different rates to FF.
3. Adding enough FF to reduce lag between setpoint and gyro causes big wobbles.

FF OFF
[image]

FF ON:
[image]

### To Reproduce

Set quad in angle mode with a non-zero feedforward value.

### Expected behavior

FF should be the derivative of the setpoint passed into the gyro loop. In angle mode that setpoint does not come directly from the RC Command but from a separate angle controller. In angle mode FF should be taken as the derivative of the setpoint created by the angle controller.

### Diff configuration

```txt
All
```

### Resources configuration

```txt
All
```

### Flight controller

All

ChrisRosser, 2022-12-03:
Looking at the code in pid.cand feedforward.c

At present FF is calculated from RawSetpoint in all modes. I think, if possible, it should use currentPidSetpoint (or similar) so that it respects the different ways the setpoint is calculated depending on the flight mode.

robertrypula, 2022-12-14:
Not sure if I understand the issue correctly but does this causes problems when tuning Feed Forward by the basement tuning method in ANGLE mode? See the video (at 8:07):

https://youtu.be/sqT4MACi3d8?t=487

Author of PIDtoolbox suggested that we can tune in ANGLE mode everything except Feed Forward. Personally I get totally different results between ANGLE/ACRO on the same sliders settings:

[image]

ChrisRosser, 2022-12-15:
Hi Robert,

Yes this is the reason. I'm working on this with the BF devs to upgrade the
angle controller to allow acro FF tuning in angle mode and improve angle
mode flight performance.

Thanks,

Chris

robertrypula, 2022-12-15:
@ChrisRosser 
"...to allow acro FF tuning in angle mode..."

That would be really, really great to have it. I can't fly proper acro in my house (or at least can't provide as much input as in angle). BTW I watched a lot of your YT videos. Keep the good work man, it really adds value to the community!

PS I guess it would be good to somehow ping PIDtoolbox author to make him aware of this thread. He explicitly mentioned it in his video that FF doesn't work with ANGLE and is don't know why. Quote from him:

```
"So in order to test feed forward we're gonna have to unfortunately
use rate mode. So although feed forward is actually working in angle
mode in betaflight 4.3 it wasn't giving the same kind of output that
we get or that we see when we're using acro mode [not sure why??]"
```

Is he available on Github? I'm sorry for "he", I don't know his real name :)

ctzsnooze, 2023-06-16:
This has been fixed now, 4.4.1 supports feedforward properly.

closed 2023-06-16 by ctzsnooze

Wildtimo, 2024-02-11:
I don't think this was fixed properly because I'm on 4.4.2 and still have this bug with high acro rates. (1200 deg/s)