betaflight/betaflight #12047: Building Targets has been made far more difficult

Quick-Flash, 2022-12-12:
### Is your feature request related to a problem? Please describe

Building a target in BF has become a lot more difficult. This is due to the new cloud building system. For developers who need to actively develop BF relying on the cloud build system is not a solution.

### Describe the solution you'd like

In the BF docs, https://github.com/betaflight/betaflight/blob/master/docs/development/Development.md?plain=1#L131 it describes the new build process that developers should make. The information included is not super great. Putting an example in the codebase for developers to modify for their needs is a better solution. 

### Describe alternatives you've considered

Do nothing and raise the bar to start developing BF code.

### Other information

Making things as easy as possible for developers is one way to help the future of BF.

sugaarK, 2022-12-14:
Lols

blckmn, 2022-12-18:
@Quick-Flash it has actually been made much easier. Whilst the documentation was a little light on, there is an element of experience required granted.

To make it even easier for developers with less experience, I have updated the documentation to give a fully working example:
https://github.com/betaflight/betaflight/pull/12060

Hope that helps.

Quick-Flash, 2022-12-19:
> Lols… you can still build hexs locally same as you have been able to all along,

No the standard unified targets no longer work and people in the discord have shown difficulty in getting the new system working if you really look.

closed 2022-12-19 by Quick-Flash

hydra, 2023-02-09:
@Quick-Flash This should help - https://github.com/betaflight/betaflight/pull/12335

hydra, 2023-02-10:
Also, there's this new PR which moves the target.h files back into the main repo: https://github.com/betaflight/betaflight/pull/12341