betaflight/betaflight #12048: MAMBA F722s FC connect error after reboot

alienwaredream, 2022-12-13:
### Describe the bug

Hi everyone, I have a mamba F722s MK2 flight controller, which I bought around 2019. I recently flashed the latest version of betaflight firmware and found that whenever the flight controller restarts, it will periodically fail. For example, the connection is normal for the first time. After clicking the "Save and Restart" button, the gyroscope and accelerometer cannot be recognized on the configuration program. Click the "Save and Restart" button again and the recognition will be successful again.
After that, I tried to flash firmware such as betaflight 4.1 4.2, inav, emuflight, etc., and there was no problem every time I restarted. So I think it's a firmware bug?
I recorded a video on YouTube https://www.youtube.com/watch?v=jzmbpyfHgIs

### To Reproduce

restarting the flight control

### Expected behavior

This problem only occurs in betaflight 4.3.0 and above

### Diff configuration

```txt
ntering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.2 Nov 28 2022 / 07:30:19 (60c9521) MSP API: 1.44
# config: manufacturer_id: DIAT, board_name: MAMBAF722, version: 03213dd9, date: 2022-11-13T13:57:08Z

# start the command batch
batch start

board_name MAMBAF722
manufacturer_id DIAT

# feature
feature -RX_PARALLEL_PWM

# master
set acc_calibration = 12,-117,23,1
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0

# end the command batch
batch end
```

### Resources configuration

```txt
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: SERIAL_RX 1
B08: CAMERA_CONTROL
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 1
C09: MOTOR 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH3 : CAMERA_CONTROL
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 3
    CH4 : DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 1
```

### Flight controller

Diatone Mamba F722s  (https://wikifpv.com/index.php/electronics/flight-controllers/item/diatone-mamba-f722s)

alienwaredream, 2022-12-14:
config
https://raw.githubusercontent.com/betaflight/unified-targets/master/configs/default/DIAT-MAMBAF722.config

haslinghuis, 2022-12-15:
This problem is intermittent present on different boards as we rely on deprecated API's using USB serial communication.

alienwaredream, 2023-01-03:
> This problem is intermittent present on different boards as we rely on deprecated API's using USB serial communication.

So is there a plan to fix it?

closed 2026-09-26 by blckmn