betaflight/betaflight #12049: 4.4.0 CRAZYBEEF4SX1280 AIO no gyro detected

L1cardo, 2022-12-14:
### Describe the bug

After flashing the firmware, bf-configurator says there is no gyro detected.
I tried to use this https://github.com/betaflight/unified-targets/blob/master/configs/default/HAMO-CRAZYBEEF4SX1280.config config, and it si still not working.

### To Reproduce

Flash 4.4.0-RC1

### Expected behavior

gyro to work

### Diff configuration

```txt
# diff

# version
# Betaflight / STM32F411 (S411) 4.4.0 Dec 14 2022 / 04:37:32 (103f238ee) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name CRAZYBEEF4SX1280
manufacturer_id HAMO

# serial
serial 1 2048 115200 57600 0 115200

# beacon
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 13 3 1700 2100 0 0
aux 3 35 2 1700 2100 0 0
aux 4 36 4 1700 2100 0 0

# adjrange
adjrange 0 0 1 900 1200 12 1 0 0
adjrange 1 0 1 1200 1800 12 1 0 0
adjrange 2 0 1 1800 2100 12 1 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWRACE  L FACTORY 5333 5373 5413 5453 5493 5533 5573 5613
vtxtable powerlevels 5
vtxtable powervalues 10 2 10 16 26
vtxtable powerlabels 0 RCE 25 100 400

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_type = PT2
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
set acc_calibration = 46,21,268,1
set rc_smoothing_auto_factor = 120
set serialrx_provider = CRSF
set dshot_idle_value = 1200
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set failsafe_delay = 4
set failsafe_recovery_delay = 20
set vbat_min_cell_voltage = 290
set vbat_warning_cell_voltage = 310
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 1
set yaw_deadband = 1
set simplified_gyro_filter = OFF
set report_cell_voltage = ON
set osd_rssi_pos = 314
set osd_link_quality_pos = 2104
set osd_rssi_dbm_pos = 312
set osd_tim_2_pos = 2455
set osd_flymode_pos = 2080
set osd_throttle_pos = 2425
set osd_vtx_channel_pos = 2133
set osd_current_pos = 288
set osd_mah_drawn_pos = 320
set osd_craft_name_pos = 2440
set osd_warnings_pos = 2345
set osd_avg_cell_voltage_pos = 2432
set osd_disarmed_pos = 2314
set osd_esc_rpm_pos = 161
set osd_flip_arrow_pos = 2112
set osd_core_temp_pos = 280
set debug_mode = RX_EXPRESSLRS_SPI
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 5
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5917
set vcd_video_system = NTSC

profile 0

# profile 0
set dterm_lpf1_type = PT2
set dterm_lpf2_type = PT2
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 65
set tpa_breakpoint = 1250

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 127
set pitch_rc_rate = 127
set yaw_rc_rate = 100
set roll_expo = 40
set pitch_expo = 40
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 75

# end the command batch
batch end
```

### Resources configuration

```txt
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: FREE
A04: PREINIT 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: PREINIT 2
A15: PREINIT 3
B00: ADC_BATT
B01: ADC_CURR
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: FREE
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 1
TIM3: FREE
TIM4:
    CH1 : MOTOR 2
    CH2 : MOTOR 3
    CH3 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 2
DMA1 Stream 1: MOTOR 1
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 3
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 4
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

HAMO/CRAZYBEEF4SX1280(STM32F411)

alienwaredream, 2022-12-14:
哎呦 这不是做模拟器连接器的大佬么！

haslinghuis, 2022-12-15:
Status report (cli) please.

Which gyro is affected?

jawz101, 2022-12-15:
For what it's worth I have a Mobula 7 ELRS whoop which uses this board and my gyro is detected.

Mine has the BMI270 gyro and my resources look different than the above output.

Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270

chrsdion, 2022-12-16:
I have the Mobula 6 ELRS with the ICM 20689 and I my gyro is detected.

codeincontext, 2022-12-19:
I've had this problem as well with the same board. I'll need to look into which gyro it has

phobos-, 2022-12-19:
@haslinghuis I suspect it might be ICM42688P. I have boards with ICM20689, MPU6000 and BMI270 and they all work.

haslinghuis, 2022-12-19:
@phobos- Yes the defines were added last week. If flashed before this time the firmware does not include `ICM42688P` 👍

JRBREWER, 2022-12-31:
I've got a darwinfpv tiny ape and I flashed 4.4 to get elrs 3.0 and I get the same problem no gyro! I also get no light on the board!

HVieiraFlyaround, 2023-01-06:
> I've got a darwinfpv tiny ape and I flashed 4.4 to get elrs 3.0 and I get the same problem no gyro! I also get no light on the board!

Apen the same to me. :(

haslinghuis, 2023-01-06:
Darwin users: try target `DARWINF4SX1280HD`

JRBREWER, 2023-01-06:
I have fixed the issue
You need to add `DARWINF4SX1280` as the target and extra option
"USE_MAX7456"

haslinghuis, 2023-01-06:
:confused: 

There is no DARWINSX1280 target only DARWINSX1280HD and it includes `USE_MAX7456`

[image]

[image]

JRBREWER, 2023-01-06:
Yes that is the one!

L1cardo, 2023-01-26:
Sorry to bother you guys! It turns out my problem was a demage problem. I even don't know when it got demage, it didn't crash once!
I reached out the manufacture and they said there is a demage on the board and replaced the aio board, then the problem is solved.

closed 2023-01-26 by haslinghuis