betaflight/betaflight #12053: Custom Defaults are not applied on FLYWOOF745NANO under 4.4.0 RC1

jseaber, 2022-12-15:


haslinghuis, 2022-12-16:
Configuration seems to have issues.

You don't need `SMARTPORT` if using `CRSF`.

Please try a classic build and report back.

jseaber, 2022-12-16:
Right, `SMARTPORT` was on by default and I did not deselect. Will try classic build and report back ASAP (quad is not with me tonight).

Should this prevent the Configurator from loading resources, though? Note that manually copying and pasting resources into CLI succeeds.

haslinghuis, 2022-12-16:
Seems there are issues with the target config. CLI is like putty - a terminal to headless firmware.

jseaber, 2022-12-16:
Confirmed: Flashing a "Classic" build successfully applies Defaults. All ports are available.

haslinghuis, 2022-12-17:
Please report `status` command (checking gyro/acc/baro/etc) and `flash_info`
Try non-classic with `SMARTPORT` removed.

jseaber, 2022-12-17:
# "Classic" build

```
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# status
MCU F745 Clock=216MHz, Vref=3.30V, Core temp=31degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 4218, max available: 32768
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MSP
System Uptime: 9 seconds, Current Time: 2022-12-17T00:16:20.840+00:00
CPU:24%, cycle time: 124, GYRO rate: 8064, RX rate: 15, System rate: 10
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 7
SD card: Not configured
Arming disable flags: RXLOSS CLI MSP RPMFILTER

# flash_info
Flash sectors=128, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=8388608
Partitions:
  0: FLASHFS   0 127
FlashFS size=8388608, usedSize=8388608
```

# SMARTPORT removed

Bug persists after Applying Defaults

```
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# status
MCU F745 Clock=216MHz, Vref=3.30V, Core temp=52degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CUSTOM DEFAULTS, size: 3701, max available: 32768
Devices detected: SPI:0, I2C:0
Gyros detected: locked

OSD: NONE
System Uptime: 23 seconds, Current Time: 2022-12-17T00:20:53.578+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 0, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
Arming disable flags: NOGYRO BOOTGRACE CLI MSP

# flash_info
Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=0, totalSize=0
Partitions:
FlashFS size=0, usedSize=0

```

Build Options for reference:
[image]

jseaber, 2022-12-17:
Oddly, I cannot reproduce this on my macbook under the same conditions, using the same USB cable and same Configurator v10.9.0 (7df8d62). Resources are always present when flashing from the mac. 😕

blckmn, 2022-12-18:
When the bug presents itself, rather than running the configuration from github, does a reset within CLI rectify anything?

haslinghuis, 2022-12-18:
`MSP_RESET_CONF` does seem to hang up communication intermittent - have to cold plug to regain access.

Same is observed with `MSP_SET_REBOOT`, (very) intermittent.

Using `defaults` in cli seems more robust (with some errors):

[image]

@jseaber can you please check using `defaults` command in `CLI` both on Windows and Mac

jseaber, 2022-12-19:
@blckmn Currently unable to replicate on the same PC, now running Configurator 10.9.0-RC2. When the bug arose previously, `defaults nosave` did not help. I'll t

@haslinghuis Here's output under Windows 10 immediatelly after running defaults

```
# status
MCU F745 Clock=216MHz, Vref=3.30V, Core temp=58degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3356, max available: 32768
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MAX7456
System Uptime: 20 seconds, Current Time: 2022-12-19T00:50:51.462+00:00
CPU:23%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 2 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 7
Arming disable flags: RXLOSS CLI MSP MOTOR_PROTO

# flash_info
Flash sectors=128, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=8388608
Partitions:
  0: FLASHFS   0 127
FlashFS size=8388608, usedSize=8388608
```

and on the Mac running Monterey:

```
# # Note: This was the mac
# status
MCU F745 Clock=216MHz, Vref=3.30V, Core temp=53degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3356, max available: 32768
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MAX7456
System Uptime: 27 seconds, Current Time: 2022-12-19T00:52:56.413+00:00
CPU:24%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 7
Arming disable flags: RXLOSS CLI MSP MOTOR_PROTO
# flash_info
Flash sectors=128, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=8388608
Partitions:
  0: FLASHFS   0 127
FlashFS size=8388608, usedSize=8388608
#
```

I'll keep experimenting.

jseaber, 2022-12-19:
Well, that didn't take long! Configurator 10.9.0-RC1 triggered the bug on the first attempt to flash a "classic" build. I see the following in the log:

```
2022-12-18 @18:57:58 -- Erased 1024 kB of flash successfully
2022-12-18 @18:58:37 -- USB device successfully closed
2022-12-18 @18:58:38 -- Query board information to preselect right firmware
2022-12-18 @18:58:38 -- Configurator failed to verify the board, if this does not work please try switching tab slowly to retry, make a new usb connection or connect first if you might have forgotten to apply custom defaults
```

Info in bugged state on Windows, Configurator 10.9.0-RC1:
```
status
MCU F745 Clock=216MHz, Vref=3.30V, Core temp=59degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CUSTOM DEFAULTS, size: 4218, max available: 32768
Devices detected: SPI:0, I2C:0
Gyros detected: locked

OSD: NONE
System Uptime: 35 seconds, Current Time: 2022-12-19T01:00:41.786+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 0, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: NOGYRO BOOTGRACE CLI MSP

# flash_info
Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=0, totalSize=0
Partitions:
FlashFS size=0, usedSize=0

# 
```

After running `defaults` on Windows, Configurator 10.9.0-RC1:

```
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# status
MCU F745 Clock=216MHz, Vref=3.30V, Core temp=59degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CUSTOM DEFAULTS, size: 4218, max available: 32768
Devices detected: SPI:0, I2C:0
Gyros detected: locked

OSD: NONE
System Uptime: 29 seconds, Current Time: 2022-12-19T01:01:47.292+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 0, System rate: 10
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: NOGYRO BOOTGRACE CLI MSP

# flash_info
Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=0, totalSize=0
Partitions:
FlashFS size=0, usedSize=0
```

Still running in the broken state, the Mac and Configurator 10.9.0-RC2 returns the same info:
```
Mac in bugged state, RC2:
#
# status
MCU F745 Clock=216MHz, Vref=3.30V, Core temp=57degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CUSTOM DEFAULTS, size: 4218, max available: 32768
Devices detected: SPI:0, I2C:0
Gyros detected: locked
OSD: NONE
System Uptime: 9 seconds, Current Time: 2022-12-19T01:03:16.695+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 0, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: NOGYRO BOOTGRACE CLI MSP
# flash_info
Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=0, totalSize=0
Partitions:
FlashFS size=0, usedSize=0
#
```

Running `defaults` returns the same info as well.

haslinghuis, 2022-12-19:
Repeat on RC2 please

jseaber, 2022-12-19:
Okay, flashing the "classic" build under Configurator 10.9.0-RC2 on Win10, log again shows:
```
2022-12-18 @19:17:35 -- Erased 1024 kB of flash successfully
2022-12-18 @19:18:14 -- USB device successfully closed
2022-12-18 @19:18:16 -- Query board information to preselect right firmware
2022-12-18 @19:18:16 -- Configurator failed to verify the board, if this does not work please try switching tab slowly to retry, make a new usb connection or connect first if you might have forgotten to apply custom defaults
```

The bug is **absent**:

```
# 

# status
MCU F745 Clock=216MHz, Vref=3.30V, Core temp=58degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 4218, max available: 32768
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MAX7456
System Uptime: 9 seconds, Current Time: 2022-12-19T01:19:05.674+00:00
CPU:24%, cycle time: 126, GYRO rate: 7936, RX rate: 15, System rate: 9
Voltage: 2 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 7
SD card: Not configured
Arming disable flags: RXLOSS CLI MSP MOTOR_PROTO

# flash_info
Flash sectors=128, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=8388608
Partitions:
  0: FLASHFS   0 127
FlashFS size=8388608, usedSize=8388608

# 
```

haslinghuis, 2022-12-19:
[image]

After flashing this output is expected as defaults have not been loaded.

But absent. Does this mean RC2 solves the issue?

jseaber, 2022-12-19:
> Bug absent. Does this mean RC2 solves the issue?

Flashing two more 4.4.0 builds under RC2 are also free of the bug.

It's difficult to be confident here, but so far, RC2 has a clean record and RC1 does not.

haslinghuis, 2023-06-01:
Closing as fixed in 4.4.2 release.

closed 2023-06-01 by haslinghuis