betaflight/betaflight #12061: TBS M8-2 GPS unavailable in Betaflight 4.4 RC1

hprompt, 2022-12-19:
### Describe the bug

Upon full chip erase and successful flashing of BF 4.4.0 RC1 via configurator 10.9.0 (RC1 or RC2) my TBS M8-2 GPS shows unavailable.

### To Reproduce

Full chip erase and flash BF 4.4.0 RC1 via configurator 10.9.0 (RC1 or RC2).
in Ports tab for UART4 under Sensor input select GPS /AUTO then save and reboot.
review Ports tab again to see if settings are saved, GPS/57600
under Configuration tab enable GPS
UBLOX Protocol
Auto Config
Use Galileo
Set Home Point Once
Ground Assistance Type North America
save and reboot 
GPS not enabled
CLI
# feature
feature GPS
unavailable

### Expected behavior

all previous BF versions from 4.2.11 to 4.3.2 have enabled GPS using the same TBS M8-2 GPS  

### Diff configuration

```txt
using same hardware from BF versions 4.2.11 to 4.3.2
T-Motor F7 F55A ProII - TmotorF7 target 4.4.0-RC1 [14-Dec-2022]

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Dec 19 2022 / 07:00:49 (103f238ee) MSP API: 1.45

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name TMOTORF7
manufacturer_id TMTR
mcu_id 0051002f3538510e35343631
signature 

# serial
serial 1 1 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 4 64 115200 57600 0 115200

# master
set acc_calibration = 24,117,118,1

profile 0

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

### Resources configuration

```txt
# resources
resource BEEPER 1 C13
resource MOTOR 1 C06
resource MOTOR 2 C07
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 B06
resource MOTOR 6 B08
resource MOTOR 7 B07
resource MOTOR 8 C09
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 B10
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 NONE
resource I2C_SDA 2 B11
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 B02
resource LED 2 NONE
resource LED 3 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_SCK 4 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MISO 4 NONE
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource SPI_MOSI 4 NONE
resource CAMERA_CONTROL 1 B09
resource ADC_BATT 1 C01
resource ADC_RSSI 1 C03
resource ADC_CURR 1 C02
resource ADC_EXT 1 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 C08
resource OSD_CS 1 B12
resource RX_SPI_CS 1 NONE
resource RX_SPI_EXTI 1 NONE
resource RX_SPI_BIND 1 NONE
resource RX_SPI_LED 1 NONE
resource RX_SPI_EXPRESSLRS_RESET 1 NONE
resource RX_SPI_EXPRESSLRS_BUSY 1 NONE
resource GYRO_EXTI 1 C04
resource GYRO_EXTI 2 NONE
resource GYRO_CS 1 A04
resource GYRO_CS 2 NONE
resource USB_DETECT 1 NONE
```

### Flight controller

T-Motor F7 F55A ProII

### Other components

TBS M8-2 GPS attached to UART4

### How are the different components wired up

TBS M8-2 - GND RX TX VCC
TmotorF7 - GND  R4 T4 +5V

### Add any other context about the problem that you think might be relevant here

also tried Development 4.4.0=zulu [pre-release]

haslinghuis, 2022-12-19:
1. Please include the build log for the firmware.
2. Did you add `GPS` as build option?
3. When enabling `GPS` on ports tab it should auto-enable the `feature`.

hprompt, 2022-12-19:
thanks for the reply
where would I get the build log for the firmware?
I didn't think about adding gps as build option, I will try and report back
thanks

hprompt, 2022-12-19:
added GPS as build option during setup -- fixed issue
thanks for all the helps
will close

closed 2022-12-19 by hprompt