betaflight/betaflight #12070: switch OSD profile with modes chanel

L1cardo, 2022-12-21:
### Is your feature request related to a problem? Please describe

Since there is an option in BF 4.4 that wo can show stick overlay on osd, so it would be nice to have a mode channel to cycle between OSD profiles. 
If I am flying my own, I won't let the stick shows on my screen. If I want to record a freestyle tutorial or just want to show my stick movements to my friends, then I can stich to OSD profile2 to display stick overlay.
[image]
[image]
[image]

### Describe the solution you'd like

switch OSD profile with modes chanel

### Describe alternatives you've considered

-

### Other information

-

goshangFPV, 2022-12-22:
@L1cardo You can do it by setting free aux-chanel to switch osd profile in adjustments tab.

muchzill4, 2022-12-22:
@L1cardo https://www.youtube.com/watch?v=eYc7djmOzs0 is a nice tutorial on how to do it. 👍

closed 2022-12-23 by haslinghuis