betaflight/betaflight #12079: Barometer not recognized

audiogek, 2022-12-23:
### Describe the bug

BF4.4.0 rc1  on MatekF405CTR the barometer(BMP280) is not recognised/working.
When I try to enable it and safe/reboot it doesn't stick and is disabled again.

### To Reproduce

Flashing to BF4.4.0rc1 reproduces it.

### Expected behavior

Add BMP280 to BF4.4.0

### Cloud build log

```txt
Too late, already back to 4.3x
```

### Diff configuration

```txt
Too late, already back to 4.3x
```

### Resources configuration

```txt
Too late, already back to 4.3x
```

### Flight controller

MatekF405ctr

haslinghuis, 2022-12-23:
Please don't forget to pick the options for your hardware like adding `BARO` to `Other Options` before flashing.

audiogek, 2022-12-23:
Oh wauw, totally missed that those 4 lines actually are dropdown menus. It's even says it when hovering over the "?" 
Thanks and sorry for my mistake.
But a small visual indicator would still be nice. 
This issue can be solved/closed/deleted now.
Thanks again!

closed 2022-12-23 by haslinghuis