betaflight/betaflight #12080: Mamba F405US - Configurator won't save GPS = On. (BF 4.4.0)

AlanzFPV123, 2022-12-23:
### Describe the bug

Started with quad on 4.2.11 with GPS working. 
1)  Flashed 4.4.0 using Configurator 10.9.0 RC2  in classic mode.  
On Ports tab, I set Uart 4 to GPS (same as it was on 4.2.11). Save and reboot successful.
On config tab, I selected GPS on, auto config, etc. Upon Save and reboot, GPS is still off. Repeated multiple times. 
2) Reflashed using cloud build, selected GPS feature. Same behaviour. GPS was off upon connection. Switching on in Config tab did not work.
3) Used CLI command 'Feature GPS' and GPS is now on and working. 

This appears to be a configurator issue. 
I suspect this is board specific because I updated a different quad yesterday to 4.4 with no such issues (Speedybee stack). 

Other notes:  I had tried BF 4.3.xx on this same quad a while back and had a similar issue with the 10.8 configurator. Finally ended up going back to 4.2.11. At that time I didn't know how to use the feature GPS command.

### To Reproduce

To reproduce I think one would need this exact board. 

### Expected behavior

I would expect the GPS "On" switch followed by Save and Reboot to result in the GPS feature to be on.

### Cloud build log

```txt
2022-12-23 @06:15:08 -- Failed to open serial port

2022-12-23 @06:15:10 -- Serial port successfully opened with ID: 24

2022-12-23 @06:15:10 -- MultiWii API version: 1.45.0

2022-12-23 @06:15:10 -- Flight controller info, identifier: BTFL, version: 4.4.0

2022-12-23 @06:15:10 -- Running firmware released on: Dec 19 2022 15:08:37

2022-12-23 @06:15:10 -- Board: DIAT/MAMBAF405US(STM32F405), version: 0

2022-12-23 @06:15:10 -- Unique device ID: 0x3400543252500120383856

2022-12-23 @06:15:10 -- Craft name: RomaF5 WTF

2022-12-23 @06:15:10 -- Arming Disabled

2022-12-23 @06:15:13 -- CLI mode detected

2022-12-23 @06:34:26 -- Serial port successfully closed

2022-12-23 @06:34:26 -- Loaded builds information for https://build.betaflight.com/api/targets from build server.

2022-12-23 @06:34:26 -- Using cached builds information for https://build.betaflight.com/api/targets.

2022-12-23 @06:34:27 -- Loaded builds information for https://build.betaflight.com/api/targets from build server.

2022-12-23 @06:34:27 -- Loaded builds information for https://build.betaflight.com/api/options from build server.

2022-12-23 @06:34:27 -- Using cached builds information for https://build.betaflight.com/api/options.

2022-12-23 @06:34:27 -- Using cached builds information for https://build.betaflight.com/api/targets.

2022-12-23 @06:34:27 -- Loaded builds information for https://build.betaflight.com/api/targets/MAMBAF405US from build server.

2022-12-23 @06:34:27 -- Loaded builds information for https://build.betaflight.com/api/builds/4.4.0-RC1/MAMBAF405US from build server.

2022-12-23 @06:34:27 -- Loaded builds information for https://build.betaflight.com/api/targets/MAMBAF405US from build server.

2022-12-23 @06:34:27 -- Using cached builds information for https://build.betaflight.com/api/builds/4.4.0-RC1/MAMBAF405US.

2022-12-23 @06:34:28 -- Query board information to preselect right firmware

2022-12-23 @06:34:28 -- Failed to open serial port
[... 530 lines cut ...]
resource ADC_BATT 1 C01
resource ADC_RSSI 1 C02
resource ADC_CURR 1 C03
resource ADC_EXT 1 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 A15
resource OSD_CS 1 B12
resource GYRO_EXTI 1 C04
resource GYRO_EXTI 2 NONE
resource GYRO_CS 1 A04
resource GYRO_CS 2 NONE
resource USB_DETECT 1 NONE
```

### Flight controller

Diatone Mamba F405US

### How are the different components wired up

GPS wired to UART 4. Was working on 4.2.11, no hardware changes, just flashed to 4.4

### Add any other context about the problem that you think might be relevant here

I think this issue dates back to the 10.8 configurator, because I experienced the same thing when updating to 4.3. Note I did  not know the 'feature GPS' command at that time, so I gave up and went back to 4..2.11 and all worked well until I wanted to upgrade to WTF-OSD, hence the need for  BF 4.4.

haslinghuis, 2022-12-23:
With Betaflight configurator 10.9-RC2 you should only have to enable GPS on the ports tab which now automatically enables the feature in configuration tab.

Good to know enabling `feature gps` in CLI is a workaround.

As you already suspect the issue is board specific please mention the GPS module vendor and model.

AlanzFPV123, 2022-12-23:
GPS module is a TBS M8.2 GPS Glonass

AlanzFPV123, 2022-12-23:
Sorry! GPS is actually a Matek GNSS and Compass M8Q-5883 (GNSS SAM-M8Q) and (compass QMC5883L)

haslinghuis, 2022-12-23:
Have ordered Matek GNNS & Compass M110-5883 and waiting for next build so I can test more.

AlanzFPV123, 2022-12-24:
Wow that’s great! It’s a good module and has a power filter, which is why I’m using on this FC which has a dirty 5v supply.  Just remaidened that quad and she flies so good. GPS working perfect.

LoneStaruUAV, 2023-04-10:
Experiencing a similar Issue with a SpeedyBee F7 Mini V2 board, GPS is on but no GPS icon and no GPS data in the OSD

closed 2026-09-26 by blckmn