betaflight/betaflight #12082: AIRBOTF4/REVO - no blackbox on 4.4

ABLomas, 2022-12-23:
### Describe the bug

AIRBOT F4 board (aka FLIP32 F4), with AIRBOTF4 or REVO target do not see onboard blackbox chip (SPI), when using 4.4 firmware ("FLASH" is selected on configurator when flashing FW).
Onboard flash works fine with stable (4.3.2) firmware.

### To Reproduce

Just flash 4.3.2 version, 4.4 and compare

### Expected behavior

onboard flash working in new release

### Diff configuration

```txt
set blackbox_device = SPIFLASH
```

### Resources configuration

```txt
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: LED 2
B05: LED 1
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: FREE
B13: FREE
B14: FREE
B15: FREE
C00: INVERTER 1
C01: FREE
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
    CH2 : DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

AIRBOT F4

haslinghuis, 2022-12-23:
As we have add the define for the blackbox to the target please try adding `FLASH_M25P16` to Custom Defines while flashing and report back.

[image]

Otherwise please try with one of these - one at the time:

    FLASH_W25P16
    FLASH_W25M512
    FLASH_W25Q128FV
    FLASH_W25N01G
    FLASH_W25M02G

haslinghuis, 2022-12-23:
Please try adding `FLASH_M25P16` to Custom Defines while flashing and report back so we can add the define for the blackbox chip used in the target config.

[image]

Otherwise please try with one of these - one at the time:

    FLASH_W25P16
    FLASH_W25M512
    FLASH_W25Q128FV
    FLASH_W25N01G
    FLASH_W25M02G

ABLomas, 2022-12-23:
Thanks, with FLASH_M25P16 it works!
So looks like it is easy fix - add missing define to defaults?
(should i close bug now, or keep it for a while until changes will be done for target?)

haslinghuis, 2022-12-23:
It will automatically close after the fix is merged.

Is it this board?

    # Betaflight / STM32F405 (S405) 4.1.0 Oct 16 2019 / 11:57:16 (c37a7c91a) MSP API: 1.42

    board_name AIRBOTF4
    manufacturer_id AIRB

or this one?

    # Betaflight / STM32F405 (S405) 4.1.0 Oct 16 2019 / 11:57:16 (c37a7c91a) MSP API: 1.42

    board_name REVO
    manufacturer_id OPEN

ABLomas, 2022-12-23:
First one.
BUT!
I don't think those two differs at all, at least visually, i have both and they are identical, visually (at least not from quick look).
Both firmwares can be flashed and AFAIK only difference between them - different pads for LED and buzzer.
And IIRC both have the same SPI flash so guess "fix" needed for both.

KarateBrot, 2022-12-25:
Now it would work without custom defines. Maybe it needs some time for the server to update.

closed 2022-12-25 by KarateBrot