betaflight/betaflight #12084: Centered post flight stats with HD-mode

Box-in-the-Jack, 2022-12-24:
### Is your feature request related to a problem? Please describe

After flying and disarm post flight stats are shown. With analog and digital links before HD mode the stats were more or less centered. Now with HD mode the stats are all the way to the left.

### Describe the solution you'd like

1.) I'd like to have an option to place the stats wherever the user wants. 
2.) Just put it in the middle, all the time.
3.) Use the whole screen.

I guess whatever has the chance to get done for bf 4.4.1 (or even 4.4.0? Please, pretty pretty please!) should be done first as the current behavior is just... weird.

### Describe alternatives you've considered

There is no alternative... is's either HD or not. Not using HD is an option, but I guess that's not the idea of an alternative.

McGiverGim, 2022-12-24:
I'm not too sure if this is a feature request or can be considered a bug... @SteveCEvans you did almost all the HD part. What do you think?

Box-in-the-Jack, 2022-12-24:
Bug or feature request, I guess that depends on the way one looks at it. Someone told me the stats placement is hardcoded in betaflight, which is fine when basically analog SD was the only kid in town. And I guess it still works as intended, no "bug" there. The canvas to "paint" the stats just changed. Then it's a feature request.
But when you just see the situation as is, ignoring the past, it might be seen as a bug.

McGiverGim, 2022-12-24:
The statistics are written here:
https://github.com/betaflight/betaflight/blob/7b70abd118a7b72e699bd097c0cea479a3f59f30/src/main/osd/osd.c#L686-L691
Is only add an offset to the second parameter of the displayWrite commands to be able to center it.
Something like this:
```c
    uint8_t offset = 0;
    if (vcdProfile->video_system == VIDEO_SYSTEM_HD && vcdProfile->canvas_cols > OSD_ELEMENT_BUFFER_LENGTH) {
        offset = (vcdProfile->canvas_rows - OSD_ELEMENT_BUFFER_LENGTH) / 2;
    }
    displayWrite(osdDisplayPort, offset + 2, y, DISPLAYPORT_ATTR_NORMAL, text);
    displayWrite(osdDisplayPort, offset + 20, y, DISPLAYPORT_ATTR_NORMAL, ":");
    displayWrite(osdDisplayPort, offset + 22, y, DISPLAYPORT_ATTR_NORMAL, value);
```
But I'm not too sure where to calculate this offset. Doing this here (in this method) will calculate it for each statistic, I suppose it must be done earlier to calculate it only one.
I'm not good at C, maybe someone can pass here and take a look at it.

closed 2022-12-26 by blckmn