betaflight/betaflight #12095: Cloud build system doesn't allow specification of EXST settings or different memory layouts.

hydra, 2022-12-26:
### Describe the bug

The EXST system requires a base address to be specified and this is per-bootloader and per-flash chip specific yet there is no way to specify this from a 'unified' target `.config`. Similiarly targets also need different memory layouts (see #10702).  Both are required.

Specifically the settings are:
a) the linker script (`LD_SCRIPT`)
b) the EXST VMA address (`EXST_ADJUST_VMA`).

Examples of shipping production boards that need this:
1) SPRacingH7EXTREME PX4 Edition - uses a different flash chip with a different VMA address, see https://github.com/spracing/betaflight/commits/spracingh7extreme-px4-edition-4.3.0-RC7 where the branch contains additional code to allow building the H7EXTREME target but for the different flash chip).
2) SPRacingH7RF (uses different linker script when build with SPRacingPixelOSD, uses different VMA address).
3) SPRacingH7CINE/NP (in #10702).

Bottom line is the cloud build system needs support for compile time options other than `#define`.

### To Reproduce

See description.

### Expected behavior

Build system should allow things other than `#define` for base target configuration, and in unified target configs.

### Diff configuration

```txt
N/A
```

### Resources configuration

```txt
N/A
```

### Flight controller

N/A

### Other components

N/A

blckmn, 2022-12-31:
At this point the focus will be on getting this working for the majority of targets, and introducing support for alternative MCUs.

Given the limited need this will be on the back burner there is some time. These targets can have hexes built and distributed separately to the configurator. Users will be able to flash using local flashing.

hydra, 2023-01-09:
Is there a repo for the cloud build server scripts somewhere?  I didn't spot it.  I could perhaps have a look and submit some PRs to help with this issue.

hydra, 2023-01-29:
ping @blckmn

blckmn, 2023-01-29:
> Is there a repo for the cloud build server scripts somewhere? I didn't spot it. I could perhaps have a look and submit some PRs to help with this issue.

No there is not currently a repo that is publicly available. This is closed source for the moment until we decide how it will operate.

In relation to your particular issue, we are focused on the ARTERY MCU support for the moment, but will consider the support for your particular target in the future. We are unlikely to give your target immediate priority as it is limited in numbers, and there are alternatives, i.e. you can build and distribute your own hex files at any stage.

We will however look into it in time, especially if the popularity among manufacturers of your particular design picks up.

closed 2023-03-08 by github-actions[bot]