betaflight/betaflight #12114: GPS not recognized after some time without power

Strat00s, 2022-12-30:
### Describe the bug

I have some problems with my GPS module (BN-220) and I have no idea what it can be. When I installed iNav on my quad, everything worked as expected. But because iNav does not support RTH without a compass, I wanted to switch to Betaflight until I get a compass. So I flashed Betaflight and configured my quad. Everything seemed to work just fine (mainly the GPS). Went flying and I just couldn't get any satellites. When I came home and connected my quad to my PC, it suddenly did not show the GPS symbol (GPS was not recognized), but it was still enabled.
So I flashed iNav back to see if that would work. And sure enough, it did. So next I again flashed Betaflight and GPS was working just fine (again). I then decided to wait a few hours to see if the same problem would happen. And yes, it did. The drone just sat on my table for 9 hours and after connecting it again to my PC, the GPS is once again not recognized. And once again, flashing iNav fixes it (but for how long?).

I tried it a few times now, always with the same result. And I should add that I never touched the GPS. It was always connected to the same UART 3. The only change is the FW, nothing else.
This issue is also not happening in iNav. The GPS is always detected there.

### To Reproduce

1. Get BN-220, flash iNav, and enable GPS (57600 baud and UBLOX)
2. Flash Betaflight and enable GPS (and set it to UBLOX). GPS works as it should.
3. Power off the quad/FC and wait for a few hours.
4. Power on the quad/FC and GPS will no longer be recognized (but flashing iNav immediately makes it work again).

### Expected behavior

GPS should always be detected.

### Diff configuration

```txt
###WARNING: NO CUSTOM DEFAULTS FOUND###

# version
# Betaflight / STM32F405 (S405) 4.3.0 Jun 14 2022 / 00:47:24 (229ac66) MSP API: 1.44
###ERROR IN diff: NO CONFIG FOUND###
# start the command batch
batch start

board_name JHEF405
manufacturer_id JHEM

# name: Flywoo Explorer

# resources
resource BEEPER 1 C13
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 B05
resource MOTOR 6 B07
resource MOTOR 7 C09
resource MOTOR 8 C08
resource PPM 1 B08
resource LED_STRIP 1 A09
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 D06
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 C14
resource SPI_SCK 1 A05
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B08
resource ADC_BATT 1 C03
resource ADC_RSSI 1 C00
resource ADC_CURR 1 C02
resource FLASH_CS 1 B03
resource OSD_CS 1 B14
resource GYRO_EXTI 1 B13
resource GYRO_CS 1 B12
resource USB_DETECT 1 A08

# timer
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer A03 AF1
# pin A03: TIM2 CH4 (AF1)
timer A02 AF1
# pin A02: TIM2 CH3 (AF1)
timer B05 AF2
# pin B05: TIM3 CH2 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer A09 AF1
# pin A09: TIM1 CH2 (AF1)

# dma
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin A03 1
# pin A03: DMA1 Stream 6 Channel 3
dma pin A02 0
# pin A02: DMA1 Stream 1 Channel 3
dma pin B05 0
# pin B05: DMA1 Stream 5 Channel 5
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin C08 0
# pin C08: DMA2 Stream 2 Channel 0
dma pin A09 0
# pin A09: DMA2 Stream 6 Channel 0

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL
feature MOTOR_STOP
feature GPS
feature LED_STRIP
feature OSD

# serial
serial 0 64 115200 57600 0 115200
serial 2 2 115200 57600 0 115200

# led
led 0 4,6::CB:8
led 1 5,6::CB:8
led 2 6,6::CB:8
led 3 7,6::CB:8

# master
set gyro_lpf1_static_hz = 137
set gyro_lpf2_static_hz = 275
set gyro_lpf1_dyn_min_hz = 137
set gyro_lpf1_dyn_max_hz = 275
set acc_calibration = -49,-26,-21,1
set mag_bustype = I2C
set mag_i2c_device = 1
set mag_hardware = NONE
set baro_bustype = I2C
set baro_i2c_device = 1
set fpv_mix_degrees = 10
set serialrx_provider = SBUS
set blackbox_device = SPIFLASH
set dshot_burst = ON
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set battery_meter = ADC
set ibata_scale = 170
set beeper_inversion = ON
set beeper_od = OFF
set gps_provider = UBLOX
set simplified_gyro_filter_multiplier = 55
set osd_vbat_pos = 14571
set osd_ah_sbar_pos = 14542
set osd_ah_pos = 14414
set osd_current_pos = 14634
set system_hse_mhz = 8
set max7456_spi_bus = 3
set dashboard_i2c_bus = 1
set pinio_box = 40,41,255,255
set flash_spi_bus = 3
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_sensor_align = CW90
set gyro_1_align_yaw = 900
set gyro_2_spibus = 1
set name = Flywoo Explorer

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 60
set dterm_lpf1_dyn_max_hz = 120
set dterm_lpf1_static_hz = 60
set dterm_lpf2_static_hz = 120
set p_pitch = 24
set i_pitch = 22
set d_pitch = 29
set p_roll = 23
set i_roll = 22
set d_roll = 28
set d_min_roll = 17
set d_min_pitch = 17
set simplified_pids_mode = OFF
set simplified_dterm_filter_multiplier = 80

rateprofile 0

# end the command batch
batch end

#
```

### Resources configuration

```txt
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: LED_STRIP
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FLASH_CS
B04: FREE
B05: FREE
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: GYRO_CS 1
B13: GYRO_EXTI
B14: OSD_CS
B15: FREE
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: LED 1
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH2 : LED_STRIP
TIM2:
    CH3 : MOTOR 4
    CH4 : MOTOR 3
TIM3:
    CH3 : MOTOR 1
    CH4 : MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: FREE

#
```

### Flight controller

JHEMCU GF16-BMI (JHEF405), https://www.aliexpress.com/item/1005003725857398.html, https://github.com/jhemcu/FC-ESC-Firmware/tree/main/GF16-BMI%2B13A%20ESC

### Other components

BN-220
HappyModel EP1, ELRS 3.1.2

### How are the different components wired up

Receiver on UART 1, GPS on UART 3.

### Add any other context about the problem that you think might be relevant here

Reflashing Betaflight does nothing. The GPS is not detected until iNav is flashed and GPS is configured there.
I might have had a similar problem (bug) with Ardupilot and iNav, where Ardupilot configured the GPS to use baudrate which iNav did not support. The GPS would not work in iNav at all but worked just fine in Ardupilot. Maybe the GPS forgets its setting after a "long" time without power and Betaflight can't communicate with it afterward?

KarateBrot, 2022-12-30:
I cannot reproduce this on my BN-220. Everything works just fine for me...

But what about this warning?
```
###WARNING: NO CUSTOM DEFAULTS FOUND###
```

See if flashing to BF 4.4 and applying custom defaults works. If you want to use BF for RTH you definitely need BF 4.4, not 4.3.

Strat00s, 2022-12-31:
> I cannot reproduce this on my BN-220. Everything works just fine for me...
> 
> But what about this warning?
> 
> ```
> ###WARNING: NO CUSTOM DEFAULTS FOUND###
> ```
> 
> See if flashing to BF 4.4 and **applying custom defaults** works. If you want to use BF for RTH you definitely need BF 4.4, not 4.3.

Tried it. Still the same. The GPS is not recognized unless I first configure it with iNav and I do not disconnect the power before flashing Betaflight.

How old is your GPS? Does the battery on the module still hold charge? Because I think that's the issue with mine. It is pretty old and it sat for a few years not doing anything. The battery is dead so the GPS does not remember any configuration. And I guess in the default configuration (or whatever is loaded then), Betaflight can't communicate with it.

And it seems to be the case because just disconnecting the GPS makes it stop working (but doing the same in iNav, it works just fine).

pitts-mo, 2022-12-31:
I remember the days when GPS needed to be configured... is it possible the stored settings on  @Strat00s  's.  specific  module can be recofigured correctly by inav but betaflight isnt catching the same miss set defaullt?

Strat00s, 2023-01-09:
So guys, any ideas about what can one do?

pitts-mo, 2023-01-10:
Does anyone know the optimal defaults that should be set on a u-blox GPS for betaflight?

@Strat00s 
**Warning: This info may help you learn more about what could be wrong. However, adjusting any GPS firmware or settings could also make the GPS completely unusable....**

So ~8-10 years ago we had to connect the GPS to a computer with an FTDI adapter to run manufacture specific software in order to configure the GPS to boot utilizing the settings our FCs needed... I believe the tool for the BN-220 would be u-center (https://www.u-blox.com/en/product/u-center) 

I could not quickly find resources for configuring a GPS for betaflight. However, I found these videos I thought would be worth a watch to get an understanding of what is under the hood... **Do not modify your GPS without accepting the risk you may make it unusable.**

firmware update? : https://www.youtube.com/watch?v=lqZ1wTd9gKU
tinkering with betaflight and u-center? : https://www.youtube.com/watch?v=qkhTBxFPtbE

good luck, 
-p

pitts-mo, 2023-01-10:
closing out some searches... figured this may help
https://dadfpv.com/how-to-configure-optimal-gps-settings-increase-update-rate-sam-m8q-m8q-5883-and-similar-ublox-based/

Strat00s, 2023-01-11:
@pitts-mo  It seems to be the same bug (or "not a feature") I had on iNav a few years back https://github.com/iNavFlight/inav/issues/6159. I used this GPS in Ardupilot and it seems to configure the GPS and store the config in flash/EEPROM. So it was configured for a 230400 baud rate the entire time and Betafligh does not consider such baud rate.
Using the u-center and setting it up for 57600 baud and storing it in flash/EEPROM did the trick. Thanks.

closed 2023-01-11 by Strat00s

KarateBrot, 2023-01-11:
Oh I see, thanks for the update.