betaflight/betaflight #12115: Add names to the UART Port configuration page

AlexandreBonneau, 2022-12-30:
### Is your feature request related to a problem? Please describe

When building multiple quads, each with different configuration, it can be daunting to remember which UART was used and for what.

### Describe the solution you'd like

I would be great if we could have a small text field next to each detected UART in the Port configuration page, when you could type a small information.

For instance you could name all your UART like this:
```
set uart1_name = "VTX HDZero"
set uart2_name = "ELRS"
set uart3_name = none
set uart4_name = none
set uart5_name = "Bluetooth"
set uart6_name = "GPS"
set uart7_name = none
```

That way, even if you haven't touched a specific quad for a long time, you can quickly know which UART are in use, and for which function

### Describe alternatives you've considered

The alternative, is to wirte this information down in a text file, make sure it's up-to-date, and make sure you can find it back later on.

pitts-mo, 2022-12-31:
Making presets and labeling them for each craft (or craft type) is the way to go.
-p

More thoughts:
The feature requested appears redundant as the user would still need to manually configure/label all the ports and there would still be a need to save the settings...  both are things that are already done.... you should also be able to diff all and selectively save the settings you wish to persist for a craft as a partial diff or even a preset (already supported)... 

This issue has prompted an interesting thought though..... what about allowing configurator to store FC MCU serial numbers and flag all the presets you want to apply?...
This would require a lot of configurator changes and still a bunch of manual work ....

bigspeedfpv, 2022-12-31:
another easier idea could just be comments for each serial number, so it doesn't have to actually change anything - just a little textbox for each different FC (could be useful for tracking gyro issues etc between identical setups)

AlexandreBonneau, 2023-01-31:
I disagree.
Making presets is hugely more complicated than writing some text in a field.
Also, having presets means you somehow have to manage versioning whenever you change a single part of your quad.

I fail to see how this is easier to manage than just inputing a small text field like :
```
VTX: HDZero Race v2
FC: iFlight mini f722
ELRS: BetaFPV micro 2.4Ghz
Misc: Added a 12V BEC to power up the VTX, since VBAT was too much
Misc: Whatch-out; the UART3 is out of order; DO NOT use it
```
...and changing a small part of this text whenever you physically modify your quad.

Real life example; I built quite a lot of quads lately, and with some small variations for the same frames (due to components shortage).
Whenever I need to update ELRS, I need to disassemble the quads to check which ELRS receiver was installed in those quads.
I really hassle...which would be easily solved with my feature requests since I would just have to plug the quad, check the 'Quad information' field and voilà. No need to take it apart.

This is only one of the many example where having the ability to add 'comments' to your quad could be helpful.

ggoraa, 2023-02-01:
I agree with @AlexandreBonneau, these features could be really useful

AlexandreBonneau, 2023-03-03:
This feature request is not stale and still valid.

AlexandreBonneau, 2023-04-04:
This is still not stale, it's just under consideration :>

AlexandreBonneau, 2023-10-26:
[image]

This is what that could look like in Betaflight configurator. Thoughts?

haslinghuis, 2023-10-26:
We gonna want a lot more info like that- like logs, maintenance, checks, why you changed X or Y, etc.
Cloud based storage or manage your own backup. Also upload blackbox for analysis etc.

Maybe not the best example: https://www.youtube.com/watch?v=shg6Q5ubhpw

AlexandreBonneau, 2023-11-21:
It's ok to want to add more info, but for this particular feature request, maybe we can focus on this basic need first?
Cloud storage seems overkill, and will not work if you do not have a reliable internet connection while testing in the field.
Maybe I did not understand what you meant be that though.

The mock-up seems to suffice for this particular feature request.

blckmn, 2026-09-26:
We're closing this as part of a clean-up of issues more than two years old. Sorry it sat here so long without a resolution, that's on us. Betaflight has changed a lot since this was opened, so if it's still relevant on the latest release, please open a new issue with current details and link back here.

closed 2026-09-26 by blckmn

AlexandreBonneau, 2026-09-27:
Well, I'd just clone this issue then, because it's still _very_ relevant! (the more time passes, the more drones you build, so the more config and quirks you need to track...)
@blckmn Wouldn't it be easier to just re-open this issue, since the need and described solution has not changed (and just need a willing implementation)?