betaflight/betaflight #12119: Sanity Check issue on BF4.4 RC2

macgyverek, 2022-12-31:
### Describe the bug

I have problem with sanity check on BF 4.4. I tested at home and Failsafe doesn't disarm a motors.  I tested Failsafe with Drop and working - immediately disable a motors. 
Next I enabled GPS Rescue and I hear that BF set failsafe but motors still working. I wait >10s and not effects.

In my opinion flying drone at home, failsafe should cut engine immediately. Sensity check function I set to ON (always).

### To Reproduce

1. Arm a quad at home (no GPS Fix);
2. Power Off Radio (Express LRS link)
3. Failsafe started
4. Motors not stoped
5. After 15 seconds still working
6. Power On radio
7. Disarm switch 
8. Motor stoped

### Expected behavior

I expected motors  immediately stoped when flying without GPS fix or indoors.

### Diff configuration

```txt
# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Dec 27 2022 / 09:28:44 (281bfb4bf) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name MAMBAF722_I2C
manufacturer_id DIAT

# name: GEPRC Cinequeen

# feature
feature GPS
feature LED_STRIP
feature ESC_SENSOR

# serial
serial 1 2 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# beacon
beacon RX_SET

# led
led 0 8,15::ATIW:0
led 1 7,15::ATIW:0

# mode_color
mode_color 6 0 2
mode_color 7 0 8

# aux
aux 0 0 0 1600 2100 0 0
aux 1 1 1 1425 1575 0 0
aux 2 2 1 1200 1350 0 0
aux 3 46 1 1675 2100 0 0
aux 4 13 3 1900 2100 0 0
aux 5 15 3 1300 2100 0 0
aux 6 39 2 900 1100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BAND_A   A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BAND_B   B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BAND_E   E CUSTOM  5705 5685 5665 5665 5885 5905 5905 5905
vtxtable band 4 AIRWAVE  F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 14 23 27 29
vtxtable powerlabels 25 200 500 800

# vtx
vtx 1 2 0 0 1 1200 1350
vtx 2 2 0 0 2 1400 1500
vtx 3 2 0 0 3 1550 1600
vtx 4 2 0 0 4 1700 2050

# rxfail
rxfail 3 s 1100
rxfail 5 s 1500

# master
set gyro_lpf1_static_hz = 300
set gyro_lpf2_static_hz = 600
set dyn_notch_count = 1
set gyro_lpf1_dyn_min_hz = 300
set gyro_lpf1_dyn_max_hz = 600
set acc_calibration = -96,21,18,1
set mag_hardware = NONE
set rc_smoothing_auto_factor = 40
set serialrx_provider = CRSF
set dshot_idle_value = 600
set dshot_bidir = ON
set motor_poles = 12
set failsafe_throttle_low_delay = 50
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 750
set vbat_warning_cell_voltage = 360
set vbat_scale = 112
set ibata_scale = 140
set beeper_dshot_beacon_tone = 3
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_min_start_dist = 25
set gps_rescue_alt_mode = CURRENT_ALT
set gps_rescue_return_alt = 20
set gps_rescue_pitch_angle_max = 32
set gps_rescue_descent_dist = 30
set gps_rescue_throttle_hover = 1300
set gps_rescue_sanity_checks = RESCUE_SANITY_ON
set gps_rescue_allow_arming_without_fix = ON
set simplified_gyro_filter_multiplier = 120
set report_cell_voltage = ON
set osd_warn_bitmask = 122879
set osd_cap_alarm = 600
set osd_tim1 = 2561
set osd_link_quality_pos = 2305
set osd_rssi_dbm_pos = 2273
set osd_tim_1_pos = 2519
set osd_vtx_channel_pos = 2497
set osd_gps_speed_pos = 2136
set osd_gps_lon_pos = 2081
set osd_gps_lat_pos = 2113
set osd_gps_sats_pos = 2104
set osd_home_dir_pos = 2127
set osd_home_dist_pos = 2295
set osd_compass_bar_pos = 2187
set osd_altitude_pos = 2327
set osd_warnings_pos = 14697
set osd_avg_cell_voltage_pos = 2508
set osd_stat_bitmask = 8783930
set vtx_band = 5
set vtx_channel = 2
set vtx_power = 1
set vtx_freq = 5695
set vcd_video_system = PAL
set gyro_1_align_yaw = 1800
set craft_name = GEPRC Cinequeen
set pilot_name = Adam Majer

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 82
set dterm_lpf1_dyn_max_hz = 165
set dterm_lpf1_static_hz = 82
set dterm_lpf2_static_hz = 165
set p_pitch = 68
set i_pitch = 109
set d_pitch = 48
set f_pitch = 150
set p_roll = 62
set i_roll = 99
set d_roll = 42
set f_roll = 138
set p_yaw = 62
set i_yaw = 99
set f_yaw = 138
set d_min_roll = 32
set d_min_pitch = 37
set simplified_master_multiplier = 115
set simplified_i_gain = 90
set simplified_d_gain = 95
set simplified_pi_gain = 120
set simplified_dmax_gain = 90
set simplified_pitch_pi_gain = 105
set simplified_dterm_filter_multiplier = 110
set tpa_mode = PD

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 75
set pitch_rc_rate = 75
set yaw_rc_rate = 75
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

# end the command batch
batch end
```

### Resources configuration

```txt
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: LED_STRIP
B04: FREE
B05: SPI_MOSI 3
B06: SERIAL_TX 1
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: MOTOR 1
C09: MOTOR 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2 : LED_STRIP
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 3
    CH4 : DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: LED_STRIP
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 1
```

### Flight controller

Diatone MAMBAF722_I2C

### Other components

GPS Matek

haslinghuis, 2022-12-31:
Failsafe does work but this setting tries to bring the quad home with in your case a throttle value of 1300:

    set failsafe_procedure = GPS-RESCUE

As it's allowed to arm without GPS sats we don't have a home position.

macgyverek, 2022-12-31:
In my opinion Senity On should detect problem - GPS Rescue + no sat + no home point and cut a motors! 

Its dangerous if you have GSP Rescue enable and flying in house (for example making cinematic shoots). 

Look there: https://github.com/betaflight/betaflight/wiki/GPS-Rescue-for-4.4 about senity:

SANITY_ON, or true failsafe in SANITY_FS_ONLY -> No Home Fix on initiation -> Immediate disarm. 

Its not working!!!

haslinghuis, 2022-12-31:
@macgyverek please test :point_up:

macgyverek, 2022-12-31:
Tomorrow I will let you know. Thx.

macgyverek, 2023-01-01:
I tested. Still FS didn't stop motors.

haslinghuis, 2023-01-01:
Please test again. Workbench tested.

macgyverek, 2023-01-02:
Yes, your fix solved this issue.

ctzsnooze, 2023-01-02:
It should not be possible to arm when there is no Home point and GPS Rescue is set, unless the user has bypassed all sanity checks or prevented the arming check.

ctzsnooze, 2023-01-02:
It may be safer to enter landing mode in the above situation, provided that Baro is available, since that will not destroy the quad by dropping it hard.

macgyverek, 2023-01-04:
> It should not be possible to arm when there is no Home point and GPS Rescue is set, unless the user has bypassed all sanity checks or prevented the arming check.

But sometimes users want flying without GPS Rescue because cruising on park with trees 2m above ground or inside buildings. Failsafe must working (drop or land).  

This is a good documentation of idea: https://github.com/betaflight/betaflight/wiki/GPS-Rescue-Mode-for-4.1-to-4.3#arm-without-a-gps-fix

ARM without GPS fix: no GPS Rescue
ARM with GPS fis: GPS Rescue Available

Its very simple and good in my opinion. 

Good option for future will be land when Baro available when Rescue will be activated in two situation:
- Flying without GPS Fix 
- Flying at minimum distance at home (now is drop??)

PR #12120 (2023-01-06): Disarm on fail-safe when `allow arming without fix` is enabled and failsafe procedure is set to GPS-RESCUE
Fixes: #12119

When `Allow arming without fix` is enabled and fail-safe is set to `GPS Rescue`, the craft will not disarm on fail-safe.

Turns out as staying at `RESCUE_INITIALIZE` the return blocked the execution of the sanity checks.

Test parameters:

    set failsafe_procedure = GPS-RESCUE
    set gps_rescue_min_start_dist = 25
    set gps_rescue_sanity_checks = RESCUE_SANITY_ON
    set gps_rescue_allow_arming_without_fix = ON

[image]
closed 2023-01-06 by haslinghuis