betaflight/betaflight #12121: FLYWOOF405S_AIO gyro issue

chrsdion, 2022-12-31:
### Describe the bug

The template is missing a gyro define. When I applied 4.4-RC3 to my board i got no dyro.

This is from 4.3 to show the gyro.
[image]

### To Reproduce

Install 4,4-RC3

### Expected behavior

Gyro should be found

### Diff configuration

```txt
# 

# diff

# version
# Betaflight / STM32F405 (S405) 4.4.0 Dec 31 2022 / 23:27:34 (2c34b3949) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name FLYWOOF405S_AIO
manufacturer_id FLWO

profile 0

rateprofile 0

# end the command batch
batch end

#
```

### Resources configuration

```txt
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: LED_STRIP
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: FLASH_CS
B04: FREE
B05: FREE
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: PREINIT 1
B13: FREE
B14: OSD_CS
B15: FREE
C00: ADC_RSSI
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: LED 1
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH2 : LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: FREE

#
```

### Flight controller

FLYWOOF405S_AIO

haslinghuis, 2022-12-31:
When above is merged this should be solved.

For now you can add `GYRO_SPI_ICM42688P ACC_SPI_ICM42688P BARO_DPS310` in Custom Defines field.

haslinghuis, 2023-01-01:
Fix is merged now. Please check again.

closed 2023-01-01 by haslinghuis

chrsdion, 2023-01-01:
Confirmed working

suprchunk, 2023-03-02:
Late to the party, but this doesn't currently work. Using 4.4 final, inputting the gyro in custom defines. No gyro detected, and locked in status.

haslinghuis, 2023-03-02:
@suprchunk Please see: http://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines and post the support ID so we can have a look. Maybe your board has other components (likely BMI270)

_Note that Custom Defines only works when enabling expert mode (it's kind of hidden after show release candidates)._
Or grab latest [configurator](https://github.com/betaflight/betaflight-configurator-nightlies/releases/)

williamleong, 2023-03-05:
@haslinghuis I'm facing the same issue with the same board (BMI270 as you said, probably an older revision). Flashed core and submitted support ID 96e2d4f8-c956-49ca-a4f9-8d7a6e56567b

williamleong, 2023-03-05:
I've also tried adding ACCGYRO_BMI270 to the custom defines in 4.4 but it is still not detected. I've verified that it is a BMI270 by flashing 4.3 and using the "status" command in the CLI.

haslinghuis, 2023-03-05:
Use latest nightly of configurator and try again

williamleong, 2023-03-05:
@haslinghuis Thanks, ACCGYRO_BMI270 in custom defines worked as expected with latest nightly (https://github.com/betaflight/betaflight-configurator-nightlies/releases/tag/v20230302.400).

@suprchunk You might want to give it a try to verify.