betaflight/betaflight #12122: CRAZYBEEF4FR BMI270 gyro missing

joeyw007, 2023-01-01:
### Describe the bug

after flashing bf 4.4.0 rc3 when on configuration page gyro is not detected.

### To Reproduce

flash bf to crazybeef4fr with BMI270 gyro, apply custom defaults, connect, go to configuration page, try to find missing gyro?

### Expected behavior

expect firmware to detect gyro

### Cloud build log

```txt
***** Commencing Build *****

VERSION: 4.4.0-RC3
COMMIT:  4.4.0-RC3
BOARD:   CRAZYBEEF4FR/STM32F411
STARTED: 2023-01-01 03:13:25Z

Want to run this build locally?

git checkout 4.4.0-RC3 && make TARGET=STM32F411 EXTRA_FLAGS="-D'BUILD_KEY=4.4.0-RC3/CRAZYBEEF4FR/0e5d954c9cb122d3503ba09352b14693' -D'BOARD_NAME=CRAZYBEEF4FR' -D'MANUFACTURER_ID=HAMO' -DCLOUD_BUILD -DTARGET_FLAGS -DUSE_ACC -DUSE_ACC_SPI_MPU6000 -DUSE_DSHOT -DUSE_FLASH -DUSE_GYRO -DUSE_GYRO_SPI_MPU6000 -DUSE_LED_STRIP -DUSE_MAX7456 -DUSE_OSD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_SERIALRX_DEFAULT -DUSE_SERIALRX_GHST -DUSE_SERIALRX_SBUS -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_TELEMETRY_GHST -DUSE_VTX"

Docker log: 

Pulling from Github (4.4.0-RC3):
From https://github.com/betaflight/betaflight
 * [new tag]             4.4.0-RC3  -> cloud_build

Checking out the branch -> cloud_build (for 4.4.0-RC3)
Switched to branch 'cloud_build'

Commencing the build (make STM32F411):
make -j ./obj/betaflight_4.4.0_STM32F411.hex
make[1]: Entering directory '/betaflight'
rm -f ./obj/main/STM32F411/.efhash_*
EF HASH -> ./obj/main/STM32F411/.efhash_226197f53d4e99c97f2991d180344c8b
%% startup_stm32f411xe.s 
%% (optimised) ./src/main/startup/system_stm32f4xx.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu.c 
%% (optimised) ./src/main/drivers/adc_stm32f4xx.c 
%% (optimised) ./src/main/drivers/bus_i2c_stm32f4xx.c 
%% (optimised) ./src/main/drivers/bus_spi_stdperiph.c 
%% (optimised) ./src/main/drivers/dma_stm32f4xx.c 
%% (optimised) ./src/main/drivers/dshot_bitbang.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_decode.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_stdperiph.c 
%% (size optimised) ./src/main/drivers/inverter.c 
%% (size optimised) ./src/main/drivers/light_ws2811strip_stdperiph.c 
%% (size optimised) ./src/main/drivers/transponder_ir_io_stdperiph.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_shared.c 
%% (optimised) ./src/main/drivers/serial_uart_stdperiph.c 
%% (optimised) ./src/main/drivers/serial_uart_stm32f4xx.c 
%% (optimised) ./src/main/drivers/system_stm32f4xx.c 
%% (optimised) ./src/main/drivers/timer_stm32f4xx.c 
%% (optimised) ./src/main/drivers/persistent.c 
%% (optimised) ./src/main/drivers/sdio_f4xx.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_fake.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu3050.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_mpu6050.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_mpu6500.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_spi_bmi160.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_spi_bmi270.c 
%% (optimised) ./src/main/drivers/accgyro/accgyro_spi_icm20649.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_spi_icm20689.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_spi_icm426xx.c 
[... 1508 lines cut ...]
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

HappyModel CrazybeeF4FR for Salifly, Larva, Mobula Purchased from RDQ

### Other components

HappyModel OVX303 VTX

### How are the different components wired up

Properly?

joeyw007, 2023-01-01:
Apologies, board is Crazybee F4 Pro v3 FrSky

joeyw007, 2023-01-01:
Can I add BMI270 driver with custom define via cloud build?

haslinghuis, 2023-01-01:
Yes add `ACCGYRO_BMI270`

joeyw007, 2023-01-01:
> Yes add `ACCGYRO_BMI270`
> 
> https://github.com/betaflight/betaflight/wiki/Flashing-firmware

Cloud build with RX_CC2500 RX_SPI ACCGYRO_BMI270 custom defines has toothpick with F4FR BMI270 board cruising around the house. I had issues with control link dropping out every second or so on this board with 4.3.0 - 4.3.2. Will test as soon as weather permits and get back to you tomorrow most likely. Thanks.

joeyw007, 2023-01-01:
Adding cloud build defines "RX_CC2500" "RX_SPI" & ACCGYRO_BMI270 Fixed gyro issue and allowed me to bind. However, in outdoor testing, range and link reliability in D8 mode is a problem. I experienced reduced range, about %10 over 4.2.11, and random failsafes on the crazybee f4 pro v3 FrSky. I have been having this issue since release of 4.3.0 and have just been running 4.3 RC1 because we fixed it once and something got changed. If need to create new PR let me know.

haslinghuis, 2023-01-01:
Yes configuration has been solved.
Control link needs a separate issue as SPI FRSKY does official not even exist according to FrSky

joeyw007, 2023-01-01:
> Yes configuration has been solved. Control link needs a separate issue as SPI FRSKY does official not even exist according to FrSky

OK will do some more testing and create separate issue if I cannot resolve with crazybeeF4FR boards. Crazybee F4 Lite and BetaFPV F4 1s whoop board work fine. The Crazybee X v1 and Crazybee F4 Pro v3, all crazybeeF4FR target, have random failsafe issues in FW 4.3.0 to 4.4.0 RC3. Will BB log help? If so, I'll solder up an openlager and get logs as soon as I have time and drop in a separate issue.

closed 2023-01-02 by blckmn

javierckr, 2023-07-28:
Same problem on CRAZYBEEF4FS, with the custom defines the gyro is detected.

joeyw007, 2023-07-28:
Wrong email address guys. I think this one was for another customer?