betaflight/betaflight #12123: Crux 3 CrazybeeF4FR cannot select SPI RX receiver mode

joeyw007, 2023-01-01:
### Describe the bug

on receiver tab cannot set receiver mode to SPI Rx

### To Reproduce

flash FC with bf 4.4.0 RC3, apply custom defaults, go to receiver tab, select SPI RX, save and reboot. Go back to receiver tab and Serial via UART is still selected in receiver mode drop down.

### Expected behavior

expect to be able to select SPI RX

### Cloud build log

```txt
***** Commencing Build *****

VERSION: 4.4.0-RC3
COMMIT:  4.4.0-RC3
BOARD:   CRAZYBEEF4FR/STM32F411
STARTED: 2023-01-01 03:13:25Z

Want to run this build locally?

git checkout 4.4.0-RC3 && make TARGET=STM32F411 EXTRA_FLAGS="-D'BUILD_KEY=4.4.0-RC3/CRAZYBEEF4FR/0e5d954c9cb122d3503ba09352b14693' -D'BOARD_NAME=CRAZYBEEF4FR' -D'MANUFACTURER_ID=HAMO' -DCLOUD_BUILD -DTARGET_FLAGS -DUSE_ACC -DUSE_ACC_SPI_MPU6000 -DUSE_DSHOT -DUSE_FLASH -DUSE_GYRO -DUSE_GYRO_SPI_MPU6000 -DUSE_LED_STRIP -DUSE_MAX7456 -DUSE_OSD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_SERIALRX_DEFAULT -DUSE_SERIALRX_GHST -DUSE_SERIALRX_SBUS -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_TELEMETRY_GHST -DUSE_VTX"

Docker log: 

Pulling from Github (4.4.0-RC3):
From https://github.com/betaflight/betaflight
 * [new tag]             4.4.0-RC3  -> cloud_build

Checking out the branch -> cloud_build (for 4.4.0-RC3)
Switched to branch 'cloud_build'

Commencing the build (make STM32F411):
make -j ./obj/betaflight_4.4.0_STM32F411.hex
make[1]: Entering directory '/betaflight'
rm -f ./obj/main/STM32F411/.efhash_*
EF HASH -> ./obj/main/STM32F411/.efhash_226197f53d4e99c97f2991d180344c8b
%% startup_stm32f411xe.s 
%% (optimised) ./src/main/startup/system_stm32f4xx.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu.c 
%% (optimised) ./src/main/drivers/adc_stm32f4xx.c 
%% (optimised) ./src/main/drivers/bus_i2c_stm32f4xx.c 
%% (optimised) ./src/main/drivers/bus_spi_stdperiph.c 
%% (optimised) ./src/main/drivers/dma_stm32f4xx.c 
%% (optimised) ./src/main/drivers/dshot_bitbang.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_decode.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_stdperiph.c 
%% (size optimised) ./src/main/drivers/inverter.c 
%% (size optimised) ./src/main/drivers/light_ws2811strip_stdperiph.c 
%% (size optimised) ./src/main/drivers/transponder_ir_io_stdperiph.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_shared.c 
%% (optimised) ./src/main/drivers/serial_uart_stdperiph.c 
%% (optimised) ./src/main/drivers/serial_uart_stm32f4xx.c 
%% (optimised) ./src/main/drivers/system_stm32f4xx.c 
%% (optimised) ./src/main/drivers/timer_stm32f4xx.c 
%% (optimised) ./src/main/drivers/persistent.c 
%% (optimised) ./src/main/drivers/sdio_f4xx.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_fake.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu3050.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_mpu6050.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_mpu6500.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_spi_bmi160.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_spi_bmi270.c 
%% (optimised) ./src/main/drivers/accgyro/accgyro_spi_icm20649.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_spi_icm20689.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_spi_icm426xx.c 
[... 544 lines cut ...]
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

CrazybeeX FrSky Comes in Crux3 BNF Purchased from RDQ

### Other components

All Integrated

haslinghuis, 2023-01-01:
Please check with adding `RX_CC2500` to custom defines

joeyw007, 2023-01-01:
> Please check with adding `RX_CC2500` to custom defines.
> 
> Does the Bind button reveal itself on receiver tab after configuration?
> 
> If this does not work what happens when you also add `RX_SPI` to custom defines?

RX_CC2500 fails. Using Custom fine RX_SPI I can now select SPI RX and FrSky D8 but no bind button and bind_rx not supported in CLI

joeyw007, 2023-01-01:
Adding cloud build defines "RX_CC2500" & "RX_SPI" solved issue with receiver mode. However, in outdoor testing, range and link reliability in D8 mode is a problem. I experienced reduced range, about %10 over 4.2.11, and random failsafes on the crazybee X v1.0 FrSky. I have been having this issue since release of 4.3.0 and have just been running 4.3 RC1 because we fixed it once and something got changed. If need to create new PR let me know.

closed 2023-01-01 by blckmn

joeyw007, 2023-01-02:
Looking at the long list of SPI FrSky issues since 4.3.0 RC1 and knowing the protocol is nearly dead. As well as my own trials and tribulations testing and trying to work out problems. I've opted to solder in nano ELRS receivers before soldering in anything else. I'm done with D8. May it Rest In Peace. "Thank you for your long and good service."

PR #12124 (2023-01-01): Add RX_SPI to RX_CC2500
Fixes: #12123

Targets have been updated: https://github.com/betaflight/unified-targets/pull/767

Build

```
make TARGET=STM32F411 \
EXTRA_FLAGS="-D'BUILD_KEY=4.4.0-RC3/CRAZYBEEF4FR0e5d954c9cb122d3503ba09352b14693' \
-D'BOARD_NAME=CRAZYBEEF4FR' -D'MANUFACTURER_ID=HAMO' -DCLOUD_BUILD -DTARGET_FLAGS \
-DUSE_ACC -DUSE_ACC_SPI_MPU6000 -DUSE_DSHOT -DUSE_FLASH -DUSE_GYRO -DUSE_GYRO_SPI_MPU6000 \
-DUSE_LED_STRIP -DUSE_MAX7456 -DUSE_OSD -DUSE_PINIO -DUSE_SERIALRX  -DUSE_SERIALRX_DEFAULT \
-DUSE_SERIALRX_SBUS -DUSE_TELEMETRY -DUSE_VTX -DUSE_RX_CC2500"
```