betaflight/betaflight #12126: Betaflight 4.4 RC3 - Can't enable Barometer

mcaraballo, 2023-01-01:
### Describe the bug

In Betaflight 4.4 RC3 you cannot turn on the Baro option via the CLI or GUI.  The GUI allows you to turn the baro switch on, the configurator reboots on save, and yet the change is not committed to the FC.  This works just fine in Betaflight 4.3.2 stable for example.

I tested this on just one flight controller so this may be limited to that hardware.  The FC is an iFlight Succex-D F7 - "EXUA/EXF722DUAL(STM32F7X2)"

If I type GET baro on a fresh flash of the Betaflight 4.4 RC3 firmware I get:
# get altitude_prefer_baro
altitude_prefer_baro = 100
Allowed range: 0 - 100

Whereas if I type GET baro on Betaflight 4.3.2 Release I get:
baro_bustype = I2C
Allowed values: NONE, I2C, SPI, SLAVE, GYROAUTO

baro_spi_device = 0
Allowed range: 0 - 5

baro_i2c_device = 2
Allowed range: 0 - 5

baro_i2c_address = 0
Allowed range: 0 - 119

baro_hardware = AUTO
Allowed values: AUTO, NONE, BMP085, MS5611, BMP280, LPS, QMP6988, BMP388, DPS310

baro_tab_size = 21
Allowed range: 1 - 48

baro_noise_lpf = 600
Allowed range: 0 - 1000

baro_cf_vel = 985
Allowed range: 0 - 1000

position_alt_baro_fallback_sats = 7
Allowed range: 3 - 49

### To Reproduce

1. Install the Betaflight 4.4 RC3 firmware fresh onto a iFlight Succex-D F7 flight controller using the Betaflight Configurator 10.9.0
2. Go into the configuration tab
3. Click on the Barometer (if supported) switch
4. Click save and reboot
5. Realize the baro is not turned on

### Expected behavior

When clicking the barometer option in the Betaflight Configurator 10.9.0 it is expected to turn on barometer support and the switch in the UI to show as on.

### Cloud build log

```txt
***** Commencing Build *****

VERSION: 4.4.0-RC3
COMMIT:  4.4.0-RC3
BOARD:   EXF722DUAL/STM32F7X2
STARTED: 2023-01-01 02:12:09Z

Want to run this build locally?

git checkout 4.4.0-RC3 && make TARGET=STM32F7X2 EXTRA_FLAGS="-D'BUILD_KEY=4.4.0-RC3/EXF722DUAL/16135ba8b3f96266f4d69aac23b95b2a' -D'BOARD_NAME=EXF722DUAL' -D'MANUFACTURER_ID=EXUA' -DCLOUD_BUILD -DTARGET_FLAGS -DUSE_ACC -DUSE_ACC_SPI_ICM20689 -DUSE_DSHOT -DUSE_FLASH -DUSE_GPS -DUSE_GYRO -DUSE_GYRO_SPI_ICM20689 -DUSE_LED_STRIP -DUSE_MAX7456 -DUSE_OSD -DUSE_OSD_HD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_SERIALRX_DEFAULT -DUSE_SERIALRX_GHST -DUSE_SERIALRX_SBUS -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_TELEMETRY_GHST -DUSE_VTX"

Docker log: 

Pulling from Github (4.4.0-RC3):
From https://github.com/betaflight/betaflight
 * [new tag]             4.4.0-RC3  -> cloud_build
[... 642 lines cut ...]
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

iFlight Succex-d F7 purchased from GetFPV

### Other components

Catalyst Machineworks Banggod Frame
DJI O3 Air Unit
Matek SAM-M8Q GPS
TBS Nano Diversity Receiver
SucceX 50A 2-6S BLHeli_32 4 in 1 ESC
iFlight Xing 2208 1300KV Motors

### How are the different components wired up

UART2 - DJI O3 Air Unit
UART4 - Matek SAM-M8Q GPS
UART3 - TBS Nano Diversity Receiver

blckmn, 2023-01-01:
Do you know what BARO is reported in 4.3.2? We can then add it to the configuration for the board.

closed 2023-01-01 by KarateBrot

hcolsen, 2023-01-01:
@blckmn I have the same issue on Diatone Mamba mk4 F722 APP 2022B.
Barometer on this is: SPL06

ally13coder, 2023-01-01:
I have the same issue on Speedybee F7v3 Barometer on board BMP 280

haslinghuis, 2023-01-01:
Speedybee F7V3 works on 4.4-RC3 - Also fixed the SDCARD;

ally13coder, 2023-01-01:
> 

no it still doesn't work.

ally13coder, 2023-01-01:
[image]

haslinghuis, 2023-01-01:
Using the 30x30 board here as test board. Maybe Speedybee has changed components. Flash using classic mode and type status in cli and copy the output here.

ally13coder, 2023-01-01:
# 

# status
MCU F722 Clock=216MHz, Vref=3.24V, Core temp=59degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 4140, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 2 locked dma shared
GYRO=BMI270, ACC=BMI270
OSD: MAX7456 (30 x 13)
BUILD KEY: 4.4.0-RC3/SPEEDYBEEF7V3/71c6e31d76cd4bfdbb5d2e2ff6e9cc44
System Uptime: 35 seconds, Current Time: 2023-01-01T16:45:42.995+00:00
CPU:10%, cycle time: 323, GYRO rate: 3095, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 22
SD card: Manufacturer 0x1d, 492544kB, 10/2016, v1.0, 'SD   '
Filesystem: Ready
Arming disable flags: RXLOSS ANGLE CLI MSP MOTOR_PROTO

#

haslinghuis, 2023-01-01:
Did it work on 4.3?

There is no trace of `BMP280`. Devices detected: I2C:0 Try adding one of these:

    BARO_BMP388
    BARO_MS5611 
    BARO_DPS310

Complete list: https://github.com/betaflight/betaflight/blob/master/docs/Manufacturer%20Design%20Guidelines.md#423-defines-for-baro

ally13coder, 2023-01-01:
yes it worked on 4.3

haslinghuis, 2023-01-01:
status report output from 4.3 please

ally13coder, 2023-01-01:
downgrade to 4.3 - now the barometer does not work either. I don't understand what to do.

haslinghuis, 2023-01-01:
Do you have other devices attached using I2C? if so detach physically as noise could disturb communication.

ally13coder, 2023-01-01:
not now. connected only to the receiver ELRS and Caddx Vista

haslinghuis, 2023-01-01:
Try replugging the board and use full chip erase while flashing. Just lets look at it without importing a diff or do any configuration.

ally13coder, 2023-01-01:
i did it. all the same. everything works except the barometer
[image]

haslinghuis, 2023-01-01:
Starts beginning to look like a hardware failure as it worked on 4.3 before and now it doesn't.

ally13coder, 2023-01-01:
Let's see if the problem appears on other flight controllers. And of course I would like to have the correct work and my particular controller. Perhaps the manufacturer will also pay attention to the problem.

PaulGrz, 2023-01-12:
I have exactly the same issue with different board, Skystars F4 F405HD2.

haslinghuis, 2023-01-12:
@PaulGrz please add custom define `BARO_SPI_BMP280` or wait for https://github.com/betaflight/unified-targets/pull/813

PaulGrz, 2023-01-12:
> @PaulGrz please add custom define `BARO_SPI_BMP280` or wait for [betaflight/unified-targets#813](https://github.com/betaflight/unified-targets/pull/813)

Actually solved it now by flashing in "Classic mode" 🤔

mcaraballo, 2023-01-15:
Sorry for the late reply.  BARO_SPI_BMP280 works for me with the iFlight
SucceX-D Mini F7 TwinG on Betaflight 4.4.0 RC3 but not on Betaflight 4.4.0
RC4.

Thanks,
--Mike C

haslinghuis, 2023-01-15:
@mcaraballo please provide board used (target), and a link where you bought the board

 iFlight SucceX-D Mini F7 TwinG seems to have `BARO_BMP280` and `BARO_DPS310` defined (I2C)

mcaraballo, 2023-01-16:
The board used (target) is EXF722DUAL.  I purchased the board in 2020 (though I am not 100% sure).  Could have been earlier in 2019.  This is precisely what it looks like https://www.getfpv.com/iflight-succex-d-f7-twing-v2-2-f722-flight-controller-hd-version.html .  It's an iFlight Succex-D (not the mini) I'm noticing some notes on GETFPV "BMP280 changed to DPS310" and I'm not sure now how to determine which I have.

mcaraballo, 2023-01-16:
Correction the target I am using is MATEKF722SE

haslinghuis, 2023-01-16:
@mcaraballo https://github.com/betaflight/unified-targets/pull/821 should solve the BARO.

mcaraballo, 2023-01-16:
> @mcaraballo <https://github.com/mcaraballo> betaflight/unified-targets#821
> <https://github.com/betaflight/unified-targets/pull/821> should solve the
> BARO.

Thanks.  So that I understand it correctly with my limited knowledge, once
the code is approved and merged into the main RC4 release, the firmware can
be downloaded and tried again.

Thanks so much,
Mike C

haslinghuis, 2023-01-16:
Yes and thank you for helping anyone else with this board configuration by creating this issue.

DHaacke, 2023-01-17:
@haslinghuis Having the same problem with no baro OR mag. Matek F722SE mini with DPS310 baro.  Worked on 4.3.2. Have two boards in front of me, one with 4.3.2 and one with 4.4 RC5.  Not in any rush, and can hold off trying classic mode if you'd like some particular information, just let me know. Good news the problem with no being able to connect to Configurator with the GPS/Compass powered up went away!

haslinghuis, 2023-01-17:
@DHaacke https://github.com/betaflight/unified-targets/blob/master/configs/default/MTKS-MATEKF722MINI.config has no baro defined.

https://github.com/betaflight/unified-targets/blob/master/configs/default/MTKS-MATEKF722SE.config has BMP280/DPS310 defined.

There is no `F722SE-MINI` target

mcaraballo, 2023-01-17:
> Yes and thank you for helping anyone else with this board configuration by creating this issue.

I just flashed Betaflight 4.4.0 RC5 and the issues with both the Baro and the Blackbox not working are now resolved!  Thank you so much for all the hard work you folks put into Betaflight!

DHaacke, 2023-01-17:
@haslinghuis Screen of it working under 4.3.2 if that helps. Sorry on the target name... should be MTKS/MATEKF722MINI.
[image]

haslinghuis, 2023-01-17:
@DHaacke I need the output of status command in cli. Also please provide URL where you got this board from.

DHaacke, 2023-01-17:
@haslinghuis Here's the status of the MATEKF722MINI with 4.3.2. It was purchased from racedayquads.com. Appreciate you looking at this!

DHaacke, 2023-01-17:
Yeah, sadly enough. Good little board. I'll try MATEKF722SE shortly and let you know. Thanks!

DHaacke, 2023-01-17:
@haslinghuis  That did the trick. Baro shows up now. However, Mag is not showing up. Note the error when applying my diff.

# get mag
###ERROR IN get: INVALID NAME: mag###

# status
MCU F722 Clock=216MHz, Vref=3.29V, Core temp=50degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3654, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000, BARO=DPS310
OSD: MAX7456 (30 x 13)
BUILD KEY: 021c0061b53b88e88794f6e5e3de47c4 (4.4.0-RC5)
System Uptime: 43 seconds, Current Time: 2023-01-17T17:44:18.122+00:00
CPU:54%, cycle time: 126, GYRO rate: 7936, RX rate: 15, System rate: 9
Voltage: 1659 * 0.01V (4S battery - OK)
I2C Errors: 0
SD card: Startup failed
Arming disable flags: RXLOSS CLI MSP

haslinghuis, 2023-01-17:
The `MAG` is onboard `GPS` device? Which GPS (link please)?

DHaacke, 2023-01-17:
I've been trying two different ones. The first is a Matek (http://www.mateksys.com/?portfolio=m8q-5883), the other the new Goku V3 (https://flywoo.net/collections/gps/products/goku-gm10-pro-v3-gps-w-compass). The Goku I haven't tinkered with much yet, but have confirmed the wiring is good. The Goku one is on 4.4, the Matek still on 4.3.2.  At the moment, not seeing mag on either. This is not critical. I hate to see you spend your precious time and expertise on this. The baro was critical, though. Really appreciate your tip on that!!

haslinghuis, 2023-01-17:
For both GPS you can add `MAG_QMC5883` in custom defines field.

Might need to add `MAG_DATA_READY_SIGNAL` but please try without first.

DHaacke, 2023-01-17:
Will do ASAP! Thanks!

DHaacke, 2023-01-17:
OK, first try with just MAG_QMC5883 resulted same, but does retain the Magnetometer switch.  Second try with MAG_DATA_READY_SIGNAL same result.  Did notice I2C errors are not zero, however.

MCU F722 Clock=216MHz, Vref=3.29V, Core temp=66degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3418, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000, BARO=DPS310
OSD: MAX7456 (30 x 13)
BUILD KEY: 4c335db31ef1babe81154419bad7903f (4.4.0-RC5)
System Uptime: 21 seconds, Current Time: 2023-01-17T20:32:11.746+00:00
CPU:25%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 1675 * 0.01V (4S battery - OK)
I2C Errors: 0
SD card: Startup failed
Arming disable flags: RXLOSS CLI MSP MOTOR_PROTO

haslinghuis, 2023-01-17:
Maybe besides `MAG_QMC5883` we also need `MAG_HMC5883`

Here is the complete list (skip the ones with SPI as we are using I2C here):

    MAG_DATA_READY_SIGNAL
    MAG_HMC5883
    MAG_SPI_HMC5883
    MAG_QMC5883
    MAG_LIS3MDL
    MAG_AK8963
    MAG_MPU925X_AK8963
    MAG_SPI_AK8963
#define USE_MAG_AK8975

https://www.betaflight.com/docs/documentation/Manufacturers/Manufacturer%20Design%20Guidelines#424-defines-for-mag

DHaacke, 2023-01-17:
So I'll try MAG_HMC5883 by itself and together with MAG_QMC5883.  I've got a 3 new GPS units coming this week, and I've got a spare MATEKF722MINI FC I'll rig to just power and GPS., and make sure it's got the SE firmware. I'll get back in a day or two. Holler if you have other thoughts... Thanks!!

DHaacke, 2023-01-18:
[image]
@haslinghuis I flashed the MATEKF722MINI with MATEKF722SE firmware (upgrading from 4.3.2 to 4.4), added MAG and MAG_QMC5883 and everything is working perfectly! GPS/Compass is the new Goku GPS-Compass V3.   From now on when the Mag doesn't show, I'll check the config to see if it's supported. Note: The mag was listed as a feature in the specs diagram. Lesson learned. Thanks for your help, as always!

chadmbrown, 2023-08-24:
TALONF7FUSION - I have the identical problem on 4.4.2. I've tried 4.4.3 as suggested by one user. I've also tried 4.5. In all cases the Baro won't enable. It also looks like many of the Baro CLI commands are no longer available in configurator. I will try to flash iNav as one user did just to make sure the Baro is working properly. 

[image]

Auge103, 2023-09-14:
Same issue on 4.4.2 and 4.5.0-zulu.
Am using the MAMBAH743 preset. Barometer is defined, but I cannot enable it nor do any baro functions exist in the CLI.
Flashing back BF4.3 does fix that issue and the baro is enabled.

haslinghuis, 2023-09-15:
Please follow https://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines