betaflight/betaflight #12127: Betaflight 4.4 RC3 - Can't choose Onboard Flash for Blackbox logging

mcaraballo, 2023-01-01:
### Describe the bug

In Betaflight 4.4 RC3 you cannot choose "Onboard Flash" for Blackbox logging via the GUI. The only option available is "serial port" or "no logging". This works just fine in Betaflight 4.3.2 stable.

I tested this on just one flight controller so this may be limited to that hardware. The FC is an iFlight Succex-D F7 - "EXUA/EXF722DUAL(STM32F7X2)"

### To Reproduce

1. Install the Betaflight 4.4 RC3 firmware fresh onto a iFlight Succex-D F7 flight controller using the Betaflight Configurator 10.9.0
2. Go into the blackbox tab
3. Try to select "Onboard Flash" for the Blackbox logging device

### Expected behavior

Expect to be able to choose "Onboard Flash for Blackbox logging as was possible in previous stable releases

### Cloud build log

```txt
***** Commencing Build *****

VERSION: 4.4.0-RC3
COMMIT:  4.4.0-RC3
BOARD:   EXF722DUAL/STM32F7X2
STARTED: 2023-01-01 02:12:09Z

Want to run this build locally?

git checkout 4.4.0-RC3 && make TARGET=STM32F7X2 EXTRA_FLAGS="-D'BUILD_KEY=4.4.0-RC3/EXF722DUAL/16135ba8b3f96266f4d69aac23b95b2a' -D'BOARD_NAME=EXF722DUAL' -D'MANUFACTURER_ID=EXUA' -DCLOUD_BUILD -DTARGET_FLAGS -DUSE_ACC -DUSE_ACC_SPI_ICM20689 -DUSE_DSHOT -DUSE_FLASH -DUSE_GPS -DUSE_GYRO -DUSE_GYRO_SPI_ICM20689 -DUSE_LED_STRIP -DUSE_MAX7456 -DUSE_OSD -DUSE_OSD_HD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_SERIALRX_DEFAULT -DUSE_SERIALRX_GHST -DUSE_SERIALRX_SBUS -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_TELEMETRY_GHST -DUSE_VTX"

Docker log: 

Pulling from Github (4.4.0-RC3):
From https://github.com/betaflight/betaflight
 * [new tag]             4.4.0-RC3  -> cloud_build

Checking out the branch -> cloud_build (for 4.4.0-RC3)
Switched to branch 'cloud_build'

Commencing the build (make STM32F7X2):
make -j ./obj/betaflight_4.4.0_STM32F7X2.hex
make[1]: Entering directory '/betaflight'
rm -f ./obj/main/STM32F7X2/.efhash_*
EF HASH -> ./obj/main/STM32F7X2/.efhash_e785ad20fb32d3fdea8d2e2f7d068a62
%% startup_stm32f722xx.s 
%% (optimised) ./src/main/startup/system_stm32f7xx.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu.c 
%% (optimised) ./src/main/drivers/adc_stm32f7xx.c 
%% (optimised) ./src/main/drivers/audio_stm32f7xx.c 
%% (speed optimised) ./src/main/drivers/bus_i2c_hal.c 
%% (size optimised) ./src/main/drivers/bus_i2c_hal_init.c 
%% (size optimised) ./src/main/drivers/bus_i2c_timing.c 
%% (optimised) ./src/main/drivers/dma_stm32f7xx.c 
%% (size optimised) ./src/main/drivers/light_ws2811strip_hal.c 
%% (size optimised) ./src/main/drivers/transponder_ir_io_hal.c 
%% (speed optimised) ./src/main/drivers/bus_spi_ll.c 
%% (optimised) ./src/main/drivers/persistent.c 
%% (optimised) ./src/main/drivers/dshot_bitbang.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_decode.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_ll.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_hal.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_shared.c 
%% (optimised) ./src/main/drivers/timer_hal.c 
%% (optimised) ./src/main/drivers/timer_stm32f7xx.c 
%% (optimised) ./src/main/drivers/system_stm32f7xx.c 
%% (optimised) ./src/main/drivers/serial_uart_hal.c 
%% (optimised) ./src/main/drivers/serial_uart_stm32f7xx.c 
%% (optimised) ./src/main/drivers/sdio_f7xx.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_fake.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu3050.c 
[... 606 lines cut ...]
DMA2 Stream 4: DSHOT_BITBANG 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

iFlight Succex-d F7 purchased from GetFPV

### Other components

Catalyst Machineworks Banggod Frame
DJI O3 Air Unit
Matek SAM-M8Q GPS
TBS Nano Diversity Receiver
SucceX 50A 2-6S BLHeli_32 4 in 1 ESC
iFlight Xing 2208 1300KV Motors

### How are the different components wired up

UART2 - DJI O3 Air Unit
UART4 - Matek SAM-M8Q GPS

haslinghuis, 2023-01-01:
Build with `FLASH_W25Q128FV` in Custom defines.

Other options:

    FLASH_M25P16
    FLASH_W25N01G
    FLASH_W25M
    FLASH_W25M512
    FLASH_W25M02G
    FLASH_W25Q128FV

Please report back which one worked so we can add it to configuration.

N3W-Be, 2023-01-03:
Same with the  HIFIONRCF7/STM32F7X2 

***** Commencing Build *****

VERSION: 4.4.0-RC3
COMMIT:  4.4.0-RC3
BOARD:   HIFIONRCF7/STM32F7X2
STARTED: 2023-01-03 21:09:16Z

Want to run this build locally?

git checkout 4.4.0-RC3 && make TARGET=STM32F7X2 EXTRA_FLAGS="-D'BUILD_KEY=4.4.0-RC3/HIFIONRCF7/aff71c7d04a02b1feafca140653a67a6' -D'BOARD_NAME=HIFIONRCF7' -D'MANUFACTURER_ID=HFOR' -DCLOUD_BUILD -DTARGET_FLAGS -DUSE_ACC -DUSE_ACC_SPI_MPU6000 -DUSE_DSHOT -DUSE_FLASH -DUSE_GPS -DUSE_GYRO -DUSE_GYRO_SPI_MPU6000 -DUSE_LED_STRIP -DUSE_MAX7456 -DUSE_OSD -DUSE_OSD_HD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX"

Docker log: 

Pulling from Github (4.4.0-RC3):
From https://github.com/betaflight/betaflight
 * [new tag]             4.4.0-RC3  -> cloud_build

Checking out the branch -> cloud_build (for 4.4.0-RC3)
Switched to branch 'cloud_build'

Commencing the build (make STM32F7X2):
make -j ./obj/betaflight_4.4.0_STM32F7X2.hex
make[1]: Entering directory '/betaflight'
rm -f ./obj/main/STM32F7X2/.efhash_*
EF HASH -> ./obj/main/STM32F7X2/.efhash_063bd623e16f9a81a6cda43ea196ee03
%% startup_stm32f722xx.s 
%% (optimised) ./src/main/startup/system_stm32f7xx.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu.c 
%% (optimised) ./src/main/drivers/adc_stm32f7xx.c 
%% (optimised) ./src/main/drivers/audio_stm32f7xx.c 
%% (speed optimised) ./src/main/drivers/bus_i2c_hal.c 
%% (size optimised) ./src/main/drivers/bus_i2c_hal_init.c 
%% (size optimised) ./src/main/drivers/bus_i2c_timing.c 
%% (optimised) ./src/main/drivers/dma_stm32f7xx.c 
%% (size optimised) ./src/main/drivers/light_ws2811strip_hal.c 
%% (size optimised) ./src/main/drivers/transponder_ir_io_hal.c 
%% (speed optimised) ./src/main/drivers/bus_spi_ll.c 
%% (optimised) ./src/main/drivers/persistent.c 
%% (optimised) ./src/main/drivers/dshot_bitbang.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_decode.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_ll.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_hal.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_shared.c 
%% (optimised) ./src/main/drivers/timer_hal.c 
%% (optimised) ./src/main/drivers/timer_stm32f7xx.c 
%% (optimised) ./src/main/drivers/system_stm32f7xx.c 
%% (optimised) ./src/main/drivers/serial_uart_hal.c 
%% (optimised) ./src/main/drivers/serial_uart_stm32f7xx.c 
%% (optimised) ./src/main/drivers/sdio_f7xx.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_fake.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu3050.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_mpu6050.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_mpu6500.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_spi_bmi160.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_spi_bmi270.c 
%% (optimised) ./src/main/drivers/accgyro/accgyro_spi_icm20649.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_spi_icm20689.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_spi_icm426xx.c 
%% (optimised) ./src/main/drivers/accgyro/accgyro_spi_l3gd20.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_spi_lsm6dso.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_spi_lsm6dso_init.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_spi_mpu6000.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_spi_mpu6500.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_spi_mpu9250.c 
%% (optimised) ./src/main/drivers/accgyro/gyro_sync.c 
%% (optimised) lib/main/BoschSensortec/BMI270-Sensor-API/bmi270_maximum_fifo.c 
%% (size optimised) ./src/main/drivers/barometer/barometer_2smpb_02b.c 
%% (size optimised) ./src/main/drivers/barometer/barometer_bmp085.c 
[... 480 lines cut ...]

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 10
set pitch_rc_rate = 10
set yaw_rc_rate = 10
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save

REK07, 2023-01-03:
Used custom defines FLASH_W25M512  to get the data flash back on the iFlight SucceX-D Mini F7 TwinG Digital Flight Controller with 
4.4 RC3

haslinghuis, 2023-01-03:
When above PR is merged it should work.

closed 2023-01-04 by blckmn

mcaraballo, 2023-01-15:
Thank you and sorry for the late reply!  This works in RC3 but not in RC4.
Suggestions?

Thanks,
--Mike C

haslinghuis, 2023-01-16:
Please try `FLASH_W25M512` in custom defines.
If that does not work try `FLASH_W25Q128FV`

Other options:

    FLASH_M25P16
    FLASH_W25M
    FLASH_W25M512
    FLASH_W25N01G
    FLASH_W25M02G
    FLASH_W25Q128FV

Please report back which one worked so we can add it to configuration `EXUA/EXF722DUAL`

REK07, 2023-01-16:
Will doSent from my iPhone

haslinghuis, 2023-01-16:
After more (re)search think it's `FLASH_M25P16` (already added - waiting for approval)

REK07, 2023-01-16:
I’ll get to it as quick as I can. Thanks.Sent from my iPhone

REK07, 2023-01-17:
I used FLASH_W25M512 to get the data flash back on the iFlight SucceX-D Mini F7 TwinG Digital Flight Controller with
4.4 RC3. Works Know

haslinghuis, 2023-01-17:
So `FLASH_M25P16` is wrong - as that was in the docs?

REK07, 2023-01-17:
This Blackbox problem was with an  iFlight Succex-D Mini F7 TwinG Stack

Rob

javamatte, 2023-01-19:
I've got a KAKUTEF4V2 board (Holybro Kakute F4 v2 Mini) that had the same problem as above (flashed from 4.3 -> 4.4-RC4 and lost my internal blackbox flash device). Flashing to RC5 did not bring my internal blackbox logging back.

Adding `FLASH_W25Q128FV` to the custom defines and re-flashing to RC5 _did_ make the flash chip reappear in BF Configurator.

I'm assuming that define need to be added to the unified target here:
https://github.com/betaflight/unified-targets/blob/dcb0dc1ea9523948dae04a5efd10f13b860513cc/configs/default/HBRO-KAKUTEF4V2.config#L4

demvlad, 2023-02-16:
Hello! I have simmilar problem in 4.4 release for Airbot  Omni NXT F7 flight controller. I use default option by flashing firmware. What additional option can i use?

haslinghuis, 2023-02-16:
If your board peripherals are not recognized please help us add the required configuration details.

1. Flash your board with the Core Only switch enabled
2. Go to the CLI tab and click the Submit Support Data button.
3. With this generated support ID we have all required information to update board configuration files.

demvlad, 2023-02-17:
Hello!
This is support data from my flight controller.

Best regards,
Vladimir

2023-02-17 @17:09:43 -- ОС: Windows

2023-02-17 @17:09:43 -- Конфигуратор: 10.9.0 (697943b)

2023-02-17 @17:09:44 -- Из GitHub загружена информация о выпуске для 
configurator.

2023-02-17 @17:09:47 -- Последовательный порт успешно открыт. ID порта: 1

2023-02-17 @17:09:47 -- MultiWii API версия: 1.45.0

2023-02-17 @17:09:47 -- Информация о полётном контроллере, идентификатор: 
BTFL, версия: 4.4.0

2023-02-17 @17:09:47 -- Используется прошивка, выпущенная в: Feb 2 2023 
14:52:51

2023-02-17 @17:09:47 -- Плата: AIRB/OMNINXT7(STM32F7X2), версия: 0

2023-02-17 @17:09:47 -- Уникальный ID устройства: 0x52002c3137511830343533

2023-02-17 @17:09:47 -- Позывной пилота:

2023-02-17 @17:09:47 -- Arming выключен

2023-02-17 @17:09:50 -- Обнаружен переход в режим командной строки

2023-02-17 @17:11:07 -- Последовательный порт успешно закрыт

2023-02-17 @17:11:08 -- Использование кэшированной информации о сборках для 
https://build.betaflight.com/api/targets.

2023-02-17 @17:11:08 -- Использование кэшированной информации о сборках для 
https://build.betaflight.com/api/targets.

2023-02-17 @17:11:08 -- Использование кэшированной информации о сборках для 
https://build.betaflight.com/api/targets.

2023-02-17 @17:11:08 -- Использование кэшированной информации о сборках для 
https://build.betaflight.com/api/targets.

2023-02-17 @17:11:09 -- Query board information to preselect right firmware

2023-02-17 @17:11:09 -- Configurator has successfully detected and verified 
the board: OMNINXT7

2023-02-17 @17:11:32 -- Success: 
https://build.betaflight.com/api/builds/438cbdb66db3eb27370e58ce9226c18d/status

2023-02-17 @17:11:33 -- Success: 
/api/builds/438cbdb66db3eb27370e58ce9226c18d/hex

2023-02-17 @17:11:54 -- MultiWii API версия: 1.45.0

2023-02-17 @17:11:54 -- Устройство - Перезагрузка для загрузки бутлоадера

2023-02-17 @17:11:55 -- USB устройство успешно подключено. ID подключения: 1

2023-02-17 @17:11:55 -- Обнаружено устройство с суммарным размером 
flash-памяти 512 KiB

2023-02-17 @17:11:55 -- Chrome API Error: Transfer failed..

2023-02-17 @17:11:55 -- Защита чтения не установлена
[... 421 lines cut ...]
set gyro_2_align_yaw = 2700
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

*** Support data submitted ***
Id: "2092a388-07ad-4255-b7cb-74c5c3d19ef0"

# copy ID and provide to the betaflight team.

Mark Haslinghuis ***@***.***> 16 февраля 2023 г. 21:27:34 
написал:

demvlad, 2023-02-18:
I have used FLASH_W25Q128FW define while loaded firmware. All good - Flash board is work. Thank you

DylanSandry, 2023-03-10:
I have a CLRACING4 FC and I'm having this issue. I'm halfway through the custom defines, but I just want to know, after the flash when it asks for custom defaults, do I hit yes or no? I just want to be sure I'm not putting in a custom define and then wiping it with custom defaults.

demvlad, 2023-03-10:
I has accepted custom defaults settings by first connection to configuration after flash firmware. All work good.