betaflight/betaflight #12128: Mamba mk4 F722 App 2022B - GPS and BARO not possible to enable

hcolsen, 2023-01-01:
### Describe the bug

BF 4.4 RC3 can`t find either the GPS or BARO

### To Reproduce

I try to enable BARO in Configuration, but after save and reboot, it is still tured off.
For the GPS part I activate the same settings and ports config as earlier BF releaces.

### Expected behavior

Expect the baro and gps to work like normal :)

### Diff configuration

```txt
# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan  1 2023 / 05:47:58 (2c34b3949) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name MAMBAF722_2022B
manufacturer_id DIAT

# name: ROMA F5

# feature
feature -TELEMETRY
feature GPS

# serial
serial 1 2 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 3 1 115200 57600 0 115200
serial 5 0 115200 0 0 115200

# map
map TREA1234

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 3 1700 2100 0 0
aux 2 2 3 1300 1700 0 0
aux 3 13 1 1700 2100 0 0
aux 4 35 2 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1          A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2          B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3          E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4          F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5          R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 5 25 100 250 1000
vtxtable powerlabels 5 25 100 250 1K

# master
set acc_calibration = 13,-37,29,1
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT600
set motor_output_reordering = 3,2,1,0,4,5,6,7
set failsafe_procedure = AUTO-LAND
set gps_provider = NMEA
set gps_auto_config = OFF
set osd_vbat_pos = 2454
set osd_tim_1_pos = 2198
set osd_tim_2_pos = 2230
set osd_remaining_time_estimate_pos = 2424
set osd_gps_speed_pos = 2081
set osd_gps_sats_pos = 2092
set osd_home_dir_pos = 2147
set osd_home_dist_pos = 2177
set osd_altitude_pos = 2166
set osd_avg_cell_voltage_pos = 2434
set osd_core_temp_pos = 2103
set osd_sys_distance_pos = 193
set osd_sys_vtx_temp_pos = 2135
set osd_stat_bitmask = 16172
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 2
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5917
set vcd_video_system = NTSC
set craft_name = ROMA F5

profile 0

# profile 0
set profile_name = stock BF
set tpa_mode = PD

rateprofile 0

# end the command batch
batch end
```

### Resources configuration

```txt
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 2
A09: MOTOR 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: FREE
B01: FREE
B02: BEEPER
B03: LED_STRIP
B04: FREE
B05: SPI_MOSI 3
B06: SERIAL_TX 1
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: PINIO 2
C01: ADC_BATT
C02: PINIO 1
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 4
C09: MOTOR 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2 : LED_STRIP
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 1
    CH4 : DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: LED_STRIP
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 3
```

### Flight controller

Diatone Mamba F722 MK4 APP 2022B

### Other components

Diatone Roma F5 v2 with GPS. Dont know the make of this GPS..

hcolsen, 2023-01-01:
I replaced the GPS with E108-GN02D from Ebyte:
https://www.ebyte.com/en/product-view-news.html?id=1125

Same problem, GPS not showing

haslinghuis, 2023-01-01:
Adding the BARO driver now. For now add `BARO_DPS310` in Custom Defines.

GPS you have to include before triggering cloud build. Also try another serial port. 

The cloud build log is missing.

hcolsen, 2023-01-01:
I just flashed back to 4.3.2 - a checked that the GPS and BARO works as normal. 
dobbelchecking the ports and GPS settings. and flashing back to 4.4 rc3.
And I`ll figure out "cloud build" 

I will update! Thanks @haslinghuis!

hcolsen, 2023-01-01:
> Adding the BARO driver now. For now add `BARO_DPS310` in Custom Defines.
> 
> GPS you have to include before triggering cloud build. Also try another serial port.
> 
> The cloud build log is missing.

After reflash with "BARO_DPS310" in custom defines it all works. I dobble checked ports in BF4.3.1 and now both gps and baro works 👯

closed 2023-01-01 by hcolsen

feivur, 2023-02-07:
Facing this issue.

Board: Diatone Mamba AIO F722 MK1, DIAT/MAMBAF722_2022B(STM32F7X2)
Flash with BARO_DPS310 in custom defines, baro enabled in configuration.
Baro works in Betaflight 4.3.2!

# status
MCU F722 Clock=216MHz, Vref=3.26V, Core temp=44degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3623, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1
GYRO=ICM42688P, ACC=ICM42688P
OSD: NONE (33 x 185)
BUILD KEY: 4f5ed44cffb3f26cbe1d1715336edcb4 (4.4.0)
System Uptime: 12 seconds, Current Time: 2023-02-07T07:16:13.602+00:00
CPU:29%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 10
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
Arming disable flags: RXLOSS CLI MSP MOTOR_PROTO