betaflight/betaflight #12129: Barometer not working on MAMBAF722_I2C (4.4.0-RC3)

klavifon, 2023-01-01:
### Describe the bug

Hello,

I flashed the newest firmware of BF (4.4.0-RC3) on my MAMBA F722_I2C flight controller and onboard barometer is not recognisable. I then tried to flash it to the last stable release (4.3.2) and there it was again showing in the configurator. It's definitely a bug. 

### To Reproduce

Flash the board with 4.0.0-RC3

### Expected behavior

I expected the barometer to work

### Diff configuration

```txt
2023-01-01 @15:51:07 -- OS: macOS

2023-01-01 @15:51:07 -- Configurator: 10.9.0-RC3 (norevision)

2023-01-01 @15:51:07 -- Loaded release information for configurator from GitHub.

2023-01-01 @15:51:17 -- Serial port successfully opened with ID: 1

2023-01-01 @15:51:17 -- MultiWii API version: 1.45.0

2023-01-01 @15:51:17 -- Flight controller info, identifier: BTFL, version: 4.4.0

2023-01-01 @15:51:17 -- Running firmware released on: Jan 1 2023 00:06:37

2023-01-01 @15:51:17 -- Board: DIAT/MAMBAF722_I2C(STM32F7X2), version: 0

2023-01-01 @15:51:17 -- Unique device ID: 0x1c00245632501520323236

2023-01-01 @15:51:17 -- Craft name: BUDDY7

2023-01-01 @15:51:17 -- Arming Disabled

2023-01-01 @15:51:30 -- Serial port successfully closed

2023-01-01 @15:51:30 -- Loaded builds information for https://build.betaflight.com/api/targets from build server.

2023-01-01 @15:51:30 -- Using cached builds information for https://build.betaflight.com/api/targets.

2023-01-01 @15:51:30 -- Loaded builds information for https://build.betaflight.com/api/targets from build server.

2023-01-01 @15:51:30 -- Using cached builds information for https://build.betaflight.com/api/targets.

2023-01-01 @15:51:31 -- Loaded builds information for https://build.betaflight.com/api/options from build server.

2023-01-01 @15:51:31 -- Loaded builds information for https://build.betaflight.com/api/options from build server.

2023-01-01 @15:51:34 -- Query board information to preselect right firmware

2023-01-01 @15:51:34 -- Configurator has successfully detected and verified the board: MAMBAF722_I2C

2023-01-01 @15:51:34 -- Loaded builds information for https://build.betaflight.com/api/targets/MAMBAF722_I2C from build server.

2023-01-01 @15:51:35 -- Loaded builds information for https://build.betaflight.com/api/builds/4.4.0-RC3/MAMBAF722_I2C from build server.

2023-01-01 @15:51:41 -- Query board information to preselect right firmware

2023-01-01 @15:51:41 -- Configurator has successfully detected and verified the board: MAMBAF722_I2C

2023-01-01 @15:52:18 -- Serial port successfully opened with ID: 4

2023-01-01 @15:52:18 -- MultiWii API version: 1.45.0

2023-01-01 @15:52:18 -- Flight controller info, identifier: BTFL, version: 4.4.0

2023-01-01 @15:52:18 -- Running firmware released on: Jan 1 2023 00:06:37

2023-01-01 @15:52:18 -- Board: DIAT/MAMBAF722_I2C(STM32F7X2), version: 0

2023-01-01 @15:52:18 -- Unique device ID: 0x1c00245632501520323236

2023-01-01 @15:52:18 -- Craft name: BUDDY7

2023-01-01 @15:52:18 -- Arming Disabled

2023-01-01 @15:52:19 -- Arming Enabled

2023-01-01 @15:52:19 -- Runaway Takeoff Prevention Enabled

2023-01-01 @15:52:19 -- Serial port successfully closed

2023-01-01 @17:02:14 -- Serial port successfully opened with ID: 5

2023-01-01 @17:02:14 -- MultiWii API version: 1.45.0

2023-01-01 @17:02:14 -- Flight controller info, identifier: BTFL, version: 4.4.0

2023-01-01 @17:02:14 -- Running firmware released on: Jan 1 2023 00:06:37

2023-01-01 @17:02:14 -- Board: DIAT/MAMBAF722_I2C(STM32F7X2), version: 0

2023-01-01 @17:02:14 -- Unique device ID: 0x1c00245632501520323236

2023-01-01 @17:02:14 -- Craft name: BUDDY7

2023-01-01 @17:02:14 -- Arming Disabled

2023-01-01 @17:02:17 -- CLI mode detected

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan  1 2023 / 00:06:37 (2c34b3949) MSP API: 1.45

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAF722_I2C
manufacturer_id DIAT
mcu_id 001c00245632501520323236
signature 

# name: BUDDY7

# feature
feature -TELEMETRY
feature GPS

# serial
serial 1 131073 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 4 2 115200 9600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1300 2100 0 0
aux 2 27 3 1700 2100 0 0
aux 3 46 1 1700 2100 0 0
aux 4 13 2 1700 2100 0 0
aux 5 26 0 1700 2100 0 0

# rxfail
rxfail 3 s 1275
rxfail 5 s 1500

# master
set gyro_lpf2_static_hz = 1000
set dyn_notch_count = 2
set dyn_notch_max_hz = 800
set gyro_lpf1_dyn_min_hz = 120
set gyro_lpf1_dyn_max_hz = 350
set acc_lpf_hz = 10
set acc_calibration = -44,-2,48,1
set min_check = 1010
set max_check = 1990
set rc_smoothing_auto_factor = 80
set rc_smoothing_setpoint_cutoff = 25
set rc_smoothing_feedforward_cutoff = 25
set serialrx_provider = CRSF
set dshot_bidir = ON
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 4500
set vbat_max_cell_voltage = 435
set vbat_scale = 113
set ibata_scale = 80
set yaw_motors_reversed = ON
set small_angle = 180
set gps_sbas_mode = AUTO
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_min_sats = 6
set simplified_gyro_filter = OFF
set osd_warn_bitmask = 57215
set osd_link_quality_pos = 3149
set osd_link_tx_power_pos = 3277
set osd_rssi_dbm_pos = 3213
set osd_flymode_pos = 3662
set osd_throttle_pos = 2506
set osd_current_pos = 2714
set osd_mah_drawn_pos = 2651
set osd_gps_speed_pos = 2378
set osd_gps_sats_pos = 2122
set osd_home_dir_pos = 2142
set osd_home_dist_pos = 2204
set osd_flight_dist_pos = 3405
set osd_altitude_pos = 18570
set osd_warnings_pos = 14808
set osd_avg_cell_voltage_pos = 2634
set osd_stat_bitmask = 10882620
set osd_displayport_device = MSP
set vcd_video_system = HD
set gyro_1_align_yaw = 1800
set craft_name = BUDDY7

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 0
set dterm_lpf1_dyn_max_hz = 0
set dterm_lpf1_dyn_expo = 0
set dterm_lpf1_static_hz = 80
set dterm_lpf2_static_hz = 140
set p_pitch = 98
set i_pitch = 131
set d_pitch = 64
set f_pitch = 204
set p_roll = 85
set i_roll = 113
set d_roll = 54
set f_roll = 178
set p_yaw = 85
set i_yaw = 113
set f_yaw = 178
set d_min_roll = 54
set d_min_pitch = 64
set feedforward_smooth_factor = 30
set feedforward_jitter_factor = 10
set simplified_master_multiplier = 165
set simplified_i_gain = 75
set simplified_d_gain = 110
set simplified_pi_gain = 115
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 90
set simplified_pitch_d_gain = 105
set simplified_pitch_pi_gain = 110
set simplified_dterm_filter = OFF
set tpa_mode = PD

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 19
set pitch_rc_rate = 19
set yaw_rc_rate = 19
set roll_expo = 56
set pitch_expo = 56
set yaw_expo = 56
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 72

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

### Resources configuration

```txt
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: SERIAL_TX 1
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 1
C09: MOTOR 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 3
    CH4 : DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 1
```

### Flight controller

MAMBA F722_I2C. It was bought along with the ESC as a stack in Europe

haslinghuis, 2023-01-01:
Please report status output while BARO is working so we can add it to target configuration

For now you could add `BARO_BMP280` to Custom Defines

klavifon, 2023-01-01:
Hello,

thanks for the reply. Sorry but I don’t know exactly how do I report status output. I flashed the board back to 4.3.2. Do you need a diff or what? Because I’m not actually a computer tech guy… A little help please.

Best regards,
Ales

haslinghuis, 2023-01-01:
:) Go the CLI tab. Type Status. Select and copy the output and paste here.

klavifon, 2023-01-01:
Now we’re talking :)

MCU F722 Clock=216MHz, Vref=3.27V, Core temp=59degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 4034, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MSP
System Uptime: 777 seconds, Current Time: 2023-01-01T17:40:05.503+00:00
CPU:57%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 1
SD card: Not configured
Arming disable flags: RXLOSS CLI MSP GPS RPMFILTER

Thanks!

haslinghuis, 2023-01-01:
Ok adding `BMP280` to configuration. Will take a little until merged as its new years day :)

For now you use 4.4 while adding `BARO_BMP280` to Custom Defines field.

klavifon, 2023-01-01:
For testing the new GPS RTH feature this is more than enough. 
Thank you mister!

Happy new year to you!

closed 2023-01-02 by blckmn

drisstron, 2023-07-14:
Same thing for HAKRC F405D, barometer perfectly working on 4.3.2 but when flashed to 4.4 for Gps rescue it does not show up.  status output for 4.3.2:
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.29V, Core temp=53degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4100, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MAX7456
System Uptime: 11 seconds, Current Time: 2023-07-14T12:07:27.853+00:00
CPU:40%, cycle time: 107, GYRO rate: 9345, RX rate: 15, System rate: 9
Voltage: 5 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: RXLOSS CLI MSP MOTOR_PROTO

haslinghuis, 2023-07-14:
@drisstron according to HAKRC it has a DPS310 baro: https://www.hakrc.com/HAKRCF405-D-Mini-FC.html on I2C port 2. But only I2C port is defined.

Please follow instructions here: https://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines and post the support ID.

drisstron, 2023-07-14:
Thank you for the reply,
i solved the problem by flashing betaflight_4.4.2_STM32F405.hex manualy
it seems there is someting with the cloud build.

haslinghuis, 2023-07-14:
Since 4.4.0 we use hardware defines for board configuration. Baro defines are missing as nobody brought it up until now.