betaflight/betaflight #12130: Barometer and magnetometer stopped working Betaflight 4.4 RC3

ally13coder, 2023-01-01:
### Describe the bug

After flash firmware 4.4, the barometer and magnetometer stopped working on my Speedybee F7v3 flight controller. BMP 280 barometer on board and compass on external module M8Q 5883. 

### To Reproduce

1. Install the Betaflight 4.4 RC3 firmware flash onto a my Speedybee F7v3 flight controller using the Betaflight Configurator 10.9.0
2. Go into the configuration tab
3. Click on the Barometer (if supported) switch
4. Click save and reboot
5. Realize the baro is not turned on

### Expected behavior

When clicking the barometer option in the Betaflight Configurator 10.9.0 it is expected to turn on barometer support and the switch in the UI to show as on.

### Cloud build log

```txt
**** Commencing Build *****

VERSION: 4.4.0-RC3
COMMIT:  4.4.0-RC3
BOARD:   SPEEDYBEEF7V3/STM32F7X2
STARTED: 2022-12-31 22:32:14Z

Want to run this build locally?

git checkout 4.4.0-RC3 && make TARGET=STM32F7X2 EXTRA_FLAGS="-D'BUILD_KEY=4.4.0-RC3/SPEEDYBEEF7V3/acf8f5e08e8332eb97beea54a885e411' -D'BOARD_NAME=SPEEDYBEEF7V3' -D'MANUFACTURER_ID=SPBE' -DCLOUD_BUILD -DTARGET_FLAGS -DUSE_ACC -DUSE_ACCGYRO_BMI270 -DUSE_BARO -DUSE_BARO_BMP280 -DUSE_DSHOT -DUSE_FLASH -DUSE_GPS -DUSE_GYRO -DUSE_LED_STRIP -DUSE_MAX7456 -DUSE_OSD -DUSE_OSD_HD -DUSE_PINIO -DUSE_SDCARD -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_SERIALRX_DEFAULT -DUSE_SERIALRX_GHST -DUSE_SERIALRX_SBUS -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_TELEMETRY_GHST -DUSE_VTX"

Docker log: 

Pulling from Github (4.4.0-RC3):
From https://github.com/betaflight/betaflight
 * [new tag]             4.4.0-RC3  -> cloud_build

Checking out the branch -> cloud_build (for 4.4.0-RC3)
Switched to branch 'cloud_build'

Commencing the build (make STM32F7X2):
make -j ./obj/betaflight_4.4.0_STM32F7X2.hex
make[1]: Entering directory '/betaflight'
rm -f ./obj/main/STM32F7X2/.efhash_*
EF HASH -> ./obj/main/STM32F7X2/.efhash_1d417307b9d2e6b9a66cf8c3274f16e5
%% startup_stm32f722xx.s 
%% (optimised) ./src/main/startup/system_stm32f7xx.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu.c 
%% (optimised) ./src/main/drivers/adc_stm32f7xx.c 
%% (optimised) ./src/main/drivers/audio_stm32f7xx.c 
%% (speed optimised) ./src/main/drivers/bus_i2c_hal.c 
%% (size optimised) ./src/main/drivers/bus_i2c_hal_init.c 
%% (size optimised) ./src/main/drivers/bus_i2c_timing.c 
%% (optimised) ./src/main/drivers/dma_stm32f7xx.c 
%% (size optimised) ./src/main/drivers/light_ws2811strip_hal.c 
%% (size optimised) ./src/main/drivers/transponder_ir_io_hal.c 
%% (speed optimised) ./src/main/drivers/bus_spi_ll.c 
%% (optimised) ./src/main/drivers/persistent.c 
%% (optimised) ./src/main/drivers/dshot_bitbang.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_decode.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_ll.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_hal.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_shared.c 
%% (optimised) ./src/main/drivers/timer_hal.c 
%% (optimised) ./src/main/drivers/timer_stm32f7xx.c 
%% (optimised) ./src/main/drivers/system_stm32f7xx.c 
%% (optimised) ./src/main/drivers/serial_uart_hal.c 
%% (optimised) ./src/main/drivers/serial_uart_stm32f7xx.c 
%% (optimised) ./src/main/drivers/sdio_f7xx.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_fake.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu3050.c 
[... 691 lines cut ...]
Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: DSHOT_BITBANG 1
DMA2 Stream 2: LED_STRIP
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: DSHOT_BITBANG 2
DMA2 Stream 7: FREE

#
```

### Flight controller

SPEEDYBEEF7V3

haslinghuis, 2023-01-01:
From @ally13coder:

`status`

```
MCU F722 Clock=216MHz, Vref=3.24V, Core temp=59degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 4140, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 2 locked dma shared
GYRO=BMI270, ACC=BMI270
OSD: MAX7456 (30 x 13)
BUILD KEY: 4.4.0-RC3/SPEEDYBEEF7V3/71c6e31d76cd4bfdbb5d2e2ff6e9cc44
System Uptime: 35 seconds, Current Time: 2023-01-01T16:45:42.995+00:00
CPU:10%, cycle time: 323, GYRO rate: 3095, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 22
SD card: Manufacturer 0x1d, 492544kB, 10/2016, v1.0, 'SD '
Filesystem: Ready
Arming disable flags: RXLOSS ANGLE CLI MSP MOTOR_PROTO
```

Downgrade to 4.3 - now the barometer does not work either.

Do you have other devices attached using I2C? if so detach physically as noise could disturb communication.
Connected only to the receiver ELRS and Caddx Vista

haslinghuis, 2023-01-01:
Having the SPEEDYBEEF7V3 30x30 so it's confirmed working on our side.