betaflight/betaflight #12133: Betaflight 4.4 RC3 - choosing SPI RX, save and reboot causes loss of comm port

jgauruder, 2023-01-02:
### Describe the bug

Betaflight 4.4 RC3 - choosing SPI RX, save and reboot causes loss of comm port

Superbee F4 lite frsky v1.0 fc works fine on 4.4 rc1 and rc2.  Using 10.9.0-rc3 configurator on MacBook.  Flashed 4.4rc3 to this fc and it flashes fine. I used classic mode and full chip erase. Apply custom defaults when prompted fine.  Reconnect to fc fine. When I go to receiver tab and change from serial to spi rx and frsky_d and click save and reboot, the USB comm port is inaccessible.   Only way to recover is bridge the two touch points on the fc with a small screw driver while reconnecting USB to enter DFU mode and reflash to 4.4 rc2 .  

### To Reproduce

flash superbee f4 lite frsky fc to 4.4 RC3.   once flashed, apply custom defaults, reconnect when done.  goto receiver tab, select SPI RX and frsky_d.  click save and reboot.   cannot access comm port any longer.  have to recover using bridging of the two points on the fc w/ small screwdriver (there is no "button" on this fc to enter DFU mode)

### Expected behavior

no loss of comm port.

### Cloud build log

```txt
***** Commencing Build *****

VERSION: 4.4.0-RC3
COMMIT:  4.4.0-RC3
BOARD:   CRAZYBEEF4FR/STM32F411
STARTED: 2023-01-02 13:12:36Z

TESTING TARGET FLAGS: -DTARGET_FLAGS -DUSE_ACC -DUSE_ACC_SPI_MPU6000 -DUSE_ACCGYRO_BMI270 -DUSE_GYRO -DUSE_GYRO_SPI_MPU6000 -DUSE_MAX7456 -DUSE_OSD -DUSE_RX_CC2500 -DUSE_RX_SPI

Want to run this build locally?

git checkout 4.4.0-RC3 && make TARGET=STM32F411 EXTRA_FLAGS="-D'BUILD_KEY=4.4.0-RC3/CRAZYBEEF4FR/4a1e142bdf449aec6f2a66ff6e00fcfd' -D'BOARD_NAME=CRAZYBEEF4FR' -D'MANUFACTURER_ID=HAMO'"

Docker log: 

Pulling from Github (4.4.0-RC3):
From https://github.com/betaflight/betaflight
 * [new tag]             4.4.0-RC3  -> cloud_build

Checking out the branch -> cloud_build (for 4.4.0-RC3)
Switched to branch 'cloud_build'

Commencing the build (make STM32F411):
make -j ./obj/betaflight_4.4.0_STM32F411.hex
make[1]: Entering directory '/betaflight'
rm -f ./obj/main/STM32F411/.efhash_*
EF HASH -> ./obj/main/STM32F411/.efhash_b3d3bfb2808805cd375e4c4574373960
%% startup_stm32f411xe.s 
%% (optimised) ./src/main/startup/system_stm32f4xx.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu.c 
%% (optimised) ./src/main/drivers/adc_stm32f4xx.c 
%% (optimised) ./src/main/drivers/bus_i2c_stm32f4xx.c 
%% (optimised) ./src/main/drivers/bus_spi_stdperiph.c 
%% (optimised) ./src/main/drivers/dma_stm32f4xx.c 
%% (optimised) ./src/main/drivers/dshot_bitbang.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_decode.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_stdperiph.c 
%% (size optimised) ./src/main/drivers/inverter.c 
%% (size optimised) ./src/main/drivers/light_ws2811strip_stdperiph.c 
%% (size optimised) ./src/main/drivers/transponder_ir_io_stdperiph.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_shared.c 
%% (optimised) ./src/main/drivers/serial_uart_stdperiph.c 
%% (optimised) ./src/main/drivers/serial_uart_stm32f4xx.c 
%% (optimised) ./src/main/drivers/system_stm32f4xx.c 
%% (optimised) ./src/main/drivers/timer_stm32f4xx.c 
%% (optimised) ./src/main/drivers/persistent.c 
%% (optimised) ./src/main/drivers/sdio_f4xx.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_fake.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu3050.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_mpu6050.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_mpu6500.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_spi_bmi160.c 
[... 701 lines cut ...]
Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

fc is from Happymodel mobula 6 hdzero - superbee f4 lite frsky v1.0

haslinghuis, 2023-01-03:
Please try configuration one step at the time. When does it fail.
The board is defined with `set rx_spi_protocol = FRSKY_X` not `FRSKY_D`

jgauruder, 2023-01-03:
If I just select spi rx (and leave at frsky_x) and do a save and reboot, I
still lose the comm port.

Regards,
Jason

haslinghuis, 2023-01-03:
Is this before or after MSP VTX configuration?

jgauruder, 2023-01-03:
This is before.
Flash (classic build, full erase)
Apply custom defaults when prompted
Reconnect after custom defaults applied
Receiver tab, change from serial to spi rx and click save and reboot... And
it is done...no usb comm port access to do anything else.  only thing I can
do is short the two pins on the fc while reconnecting physically to USB to
enter DFU mode to reflash back to pre rc 3

Same steps in rc2 works just fine.

haslinghuis, 2023-01-03:
Would have to flash nightly builds between RC2 and RC3 to establish where it fails, did not see anything touching in release notes.

jgauruder, 2023-01-03:
I can see about doing that.  I would start in the middle date between rc2
and 3 and work backwards or forwards in timeline based on duplication of
issue or not until I hopefully find the last nightly that was without
symptom.     I've not flashed a nightly before, but I see on GitHub the
assets are named like betaflight_4.4.0_STM32F411_1b98917.hex but they don't
have a file that has crazybeef4fr in the name.     Would I just flash the
STM32F411 named file and proceed with the same steps I did to try and
reproduce ?

haslinghuis, 2023-01-03:
Use auto-detect button and use `Load Firmware local `button.

jgauruder, 2023-01-03:
I did the following before I saw your last comment.  Hopefully this suffices 👍 
I used BF configurator 10.8.0 for flashing nightly builds (10.9.0 RC3 does not have nightly builds in the drop down under development)

Once flashed, closed BF configurator 10.8.0 and launched 10.9.0 RC3 and applied custom defaults and then reconnected, then receiver tab, change to SPI RX, save and reboot.

4.4.0 #2909 - good
4.4.0 #2913 - good
4.4.0 #2914 - good
4.4.0 #2915 - symptom appears

haslinghuis, 2023-01-03:
https://ci.betaflight.tech/job/Betaflight/2915/ ?

jgauruder, 2023-01-03:
[image]
yes #2915 - attached is the screenshot of the firmware flashing I did - with #2915 being the first build that produced the symptom.

haslinghuis, 2023-01-03:
https://github.com/betaflight/betaflight/pull/12111

jgauruder, 2023-01-03:
I did not do any "custom defines" that I am aware of.  Should I try to flash #2916 and try to reproduce symptom?

haslinghuis, 2023-01-03:
Yes please.

haslinghuis, 2023-01-03:
@Chillshot This ticket is about FR SPI protocol. You need to select CRSF for ELRS.

jgauruder, 2023-01-03:
4.4.0 #2916 - symptom appears just like with #2915

jgauruder, 2023-01-04:
@haslinghuis should I try a custom define?  I've just been doing the basic "classic" build

haslinghuis, 2023-01-04:
Try: `-RX_SPI` or `-RX_CC2500 (not both) with the minus in front.`

jgauruder, 2023-01-04:
[log-jan-4-2023.txt](https://github.com/betaflight/betaflight/files/10344722/log-jan-4-2023.txt)
[image]
[BTFL_cli_20230104_085650.txt](https://github.com/betaflight/betaflight/files/10344723/BTFL_cli_20230104_085650.txt)
symptom is NOT observed flashing 4.4.0 RC3 using custom define : -RX_SPI 
I did not try the -RX_CC2500 as i had success with the first option tried (-RX_SPI).
was able to flash 4.4.0 RC3 with that single custom define, then apply custom defaults after flashing.  then reconnect and receiver tab, spi rx, save and reboot...usb comm port still accessible. was able to paste in all the diff all previously saved from rc2 with two exceptions that i believe are benign for this fc as it does not have these components :

# set mag_hardware = NONE
###ERROR IN set: INVALID NAME: mag_hardware = NONE###

# set baro_hardware = NONE
###ERROR IN set: INVALID NAME: baro_hardware = NONE###
 
so in summary, for this bug/issue, the workaround was to not use classic build but rather to use the new cloud style build and add in custom define : -RX_SPI

haslinghuis, 2023-01-04:
Good news.

We need to revisit the classic build and have to check on our side and add something like 
#ifdef USE_RX_CC2500
#undef USE_RX_SPI
#endif

blckmn, 2023-01-04:
Question is what is USE_RX_SPI gating if not to add the SPI code for CC2500.

I can remove the adding of it in cloud build. It's just a mapping.

Why does it exist at all?

haslinghuis, 2023-01-04:
Afaik it is used for

- SPI configuration (`rx_spi_protocol`, `rx_spi_bus` and `rx_spi_led_inversion`)
- Bind RX button also serving other protocols.

We could try remove the mapping only if `USE_RX_CC2500` is defined as it seems to have it's own SPI implementation.

jgauruder, 2023-01-06:
Hmmm...you are holding down boot button while applying power (USB connection to your computer) ?
I know for the superbee f4 lite frsky there is no boot button , but rather two pads I have to bridge with a small metal object.    It usually takes me 5 or 6 tries to bridge it correctly while plugging in USB .

Whodeany, 2023-01-06:
I mustn't have been pressing the boot button correctly before, its in an awkward spot, got it into DFU for a reflash now.

haslinghuis, 2023-01-06:
What happens if you use `-RX_SPI` in custom defines when flashing?

Whodeany, 2023-01-07:
Used -RX_SPI in user defines and it reverted to -RX_SERIAL after flashing

kvillnv, 2023-01-11:
Same problem here on Betaflight 4.4 RC3 and RC4.

Board: CRAZYBEEF4FR

I select FrSky and SPI, then press save. 
After that the FC can't connect to BF anymore and is stuck unbound.

Only workaround was to boot in DFU and flash V4.4 RC2.

jgauruder, 2023-01-11:
@kvillnv does the workaround work for you ? Custom define value :   -RX_SPI

kvillnv, 2023-01-11:
@jgauruder Thank you. Yes it works! I can now fly and use betaflight configurator.
I installed RC4 with Custom define value : -RX_SPI , exactly as shown in your screenshot https://github.com/betaflight/betaflight/issues/12133#issuecomment-1370972063

Previously I was using the classic mode.

tim30k, 2023-01-23:
One more bricked crazybee F4 with 4.4 RC5 :( Found the thread too late
Somehow bridging the boot pads does not put the board in DFU, still can't access it. Will try to solder the pads...

jgauruder, 2023-01-24:
@tim30k keep trying...I know for me the tiny flathead screwdriver I use takes me 3 to 5 times to get it right.   Remember you have to bridge the pads while then plugging in USB power in betaflight configurator...and for me i never get it into DFU the first attempt.  

I guess soldering would (should) work...but then you have to unsolder it after when done.

tim30k, 2023-01-24:
@jgauruder and just like you said, after 6-7 attempts it finally worked! Thank you. I have already put the quad to the "might fix it some day" box, but it is flying again 😁

hartz21, 2023-01-25:
Having the same issue however the custom define does not resolve, tried RC 3-5

Shura1oplot, 2023-01-30:
The same problem with CRAZYBEEF4FR, but only if build the firmware by myself after merging commit tagged 4.4.0 with #12067. Works fine if use online build of 4.4.0.

closed 2023-03-09 by github-actions[bot]

suprchunk, 2023-03-18:
Same issue with iFlight_Blitz-f411rx