betaflight/betaflight #12134: hdzero osd

DozerFpv, 2023-01-03:
### Describe the bug

post data not viewable after disarm.
board_name JHEF405PRO
manufacturer_id JHEF
mcu_id 003d004f394e500820333942

### To Reproduce

same outcome after every flight

### Expected behavior

no post data after disarm

### Cloud build log

```txt
VERSION: 4.4.0-RC3
COMMIT:  4.4.0-RC3
BOARD:   JHEF405PRO/STM32F405
STARTED: 2023-01-03 00:12:58Z

Want to run this build locally?

git checkout 4.4.0-RC3 && make TARGET=STM32F405 EXTRA_FLAGS="-D'BUILD_KEY=4.4.0-RC3/JHEF405PRO/e48cdf22af36d2f4f6129cdfe06ab198' -D'BOARD_NAME=JHEF405PRO' -D'MANUFACTURER_ID=JHEF' -DCLOUD_BUILD -DTARGET_FLAGS -DUSE_ACC -DUSE_ACC_SPI_ICM42688P -DUSE_ACC_SPI_MPU6000 -DUSE_BARO -DUSE_BARO_BMP280 -DUSE_BARO_DPS310 -DUSE_DSHOT -DUSE_FLASH -DUSE_FLASH_W25Q128FV -DUSE_GYRO -DUSE_GYRO_SPI_ICM42688P -DUSE_GYRO_SPI_MPU6000 -DUSE_LED_STRIP -DUSE_MAX7456 -DUSE_OSD -DUSE_OSD_HD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_SERIALRX_DEFAULT -DUSE_SERIALRX_GHST -DUSE_SERIALRX_SBUS -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_TELEMETRY_GHST -DUSE_VTX"

Docker log: 

Pulling from Github (4.4.0-RC3):
From https://github.com/betaflight/betaflight
 * [new tag]             4.4.0-RC3  -> cloud_build

Checking out the branch -> cloud_build (for 4.4.0-RC3)
Switched to branch 'cloud_build'

Commencing the build (make STM32F405):
make -j ./obj/betaflight_4.4.0_STM32F405.hex
make[1]: Entering directory '/betaflight'
rm -f ./obj/main/STM32F405/.efhash_*
EF HASH -> ./obj/main/STM32F405/.efhash_d2ac473cfb3ff05350215c4e32e0d2f1
%% startup_stm32f40xx.s 
%% (optimised) ./src/main/startup/system_stm32f4xx.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu.c 
%% (optimised) ./src/main/drivers/adc_stm32f4xx.c 
%% (optimised) ./src/main/drivers/bus_i2c_stm32f4xx.c 
%% (optimised) ./src/main/drivers/bus_spi_stdperiph.c 
%% (optimised) ./src/main/drivers/dma_stm32f4xx.c 
%% (optimised) ./src/main/drivers/dshot_bitbang.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_decode.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_stdperiph.c 
%% (size optimised) ./src/main/drivers/inverter.c 
%% (size optimised) ./src/main/drivers/light_ws2811strip_stdperiph.c 
%% (size optimised) ./src/main/drivers/transponder_ir_io_stdperiph.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_shared.c 
%% (optimised) ./src/main/drivers/serial_uart_stdperiph.c 
%% (optimised) ./src/main/drivers/serial_uart_stm32f4xx.c 
%% (optimised) ./src/main/drivers/system_stm32f4xx.c 
%% (optimised) ./src/main/drivers/timer_stm32f4xx.c 
%% (optimised) ./src/main/drivers/persistent.c 
%% (optimised) ./src/main/drivers/sdio_f4xx.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_fake.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu3050.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_mpu6050.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_mpu6500.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_spi_bmi160.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_spi_bmi270.c 
%% (optimised) ./src/main/drivers/accgyro/accgyro_spi_icm20649.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_spi_icm20689.c 
[... 661 lines cut ...]
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

JHEF405PRO  jhemcu

### Other components

hdzero whoop lite vtx

### How are the different components wired up

vtx on uuart 2
built in uart base receiver on uart 1

jgauruder, 2023-01-04:
does it look like this attached where the stats definitions are shown in the middle, but the actual stats are basically far off the screen to the right ? This is mine after upgrade to rc3
[image]

jgauruder, 2023-01-04:
changing from "hd" to "auto" on osd tab resolved this issue for me.
[image]
this line is now removed in my diff all output : 
set vcd_video_system = HD

DozerFpv, 2023-01-05:
yes

closed 2023-02-11 by github-actions[bot]