betaflight/betaflight #12136: SD card corrupted BF 4.4.0 RC3

aowi7280, 2023-01-03:
### Describe the bug

It may be a coincidence, but my first day flying 4.4.0 RC3 and when I went to download the BB files, my computer would not read the SD card. I was getting correct BB file numbers on the post flight OSD screen for some of them, but I didn't look every flight, about 8 flights. Windows 10 says I need to format the card, Lexar x633 16GB. I couldn't format or even use disk part to remove the write protection that it said was there. I used the Classic Mode to flash BF. RC2 was working fine.

### To Reproduce

Flash 4.4.0 RC3. Fly several packs. Remove SD card from FC and put into Windows PC.

### Expected behavior

SD saves BB files without card corruption.

### Diff configuration

```txt
# Betaflight / STM32F405 (S405) 4.4.0 Jan  1 2023 / 15:30:38 (2c34b3949) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id AIRB

# name: Androne

# resources
resource MOTOR 5 NONE
resource SONAR_TRIGGER 1 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 11 A01
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11

# feature
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature LED_STRIP

# serial
serial 0 64 115200 57600 0 115200
serial 5 2 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# led
led 0 6,2::A:6
[... 342 lines cut ...]
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: SPI_MISO 3
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: ADC 2
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

board_name OMNIBUSF4SD manufacturer_id AIRB

### Other components

R9mm receiver flashed with latest ELRS.

haslinghuis, 2023-01-03:
We can't tell if it's a coincidence.

Could you please try RC2 again to see if it's a software issue or an hardware issue or try another SDC.

haslinghuis, 2023-01-07:
Should also work on RC3 then

aowi7280, 2023-01-09:
Flew 7 packs today on RC3 with a new SD card. It seemed fine.
BTW, I found these gps rescue PIDs work much better than defaults on my heavy 6", 4S. Throttle defaults are fine. Maybe the 50 degrees per second pitch angle needs to be reduced. 
gps_rescue_velocity_p = 5
gps_rescue_velocity_i = 7
gps_rescue_velocity_d = 7
gps_rescue_pitch_angle_max = 30

haslinghuis, 2023-01-09:
@ctzsnooze please take note about GPS Rescue PIDs above.

@aowi7280 Thanks. Issue is resolved by replacing SDC.

closed 2023-01-09 by haslinghuis