betaflight/betaflight #12137: Translate Betaflight firmware now that we have cloud build

McGiverGim, 2023-01-03:
### Is your feature request related to a problem? Please describe

I want that the CMS menus, OSD messsages and others elements appear in my own language.

### Describe the solution you'd like

Now that we have cloud build, we can have the literal strings as defines or something similar in it's own file, use Crowdin to translate them and let the cloud build generate the firmware file in the language of the user if he wants.

### Describe alternatives you've considered

All the languages inside the firmware, but this will take a lot of space.

### Other information

Nothing.

McGiverGim, 2023-01-04:
If someone wants to work on this, I've seen this tutorial that can help: https://levelup.gitconnected.com/how-to-make-a-multi-language-localized-application-in-c-8862e1d1f64f

I've played a little with it, but I don't know how to put all the pieces in place in the Betaflight code.

McGiverGim, 2023-01-04:
I liked the gettext solution, but it seems to do the magic at runtime, and we need to do it at compile time, to not waste FC space with translations. Someone has a better idea?

Quick-Flash, 2023-01-24:
> I liked the gettext solution, but it seems to do the magic at runtime, and we need to do it at compile time, to not waste FC space with translations. Someone has a better idea?

Do something sort of like how the settings and blackbox now use a unified name. Have a folder that just contains all the names for each language, then put each language folder under a language define. Use that unified language name wherever it appears. Then you could even make the cli settings a per language thing. You could even set it up so that English is default and the only language codebase, then have a separate repo with all the languages, so that even in the middle of a release you can have new languages added and improvements to the translations.

McGiverGim, 2023-01-25:
Yes, that was the only solution that I have found, but it a little more complicated than this:
1. As you say, the easiest way is to have several files, one for each language, with defines for each string that must be translated.
2. It must be done in some way that if one string is missing, then the English one must be used.
3. I liked `gettext` because the "string" remains in the code, making it more friendly. is better `printf("Profile selected")` than `printf(TRANSLATION_1)`. I was thinking maybe that simply giving a long name to the translation define can help, something like this: `printf(T_PROFILE_SELECTED)` or, if is commonly accepted in C code the undercase defines `printf(T_Profile_selected)`.
4. Not only that, I suppose that if the translation is longer than the original string, and maybe it must be included at some fixed size variable, weirds things can happen if the translation is bigger. I was thinking about adding some kind of macro to the define, so the translation files will end with something like: `#define T_PROFILE_SELECTED T("Profile selected", 20)` being the number the max size that can be used for the string. If bigget, the `T` macro must cut it to the max size.
5. I liked the `gettext` solution too because the `po` files generated are very common and accepted in translations portals. We use Crowdin for that, so we need to transform the "English language defines file" to some other more common format, maybe JSON, upload to Crowdin, and from time to time, download the translations, and transform them to "other languages defines file".

I think that's all. I have not expertise in C code, so I'm not too sure how to do some of the things explained here. But if nobody takes it, maybe I can try to do something for the next version of Betaflight.

McGiverGim, 2023-01-25:
Contacting with support from Crowdin, the formats that let specify size are: [XLSX or CSV](https://support.crowdin.com/csv-xlsx-configuration/), [XLIFF](https://store.crowdin.com/xliff), [Android XML](https://store.crowdin.com/android-xml). CSV can serve, but needs some manual config after uploading the file. Android format can serve too. Both of them will be easy to generate from the defines I think.

closed 2026-09-26 by blckmn