betaflight/betaflight #12138: GEPRC_F722_AIO BF 4.4 RC1/2/3/Zulu no gyro found

thebiggestgit, 2023-01-03:
### Describe the bug

When flashing with any of the 4.4 current releases RC1/2/3/Zulu (as of 2.1.22) the gyro is not detected on the board, works fine on 4.3.2 and below releases.

### To Reproduce

Flash the board GEPRC_F722_AIO with any of the Betaflight 4.4 current releases, choose to apply defaults after. Gyro is not detected and so drone can not be flown, unless reverting back to 4.3.2 (or below).  This is on the factory built GepRC Cinelog 35 6S HD/Caddx Vista drone.

### Expected behavior

Expect board to have gyro and accelerometer detected and function as on earlier 4.3.2 releases and below with the board betaflight defaults applied.

### Cloud build log

```txt
***** Commencing Build *****

VERSION: 4.4.0-RC3
COMMIT:  4.4.0-RC3
BOARD:   GEPRC_F722_AIO/STM32F7X2
STARTED: 2023-01-02 23:15:10Z

Want to run this build locally?

git checkout 4.4.0-RC3 && make TARGET=STM32F7X2 EXTRA_FLAGS="-D'BUILD_KEY=4.4.0-RC3/GEPRC_F722_AIO/ad8408193339f1d5e0237fdb060eaa83' -D'BOARD_NAME=GEPRC_F722_AIO' -D'MANUFACTURER_ID=GEPR' -DCLOUD_BUILD -DTARGET_FLAGS -DUSE_ACC -DUSE_ACC_SPI_ICM42688P -DUSE_ACCGYRO_BMI270 -DUSE_DSHOT -DUSE_FLASH -DUSE_FLASH_W25Q128FV -DUSE_GPS -DUSE_GYRO -DUSE_GYRO_SPI_ICM42688P -DUSE_LED_STRIP -DUSE_MAX7456 -DUSE_OSD -DUSE_OSD_HD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX"

Docker log: 

Pulling from Github (4.4.0-RC3):
From https://github.com/betaflight/betaflight
 * [new tag]             4.4.0-RC3  -> cloud_build

Checking out the branch -> cloud_build (for 4.4.0-RC3)
Switched to branch 'cloud_build'

Commencing the build (make STM32F7X2):
make -j ./obj/betaflight_4.4.0_STM32F7X2.hex
make[1]: Entering directory '/betaflight'
rm -f ./obj/main/STM32F7X2/.efhash_*
EF HASH -> ./obj/main/STM32F7X2/.efhash_10733d1243f1bced6ddb9c0f8b814ca3
%% startup_stm32f722xx.s 
%% (optimised) ./src/main/startup/system_stm32f7xx.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu.c 
%% (optimised) ./src/main/drivers/adc_stm32f7xx.c 
%% (optimised) ./src/main/drivers/audio_stm32f7xx.c 
%% (speed optimised) ./src/main/drivers/bus_i2c_hal.c 
%% (size optimised) ./src/main/drivers/bus_i2c_hal_init.c 
%% (size optimised) ./src/main/drivers/bus_i2c_timing.c 
%% (optimised) ./src/main/drivers/dma_stm32f7xx.c 
%% (size optimised) ./src/main/drivers/light_ws2811strip_hal.c 
%% (size optimised) ./src/main/drivers/transponder_ir_io_hal.c 
%% (speed optimised) ./src/main/drivers/bus_spi_ll.c 
%% (optimised) ./src/main/drivers/persistent.c 
%% (optimised) ./src/main/drivers/dshot_bitbang.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_decode.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_ll.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_hal.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_shared.c 
%% (optimised) ./src/main/drivers/timer_hal.c 
%% (optimised) ./src/main/drivers/timer_stm32f7xx.c 
%% (optimised) ./src/main/drivers/system_stm32f7xx.c 
%% (optimised) ./src/main/drivers/serial_uart_hal.c 
%% (optimised) ./src/main/drivers/serial_uart_stm32f7xx.c 
%% (optimised) ./src/main/drivers/sdio_f7xx.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_fake.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu3050.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_mpu6050.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_mpu6500.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_spi_bmi160.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_spi_bmi270.c 
[... 729 lines cut ...]
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

GEPRC_F722_AIO (supplied with Cinelog 35 HD 6S Crossfire by GepRC))

### Other components

UART1 - MSP - Caddx Vista DJI Video
UART2 - Serial RX - TBS Crossfire Nano
UART 4 - GPS Serial BN 820T, was same behavour without this fitted.

haslinghuis, 2023-01-03:
Changelog:

June 14th,2022: Gyro updated to BMI270/42688.
June 25th,2022:Mounting Hole Size changed from 26.5×26.5mm to 25.5×25.5mm.

https://geprc.com/product/gep-f722-35a-aio-f722-fc-35a-2-6s-8bits-bls-esc-25-5mm/

Please provide status output when flashed with 4.3.2

thebiggestgit, 2023-01-03:
Thanks, just flashed to 4.4 RC3 diff and status below - no gyro detected as before:

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan  2 2023 / 22:32:17 (1fdf2ece0) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name GEPRC_F722_AIO
manufacturer_id GEPR

profile 0

rateprofile 0

# end the command batch
batch end
*******************************************************************************
# status
MCU F722 Clock=216MHz, Vref=3.27V, Core temp=41degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CUSTOM DEFAULTS, size: 3653, max available: 16384
Devices detected: SPI:0, I2C:0
Gyros detected: locked

OSD: MAX7456 (30 x 13)
BUILD KEY: 4.4.0-zulu/GEPRC_F722_AIO/877d30b575141e67b09c2966be867110
System Uptime: 18 seconds, Current Time: 2023-01-03T19:27:49.048+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 0, System rate: 9
Voltage: 437 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: NOGYRO BOOTGRACE CLI MSP

haslinghuis, 2023-01-03:
When above PR is merged you should find your GYRO. For now you could add in custom defines:

`GYRO_SPI_MPU6000 ACC_SPI_MPU6000`

thebiggestgit, 2023-01-03:
Excellent thank you very much, and for providing a fix so quickly - it now recognises the gyro/accelerometer when supplying those parameters. Great it will work in the future too.

closed 2023-01-04 by blckmn

andrewfpl, 2023-01-19:
> Excellent thank you very much, and for providing a fix so quickly - it now recognises the gyro/accelerometer when supplying those parameters. Great it will work in the future too.

How to add in custom defines:

GYRO_SPI_MPU6000 ACC_SPI_MPU6000