betaflight/betaflight #12140: BF 4.4.0 RC3 Jeti Telemetrie broken

RoadyFPV, 2023-01-04:
### Describe the bug

The Jeti Telemetry is broken. RC Link works fine but the Telemetry does not.

I Use the Classic mode for flashing. If i use Build Configuration the Baro is missing too.

### To Reproduce

Flashing 4.3.2 and telemetry works great, with 4.4.0 RC3 not

### Expected behavior

Should work as in 4.3.2

### Diff configuration

```txt
Entering CLI Mode, type 'exit' to retu

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan  1 2023 / 11:34:00 (2c34b3949) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name MAMBAF722_I2C
manufacturer_id DIAT

# name: Chimera 4

# feature
feature -AIRMODE
feature MOTOR_STOP
feature GPS
feature ESC_SENSOR

# serial
serial 1 131073 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# beeper
beeper -ARMED
beeper -ON_USB

# aux
aux 0 0 1 1700 2100 0 0
aux 1 1 0 1700 2100 0 0
aux 2 13 3 1700 2100 0 0
aux 3 28 0 900 1300 0 0
aux 4 35 3 900 1300 0 0
aux 5 36 10 1700 2100 0 0

# rxfail
rxfail 4 s 2000
rxfail 6 s 1000
rxfail 7 s 2000

# master
set gyro_lpf1_static_hz = 375
set gyro_lpf2_static_hz = 750
set dyn_notch_count = 2
set dyn_notch_min_hz = 125
set dyn_notch_max_hz = 850
set gyro_lpf1_dyn_min_hz = 375
set gyro_lpf1_dyn_max_hz = 750
set acc_lpf_hz = 10
set acc_calibration = -9,-31,9,1
set mag_hardware = NONE
set min_check = 1020
set max_check = 1980
set rssi_channel = 7
set serialrx_provider = JETIEXBUS
set blackbox_device = NONE
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set bat_capacity = 650
set vbat_scale = 111
set ibata_scale = 138
set yaw_motors_reversed = ON
set small_angle = 180
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set pid_process_denom = 2
set simplified_gyro_filter_multiplier = 150
set osd_warn_bitmask = 16383
set osd_cap_alarm = 425
set osd_alt_alarm = 25
set osd_tim1 = 0
set osd_tim2 = 1
set osd_vbat_pos = 2049
set osd_rssi_pos = 3096
set osd_flymode_pos = 2737
set osd_ah_pos = 79
set osd_current_pos = 2058
set osd_mah_drawn_pos = 2241
set osd_craft_name_pos = 3077
set osd_gps_speed_pos = 2718
set osd_gps_sats_pos = 3090
set osd_home_dir_pos = 2110
set osd_home_dist_pos = 2077
set osd_flight_dist_pos = 2305
set osd_compass_bar_pos = 11
set osd_altitude_pos = 3477
set osd_power_pos = 2092
set osd_warnings_pos = 2682
set osd_avg_cell_voltage_pos = 2081
set osd_battery_usage_pos = 2745
set osd_disarmed_pos = 2491
set osd_esc_tmp_pos = 2401
set osd_flip_arrow_pos = 2334
set osd_core_temp_pos = 2433
set osd_stat_bitmask = 6309672
set osd_displayport_device = MSP
set vcd_video_system = HD
set gyro_1_align_yaw = 1800
set rpm_filter_harmonics = 2
set craft_name = Chimera 4

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 90
set dterm_lpf1_dyn_max_hz = 180
set dterm_lpf1_static_hz = 90
set dterm_lpf2_static_hz = 180
set yaw_lowpass_hz = 0
set p_pitch = 70
set i_pitch = 126
set d_pitch = 45
set f_pitch = 159
set p_roll = 67
set i_roll = 120
set f_roll = 153
set p_yaw = 67
set i_yaw = 120
set f_yaw = 153
set d_min_roll = 40
set d_min_pitch = 45
set thrust_linear = 10
set dyn_idle_min_rpm = 35
set simplified_master_multiplier = 150
set simplified_d_gain = 90
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 85
set simplified_dterm_filter_multiplier = 120

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 100
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 20
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 65

# end the command batch
batch end

#
```

### Resources configuration

```txt
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: SERIAL_TX 1
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: MOTOR 1
C09: MOTOR 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 3
    CH4 : DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 1

#
```

### Flight controller

MAMBAF722_I2C

RoadyFPV, 2023-01-06:
4.4.0 RC1 OK, 
4.4.0 RC2 OK, 
4.4.0 RC3 Telemetry broken

haslinghuis, 2023-01-06:
Please try adding `TELEMETRY_JETIEXBUS` in custom defines.

RoadyFPV, 2023-01-07:
@haslinghuis Telemetry is working fine now.

Baro works fine too now, did u change something with the Baro?

RoadyFPV, 2023-01-07:
There is soming wrong in 4.4.0 RC

Vario dosnt work, with 4.3.2 is fine.

haslinghuis, 2023-01-07:
Please add `VARIO` in custom defines.

haslinghuis, 2023-01-07:
Yes I added BARO driver to this target 4 days ago.

RoadyFPV, 2023-01-07:
There will be an Error while cloud building

***** Commencing Build *****

VERSION: 4.4.0-RC3
COMMIT:  4.4.0-RC3
BOARD:   MAMBAF722_I2C/STM32F7X2
STARTED: 2023-01-07 13:42:05Z

Want to run this build locally (note "origin" needs to be the Betaflight Repo, not your fork)?

git fetch origin 4.4.0-RC3 && git checkout -f FETCH_HEAD && make TARGET=STM32F7X2 EXTRA_FLAGS="-D'BUILD_KEY=4.4.0-RC3/MAMBAF722_I2C/8af13e736592eb1db55c7d1417c23f21' -D'BOARD_NAME=MAMBAF722_I2C' -D'MANUFACTURER_ID=DIAT' -DCLOUD_BUILD -DTARGET_FLAGS -DUSE_ACC -DUSE_ACC_SPI_MPU6000 -DUSE_BARO -DUSE_BARO_BMP280 -DUSE_DSHOT -DUSE_FLASH -DUSE_FLASH_W25Q128FV -DUSE_GPS -DUSE_GYRO -DUSE_GYRO_SPI_MPU6000 -DUSE_LED_STRIP -DUSE_MAX7456 -DUSE_OSD -DUSE_OSD_HD -DUSE_PINIO -DUSE_PWM -DUSE_SERIALRX -DUSE_SERIALRX_JETIEXBUS -DUSE_TELEMETRY -DUSE_TELEMETRY_JETIEXBUS -DUSE_VARIO -DUSE_VTX"

Docker log: 

Pulling from Github (4.4.0-RC3):
From https://github.com/betaflight/betaflight
 * [new tag]             4.4.0-RC3  -> CLOUD_HEAD

Checking out the branch -> CLOUD_HEAD (for 4.4.0-RC3)
Switched to branch 'CLOUD_HEAD'

Commencing the build (make STM32F7X2):
make -j ./obj/betaflight_4.4.0_STM32F7X2.hex
make[1]: Entering directory '/betaflight'
rm -f ./obj/main/STM32F7X2/.efhash_*
EF HASH -> ./obj/main/STM32F7X2/.efhash_13fa9621c41f2585364106c3fbd0d24f
%% startup_stm32f722xx.s 
%% (optimised) ./src/main/startup/system_stm32f7xx.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu.c 
%% (optimised) ./src/main/drivers/adc_stm32f7xx.c 
%% (optimised) ./src/main/drivers/audio_stm32f7xx.c 
%% (speed optimised) ./src/main/drivers/bus_i2c_hal.c 
%% (size optimised) ./src/main/drivers/bus_i2c_hal_init.c 
%% (size optimised) ./src/main/drivers/bus_i2c_timing.c 
%% (optimised) ./src/main/drivers/dma_stm32f7xx.c 
%% (size optimised) ./src/main/drivers/light_ws2811strip_hal.c 
%% (size optimised) ./src/main/drivers/transponder_ir_io_hal.c 
In file included from ./src/main/platform.h:119,
                 from ./src/main/drivers/accgyro/accgyro_mpu.c:26:
./src/main/target/common_pre.h:509: error: "USE_VARIO" redefined [-Werror]
  509 | #define USE_VARIO
      | 
<command-line>: note: this is the location of the previous definition
In file included from ./src/main/platform.h:119,
                 from ./src/main/drivers/bus_i2c_hal_init.c:25:
./src/main/target/common_pre.h:509: error: "USE_VARIO" redefined [-Werror]
  509 | #define USE_VARIO
      | 
<command-line>: note: this is the location of the previous definition
In file included from ./src/main/platform.h:119,
                 from ./src/main/startup/system_stm32f7xx.c:69:
./src/main/target/common_pre.h:509: error: "USE_VARIO" redefined [-Werror]
  509 | #define USE_VARIO
      | 
<command-line>: note: this is the location of the previous definition
In file included from ./src/main/platform.h:119,
                 from ./src/main/drivers/audio_stm32f7xx.c:25:
./src/main/target/common_pre.h:509: error: "USE_VARIO" redefined [-Werror]
  509 | #define USE_VARIO
      | 
<command-line>: note: this is the location of the previous definition
In file included from ./src/main/platform.h:119,
                 from ./src/main/drivers/bus_i2c_timing.c:23:
./src/main/target/common_pre.h:509: error: "USE_VARIO" redefined [-Werror]
  509 | #define USE_VARIO
      | 
<command-line>: note: this is the location of the previous definition
In file included from ./src/main/platform.h:119,
                 from ./src/main/drivers/dma_stm32f7xx.c:25:
[... 48 lines cut ...]
<command-line>: note: this is the location of the previous definition
cc1: all warnings being treated as errors
make[1]: *** [Makefile:411: obj/main/STM32F7X2/drivers/transponder_ir_io_hal.o] Error 1
In file included from ./src/main/platform.h:119,
                 from ./src/main/drivers/bus_spi_ll.c:25:
./src/main/target/common_pre.h:509: error: "USE_VARIO" redefined [-Werror]
  509 | #define USE_VARIO
      | 
<command-line>: note: this is the location of the previous definition
In file included from ./src/main/platform.h:119,
                 from ./src/main/drivers/persistent.c:27:
./src/main/target/common_pre.h:509: error: "USE_VARIO" redefined [-Werror]
  509 | #define USE_VARIO
      | 
<command-line>: note: this is the location of the previous definition
cc1: all warnings being treated as errors
make[1]: *** [Makefile:411: obj/main/STM32F7X2/drivers/persistent.o] Error 1
cc1: all warnings being treated as errors
make[1]: *** [Makefile:411: obj/main/STM32F7X2/drivers/bus_spi_ll.o] Error 1
make[1]: Leaving directory '/betaflight'
make: *** [Makefile:527: hex] Error 2
make failed with 2

 ***** Completed Build *****
BUILD TIME: 2023-01-07 13:42:09Z --- 3.929135

RoadyFPV, 2023-01-07:
[image]

Are the Custom define correct?

haslinghuis, 2023-01-07:
@RoadyFPV my bad, `[USE_]VARIO` is already defined and depends on `[USE_]BARO` and `[USE_]GPS` which are in your build parameters. To further invesigate:

1. Please provide info about specific VARIO device being used and how it's wired up.
2. Also please check flashing a classic build to see if there is a problem in 4.4 base.

@blckmn can you please add mapping for `USE_TELEMETRY_JETIEXBUS` when radioprotocol is `USE_SERIALRX_JETIEXBUS`

RoadyFPV, 2023-01-07:
How can i see the information about the VARIO being used? I dont have wired up the VARIO, its onboard. 
if i type get Baro in CLI it is on AUTO
With Classic build Telemetry dont work.

haslinghuis, 2023-01-07:
What happens is you `set baro_hardware = BMP280` and `save`?

RoadyFPV, 2023-01-07:
Same,
Altitude working, Vario not.

blckmn, 2023-01-07:
> @blckmn can you please add mapping for `USE_TELEMETRY_JETIEXBUS` when radioprotocol is `USE_SERIALRX_JETIEXBUS`

This is already there.

haslinghuis, 2023-01-08:
Only for targets > 512K

RoadyFPV, 2023-01-08:
I have another problem found,

The FC lead to freeze after a couple of minutes.
Did you use GCC 10 now? Seems like the same problem INAV have since using GCC10

https://github.com/iNavFlight/inav/issues/7672

RoadyFPV, 2023-01-08:
@marv-t maybe you can assist here in this case and have a look?

haslinghuis, 2023-01-08:
We went from GCC `9.3.1` to `10.3.1` on Jul 14, 2022 for BF 4.4..

You could type `#12146` in select commit field to test the build with GCC `11.3.1` - you still need to add telemetry manually?

RoadyFPV, 2023-01-08:
get an error 

***** Commencing Build *****

VERSION: 4.4.0-zulu
COMMIT:  pull/12146/head
BOARD:   MAMBAF722_I2C/STM32F7X2
STARTED: 2023-01-08 13:57:18Z

Want to run this build locally (note "origin" needs to be the Betaflight Repo, not your fork)?

git fetch origin pull/12146/head && git checkout -f FETCH_HEAD && make TARGET=STM32F7X2 EXTRA_FLAGS="-D'BUILD_KEY=4.4.0-zulu/MAMBAF722_I2C/e088f88d3a99d9a26217ea859025c93a' -D'BOARD_NAME=MAMBAF722_I2C' -D'MANUFACTURER_ID=DIAT' -DCLOUD_BUILD -DTARGET_FLAGS -DUSE_ACC -DUSE_ACC_SPI_MPU6000 -DUSE_BARO -DUSE_BARO_BMP280 -DUSE_DSHOT -DUSE_FLASH -DUSE_FLASH_W25Q128FV -DUSE_GPS -DUSE_GYRO -DUSE_GYRO_SPI_MPU6000 -DUSE_MAX7456 -DUSE_OSD -DUSE_OSD_HD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_JETIEXBUS -DUSE_TELEMETRY -DUSE_TELEMETRY_JETIEXBUS"

Docker log: 

Pulling from Github (pull/12146/head):
From https://github.com/betaflight/betaflight
 * [new ref]             refs/pull/12146/head -> CLOUD_HEAD

Checking out the branch -> CLOUD_HEAD (for pull/12146/head)
Switched to branch 'CLOUD_HEAD'

Commencing the build (make STM32F7X2):
make: arm-none-eabi-gcc: No such file or directory
make/tools.mk:270: *** **ERROR** arm-none-eabi-gcc not in the PATH. Run 'make arm_sdk_install' to install automatically in the tools folder of this repo.  Stop.
make failed with 2

 ***** Completed Build *****
BUILD TIME: 2023-01-08 13:57:20Z --- 1.8988382
[image]

haslinghuis, 2023-01-08:
[betaflight_4.4.0_STM32F7X2.zip](https://github.com/betaflight/betaflight/files/10368614/betaflight_4.4.0_STM32F7X2.zip)

Use auto-detect, then choose load firmware local and hit flash firmware. This hex is a cloud build with above parameters using GCC 11.3.1

RoadyFPV, 2023-01-08:
Telemetry is working, Telemetry Value Altitude ok, Telemetry Value Vario not working.

FC freeze after a couple minutes too

RoadyFPV, 2023-01-08:
Can u make me a file wit gcc 9, i will check if there is a freeze.

haslinghuis, 2023-01-08:
Cloud build with above parameters using GCC 9.3.1: [betaflight_4.4.0_STM32F7X2.zip](https://github.com/betaflight/betaflight/files/10368806/betaflight_4.4.0_STM32F7X2.zip)

RoadyFPV, 2023-01-08:
No FC freeze, 
Telemetry working, only the VARIO Value not.

haslinghuis, 2023-01-08:
This build uses full firmware version (made a fix here to include JETIEXBUS telemetry) using GCC 9.3.1
[betaflight_4.4.0_STM32F7X2.zip](https://github.com/betaflight/betaflight/files/10369572/betaflight_4.4.0_STM32F7X2.zip)

Please check Vario again.

blckmn, 2023-01-08:
> Only for targets > 512K

If you select cloud build - and use serialrx_jetibus, telemetry is automatically included.

haslinghuis, 2023-01-08:
@blckmn currently some RX and telemetry is hiding in the > 512K block. Moving them to non cloud build overflows flash size.

haslinghuis, 2023-01-09:
@RoadyFPV 

> 4.4.0 RC1 OK,
> 4.4.0 RC2 OK,
> 4.4.0 RC3 Telemetry broken

Did it freeze with RC1 or RC2 as you said there were no issue (except VARIO maybe)?

RoadyFPV, 2023-01-09:
With the rc1 and 2, I only checked whether telemetry data arrived at all. I'll check again tonight to see if the FC freezes.

I will test your second file too

RoadyFPV, 2023-01-09:
RC1 Classic Mode = Telemetry working, no Vario, no freeze, 12min test runtime, blue led flashing
RC2 Classic Mode = Telemetry working, no Vario, no freeze, 18min test runtime, blue led flashing
RC3 Classic Mode = Telemetry not working, ----, no freeze, 11min test runtime, blue led flashing

RC3 Cloud Build, Radio Protocol JETIEXBUS, Custom Define TELEMERTY_JETIEXBUS = Telemetry working, no Vario, freeze at around 1min (second test around 2min not a freeze but a restart of fc), blue led not flashing

RoadyFPV, 2023-01-10:
> This build uses full firmware version (made a fix here to include JETIEXBUS telemetry) using GCC 9.3.1 [betaflight_4.4.0_STM32F7X2.zip](https://github.com/betaflight/betaflight/files/10369572/betaflight_4.4.0_STM32F7X2.zip)
> 
> Please check Vario again.

Your file = Telemetry working, no Vario, no freeze, blue led flashing 14min runtime

The Vario Telemetry is working but ther are no Values for it in the Transmitter 
[image]

[image]

haslinghuis, 2023-01-10:
FYI. We now have moved `JETI` out of non-cloud build as flash size is already overflowing on some work.

There is nothing obvious for `VARIO` not working as that part of the code has not changed as we tested a full build on 4.4.

RoadyFPV, 2023-01-12:
@marv-t If you have some time, could you please take a look whats going on here?
If you need support for testing, let me know it.

RoadyFPV, 2023-01-12:
Why freezes the fc with gcc10 and with gcc9 not?

RoadyFPV, 2023-01-14:
build with gcc10 and disable telemetry option in configurator there is no freeze

RoadyFPV, 2023-01-14:
Matek H743 4.4.0 zulu no freezes with telemetry enabled

RoadyFPV, 2023-01-14:
Matek H743 Freeze around 12min50s
seems like GCC10 broke the jeti telemetry code

RoadyFPV, 2023-01-22:
@haslinghuis is it possible to choose the gcc version on cloud build

haslinghuis, 2023-01-23:
Maybe with a PR would build GCC 9. But also see Vario is not working with that build for you.

Checking the sensor output on your screen now. Is telemetry sensor data (9. Vario) the last one you receive?
As with GPS you should have additional sensors:

```
"BF D1",           "",         EX_TYPE_DES,   0              },     // device descripton
    {"Voltage",         "V",        EX_TYPE_22b,   DECIMAL_MASK(1)},
    {"Current",         "A",        EX_TYPE_22b,   DECIMAL_MASK(2)},
    {"Altitude",        "m",        EX_TYPE_22b,   DECIMAL_MASK(2)},
    {"Capacity",        "mAh",      EX_TYPE_22b,   DECIMAL_MASK(0)},
    {"Power",           "W",        EX_TYPE_22b,   DECIMAL_MASK(1)},
    {"Roll angle",      "\xB0",     EX_TYPE_22b,   DECIMAL_MASK(1)},
    {"Pitch angle",     "\xB0",     EX_TYPE_22b,   DECIMAL_MASK(1)},
    {"Heading",         "\xB0",     EX_TYPE_22b,   DECIMAL_MASK(1)},
    {"Vario",           "m/s",      EX_TYPE_22b,   DECIMAL_MASK(2)},
    {"GPS Sats",        "",         EX_TYPE_22b,   DECIMAL_MASK(0)},
    {"GPS Long",        "",         EX_TYPE_GPS,   DECIMAL_MASK(0)},
    {"GPS Lat",         "",         EX_TYPE_GPS,   DECIMAL_MASK(0)},
    {"GPS Speed",       "m/s",      EX_TYPE_22b,   DECIMAL_MASK(2)},
    {"GPS H-Distance",  "m",        EX_TYPE_22b,   DECIMAL_MASK(0)},
    {"GPS H-Direction", "\xB0",     EX_TYPE_22b,   DECIMAL_MASK(1)},
    {"BF D2",           "",         EX_TYPE_DES,   0              },     // device descripton
    {"GPS Heading",     "\xB0",     EX_TYPE_22b,   DECIMAL_MASK(1)},
    {"GPS Altitude",    "m",        EX_TYPE_22b,   DECIMAL_MASK(2)},
    {"G-Force X",       "",         EX_TYPE_22b,   DECIMAL_MASK(3)},
    {"G-Force Y",       "",         EX_TYPE_22b,   DECIMAL_MASK(3)},
    {"G-Force Z",       "",         EX_TYPE_22b,   DECIMAL_MASK(3)}
};
```
https://github.com/betaflight/betaflight/pull/6768/files

RoadyFPV, 2023-01-23:
I have tested 4.4.0 RC 5 today with a cloud build and have no freeze, this confuses me, i will run more tests tomorrow.
I will go back to 4.3.2 as well and check the vario.

The Altitude and Vario is transmitted back, but the Vario have no Values in it. Vario Sensor is present and not blinking.

RoadyFPV, 2023-01-23:
Iam recieve all 20 Values. GPS is on board. maybe is the vario bound to the GPS calculations?

haslinghuis, 2023-01-23:
Which version are we on?

http://www.hacker-motor.com/daten/anleitungen/jeti/ReleaseNotes_EN.pdf

RoadyFPV, 2023-01-24:
Iam at newest one V5.07

RoadyFPV, 2023-01-24:
Thats Not a problem of the Rex rx. If u want a Vario function with the assist reciever you have to buy this and unlock the device in the rx. This is only for rex assist.
Iam use a rex3.

The main problem is GCC10. Inav have the same freeze issue since using GCC10. But vario is working. I will check that again as well

The vario are generated in the code.
I think the problem have somting to do with the new baro driver thing above... at 4.4.0

RoadyFPV, 2023-01-24:
@marv-t if you want to have a look and need a mamba f722 i2c, a have a brand new that i can send you for testing.

RoadyFPV, 2023-01-24:
4.3.2 Vario is working perfect, all work as it shoud, i am get 19 senor values and 2 descriptions (BF D1 and BF D2) Heading is missing in case of no mag.

4.4.0 RC5 there are only 18 sensor values and 2 description. Heading **and Vario** is missing now. 
Freeze tests running right now

haslinghuis, 2023-01-24:
There were changes made in flight/position.c related to Vario. For testing - what happens if you disable GPS?

RoadyFPV, 2023-01-24:
2 runs without freeze on 4.4.0 RC5

Disabled GPS = recieving 10 sensors, Vario is missing too

RoadyFPV, 2023-01-24:
i think there is a missmatch where the telemetry is getting the vario calculation data maybe

RoadyFPV, 2023-01-24:
Right now in the moment, i am getting a freeze. damn thats crazy

haslinghuis, 2023-01-24:
With latest development build the debug mode `ALTITUDE` has been fixed. With this enabled you can read the `estimatedVario` value in sensor tab.
```
#ifdef USE_VARIO
    DEBUG_SET(DEBUG_ALTITUDE, 3, estimatedVario);
#endif
```

Other changes I see could have an affect: `TASK_ALTITUDE_RATE_HZ`  has been changed from 120 to 100
There are changes in baro sensor too.

haslinghuis, 2023-01-25:
@RoadyFPV please check this build by adding `#12233` in the select commit field.

RoadyFPV, 2023-01-25:
> @RoadyFPV please check this build by adding `#12233` in the select commit field.

Same Vario is missing

RoadyFPV, 2023-01-25:
[image]

Something to do with that?

RoadyFPV, 2023-01-29:
Where can i define the vario sensor?  
The code dont recognize that a vario sensor is on board. Thats why its missing

haslinghuis, 2023-01-29:
What is the value of `telemetry_disabled_vario` in CLI ?

RoadyFPV, 2023-01-29:
# get telemetry_disabled_vario
telemetry_disabled_vario = OFF
Allowed values: OFF, ON

haslinghuis, 2023-01-29:
Made a minor change to the PR, please flash again with `#12233` in the commit field and `VARIO` in custom defines.

RoadyFPV, 2023-01-29:
[image]

***** Commencing Build *****

VERSION: 4.5.0-zulu
COMMIT:  pull/12233/head
BOARD:   MAMBAF722_I2C/STM32F7X2
STARTED: 2023-01-29 16:14:50Z

Want to run this build locally (note "origin" needs to be the Betaflight Repo, not your fork)?

git fetch origin pull/12233/head && git checkout -f FETCH_HEAD && make TARGET=STM32F7X2 EXTRA_FLAGS="-D'BUILD_KEY=f4d73c2725dfcde22b3e958b88f03210' -D'RELEASE_NAME=4.5.0-zulu' -D'BOARD_NAME=MAMBAF722_I2C' -D'MANUFACTURER_ID=DIAT' -DCLOUD_BUILD -DUSE_ACC -DUSE_ACC_SPI_MPU6000 -DUSE_ACC_SPI_MPU6500 -DUSE_BARO -DUSE_BARO_BMP280 -DUSE_DSHOT -DUSE_FLASH -DUSE_FLASH_W25Q128FV -DUSE_GPS -DUSE_GYRO -DUSE_GYRO_SPI_MPU6000 -DUSE_GYRO_SPI_MPU6500 -DUSE_MAX7456 -DUSE_OSD -DUSE_OSD_HD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_JETIEXBUS -DUSE_TELEMETRY -DUSE_TELEMETRY_JETIEXBUS -DUSE_VARIO"

Docker log: 

Pulling from Github (pull/12233/head):
From https://github.com/betaflight/betaflight
 * [new ref]             refs/pull/12233/head -> CLOUD_HEAD

Checking out the branch -> CLOUD_HEAD (for pull/12233/head)
Switched to branch 'CLOUD_HEAD'

Commencing the build (make STM32F7X2):
make -j ./obj/betaflight_4.5.0_STM32F7X2.hex
make[1]: Entering directory '/betaflight'
rm -f ./obj/main/STM32F7X2/.efhash_*
EF HASH -> ./obj/main/STM32F7X2/.efhash_606000fe1d8fb166358acb2afb9d9c4d
%% startup_stm32f722xx.s 
%% (optimised) ./src/main/startup/system_stm32f7xx.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu.c 
%% (optimised) ./src/main/drivers/adc_stm32f7xx.c 
%% (optimised) ./src/main/drivers/audio_stm32f7xx.c 
%% (speed optimised) ./src/main/drivers/bus_i2c_hal.c 
%% (size optimised) ./src/main/drivers/bus_i2c_hal_init.c 
%% (size optimised) ./src/main/drivers/bus_i2c_timing.c 
%% (optimised) ./src/main/drivers/dma_stm32f7xx.c 
%% (size optimised) ./src/main/drivers/light_ws2811strip_hal.c 
%% (size optimised) ./src/main/drivers/transponder_ir_io_hal.c 
%% (speed optimised) ./src/main/drivers/bus_spi_ll.c 
%% (optimised) ./src/main/drivers/persistent.c 
%% (optimised) ./src/main/drivers/dshot_bitbang.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_decode.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_ll.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_hal.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_shared.c 
%% (optimised) ./src/main/drivers/timer_hal.c 
%% (optimised) ./src/main/drivers/timer_stm32f7xx.c 
%% (optimised) ./src/main/drivers/system_stm32f7xx.c 
%% (optimised) ./src/main/drivers/serial_uart_hal.c 
%% (optimised) ./src/main/drivers/serial_uart_stm32f7xx.c 
%% (optimised) ./src/main/drivers/sdio_f7xx.c 
%% arm_bitreversal2.S 
%% (optimised) ./src/main/drivers/flash.c 
%% (optimised) ./src/main/drivers/flash_m25p16.c 
%% (optimised) ./src/main/drivers/flash_w25n01g.c 
%% (optimised) ./src/main/drivers/flash_w25q128fv.c 
%% (optimised) ./src/main/drivers/flash_w25m.c 
%% (optimised) ./src/main/io/flashfs.c 
%% (optimised) ./src/main/drivers/usb_msc_common.c 
%% (optimised) ./src/main/drivers/usb_msc_f7xx.c 
%% (optimised) ./src/main/msc/usbd_storage.c 
%% (optimised) ./src/main/msc/usbd_storage_emfat.c 
%% (optimised) ./src/main/msc/emfat.c 
%% (optimised) ./src/main/msc/emfat_file.c 
%% (optimised) ./src/main/msc/usbd_storage_sdio.c 
%% (optimised) ./src/main/msc/usbd_storage_sd_spi.c 
%% (optimised) ./src/main/drivers/sdcard.c 
%% (optimised) ./src/main/drivers/sdcard_spi.c 
%% (optimised) ./src/main/drivers/sdcard_sdio_baremetal.c 
%% (optimised) ./src/main/drivers/sdcard_standard.c 
[... 189 lines cut ...]
./src/main/flight/position.c:234:46: note: each undeclared identifier is reported only once for each function it appears in
./src/main/flight/position.c:118:20: error: unused variable 'gpsAltOffset' [-Werror=unused-variable]
  118 |     static int32_t gpsAltOffset = 0;
      |                    ^~~~~~~~~~~~
./src/main/flight/position.c:117:20: error: unused variable 'baroAltOffset' [-Werror=unused-variable]
  117 |     static int32_t baroAltOffset = 0;
      |                    ^~~~~~~~~~~~~
At top level:
./src/main/flight/position.c:117:20: error: 'baroAltOffset' defined but not used [-Werror=unused-variable]
./src/main/flight/position.c:118:20: error: 'gpsAltOffset' defined but not used [-Werror=unused-variable]
  118 |     static int32_t gpsAltOffset = 0;
      |                    ^~~~~~~~~~~~
cc1: all warnings being treated as errors
%% (size optimised) ./src/main/io/serial_4way_stk500v2.c 
make[1]: *** [Makefile:418: obj/main/STM32F7X2/flight/position.o] Error 1
make[1]: *** Waiting for unfinished jobs....
%% (speed optimised) ./src/main/rx/ibus.c 
make[1]: Leaving directory '/betaflight'
make: *** [Makefile:530: hex] Error 2
make failed with 2

 ***** Completed Build *****
BUILD TIME: 2023-01-29 16:14:58Z --- 7.61

Submitted: 1/29/2023 4:14:48 PM, wait time: 9.81

haslinghuis, 2023-01-29:
Sorted above error now

RoadyFPV, 2023-01-30:
It dont recognize the whole VARIO sensor. the Sensor list is getting send at the first time the fc boot i think the first 5 sec.

if the vario sensor getting enabled after arm then it will not transmitted

[70 more replies]

closed 2023-10-21 by SteveCEvans