betaflight/betaflight #12141: meteor85 can not arm with RC3

motion0803, 2023-01-04:
### Describe the bug

meteor85 (BMI270) can not arm with RC3, always complain "calib".
With RC2, still has chance to arm as long as set gyro_calib_noise_limit = 200

### To Reproduce

flash meteor85 with RC3

### Expected behavior

not arm (calib)
betaflight shows "calibrating"

### Diff configuration

```txt
board_name BETAFPVF4SX1280
manufacturer_id BEFH
mcu_id 003c001f3535510636323431
signature 

# name: M85

# feature
feature -RX_SERIAL
feature -SOFTSERIAL
feature -TELEMETRY
feature -LED_STRIP
feature -AIRMODE

# serial
serial 1 2048 115200 57600 0 115200

# beeper
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 900 1800 0 0
aux 1 1 1 1300 2100 0 0
aux 2 13 6 1350 1650 0 0
aux 3 35 6 1875 2100 0 0

# adjrange
adjrange 0 0 2 900 2100 29 2 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 100 200 350

# vtx
vtx 0 7 0 0 1 900 1150
vtx 1 7 0 0 2 1150 1550
vtx 2 7 0 0 3 1550 1850
vtx 3 7 0 0 4 1850 2100

# rxfail
rxfail 3 s 1200

# master
set gyro_lpf2_static_hz = 0
set gyro_calib_duration = 200
set gyro_calib_noise_limit = 200
set dyn_notch_count = 0
set dyn_notch_q = 400
set acc_calibration = -149,-165,21,1
set fpv_mix_degrees = 25
set blackbox_device = NONE
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set failsafe_throttle = 1200
set failsafe_procedure = AUTO-LAND
set bat_capacity = 450
set vbat_max_cell_voltage = 451
set vbat_min_cell_voltage = 342
set vbat_warning_cell_voltage = 352
set yaw_motors_reversed = ON
set small_angle = 180
set osd_warn_bitmask = 949
set osd_cap_alarm = 400
set osd_tim1 = 1282
set osd_link_quality_pos = 6537
set osd_rssi_dbm_pos = 6544
set osd_tim_2_pos = 14743
set osd_vtx_channel_pos = 6145
set osd_crosshairs_pos = 6350
set osd_ah_pos = 6222
set osd_current_pos = 14720
set osd_mah_drawn_pos = 14712
set osd_craft_name_pos = 14349
set osd_warnings_pos = 14603
set osd_avg_cell_voltage_pos = 14688
set osd_disarmed_pos = 14443
set osd_flip_arrow_pos = 14561
set osd_stat_bitmask = 8603301
set osd_profile = 2
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 1
set vtx_freq = 5732
set expresslrs_uid = 10
set expresslrs_rate_index = 1
set expresslrs_switch_mode = HYBRID
set craft_name = M85

profile 0

# profile 0
set yaw_lowpass_hz = 0
set i_pitch = 40
set d_pitch = 34
set i_roll = 40
set d_roll = 30
set i_yaw = 40
set d_min_roll = 0
set d_min_pitch = 0
set dyn_idle_min_rpm = 40
set simplified_pids_mode = OFF
set tpa_mode = PD
set tpa_rate = 70
set tpa_breakpoint = 1600

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 33
set thr_expo = 83
set rates_type = BETAFLIGHT
set roll_rc_rate = 70
set pitch_rc_rate = 70
set yaw_rc_rate = 70
set roll_expo = 70
set pitch_expo = 70
set yaw_expo = 70
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

### Resources configuration

```txt
# resources
resource BEEPER 1 A14
resource MOTOR 1 B08
resource MOTOR 2 A00
resource MOTOR 3 B10
resource MOTOR 4 B07
resource MOTOR 5 A08
resource MOTOR 6 A10
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource LED_STRIP 1 B01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A08
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 A08
resource SERIAL_RX 12 NONE
resource INVERTER 1 NONE
resource INVERTER 2 NONE
resource INVERTER 3 NONE
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
resource INVERTER 8 NONE
resource INVERTER 9 NONE
resource INVERTER 10 NONE
resource INVERTER 11 NONE
resource INVERTER 12 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SDA 1 NONE
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource LED 1 C14
resource LED 2 NONE
resource LED 3 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource ESCSERIAL 1 NONE
resource CAMERA_CONTROL 1 NONE
resource ADC_BATT 1 A01
resource ADC_RSSI 1 NONE
resource ADC_CURR 1 B00
resource ADC_EXT 1 NONE
resource PINIO 1 NONE
resource PINIO 2 NONE
resource PINIO 3 NONE
resource PINIO 4 NONE
resource OSD_CS 1 B12
resource RX_SPI_CS 1 A15
resource RX_SPI_EXTI 1 C13
resource RX_SPI_BIND 1 B02
resource RX_SPI_LED 1 C15
resource RX_SPI_EXPRESSLRS_RESET 1 B09
resource RX_SPI_EXPRESSLRS_BUSY 1 A13
resource GYRO_EXTI 1 B06
resource GYRO_EXTI 2 NONE
resource GYRO_CS 1 A04
resource GYRO_CS 2 NONE
resource USB_DETECT 1 NONE
resource PULLUP 1 NONE
resource PULLUP 2 NONE
resource PULLUP 3 NONE
resource PULLUP 4 NONE
resource PULLDOWN 1 NONE
resource PULLDOWN 2 NONE
resource PULLDOWN 3 NONE
resource PULLDOWN 4 NONE
```

### Flight controller

stm32f411

motion0803, 2023-01-05:
Bluejay 
 C_X_30, v0.18.

haslinghuis, 2023-01-05:
1. Try with default settngs.
2. Did you apply some preset (and if so which one?)
3. You have disabled gyro filters?

Try adding:
```
set gyro_lpf2_type = PT2
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 575
save
```
Test now. After testing You could also add:
```
simplified_pids_mode = RPY
set iterm_relax = RP_INC
set p_pitch = 37
set i_pitch = 70
set d_pitch = 30
set f_pitch = 118
set p_roll = 34
set i_roll = 64
set d_roll = 27
set f_roll = 108
set p_yaw = 34
set i_yaw = 64
set f_yaw = 108
set d_min_roll = 27
set d_min_pitch = 30
set feedforward_smooth_factor = 30
set dyn_idle_min_rpm = 4
set simplified_master_multiplier = 90
set simplified_i_gain = 105
set simplified_pi_gain = 85
set simplified_dmax_gain = 0
set simplified_pitch_pi_gain = 105
set tpa_rate = 55
simplified_tuning apply
save
```

motion0803, 2023-01-05:
hi

    Did you apply some preset (and if so which one?)
NO
    You have disabled gyro filters?
no, pls see dump below.

===============================
# dump

# version
# Betaflight / STM32F411 (S411) 4.4.0 Dec 27 2022 / 11:31:32 (281bfb4bf) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name BETAFPVF4SX1280
manufacturer_id BEFH

# name: M85

# resources
resource BEEPER 1 A14
resource MOTOR 1 B08
resource MOTOR 2 A00
resource MOTOR 3 B10
resource MOTOR 4 B07
resource MOTOR 5 A08
resource MOTOR 6 A10
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource LED_STRIP 1 B01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A08
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 A08
resource SERIAL_RX 12 NONE
resource INVERTER 1 NONE
resource INVERTER 2 NONE
resource INVERTER 3 NONE
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
resource INVERTER 8 NONE
resource INVERTER 9 NONE
resource INVERTER 10 NONE
resource INVERTER 11 NONE
resource INVERTER 12 NONE
[... 949 lines cut ...]
set thr_mid = 33
set thr_expo = 83
set rates_type = BETAFLIGHT
set quickrates_rc_expo = OFF
set roll_rc_rate = 70
set pitch_rc_rate = 70
set yaw_rc_rate = 70
set roll_expo = 70
set pitch_expo = 70
set yaw_expo = 70
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# end the command batch
batch end

#

closed 2023-02-12 by github-actions[bot]