betaflight/betaflight #12142: Barometer not working on MAMBAF405US_I2C (4.4.0-RC3) #12129

rdahlin, 2023-01-04:
### Describe the bug

It seems like tit's the same issue as in #12129.

If I revert to 4.3.2 I have the barometer again with the following info from "status"
----------------
# status
MCU F40X Clock=168MHz (PLLP-HSI), Vref=3.30V, Core temp=43degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4100, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MAX7456
System Uptime: 29 seconds, Current Time: 2023-01-04T19:00:38.513+00:00
CPU:8%, cycle time: 135, GYRO rate: 7407, RX rate: 495, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: CLI MS

### To Reproduce

Fresh install of 4.4.0-RC3 on a MAMBAF405US_I2C 

### Expected behavior

Be able to use the barometer.

### Cloud build log

```txt
#
```

### Diff configuration

```txt
#
```

### Resources configuration

```txt
#
```

### Flight controller

MAMBAF405US_I2C

rdahlin, 2023-01-04:
Just tested to add BARO_BMP280 to Custom Defines field and it seem to work.

haslinghuis, 2023-01-04:
Thanks, when above PR is merged it will work without custom define.

closed 2023-01-05 by blckmn