betaflight/betaflight #12145: DarwinFPV 15A F411 board - No Gyro/Accel after flashing to 4.4 RC3

chaserporter, 2023-01-06:
### Describe the bug

I flashed the DarwinFPV 15A F411 AIO (non-elrs version) with the latest RC3 and after applying custom defaults it doesn't find the gyro or accelerometer. I'm flashing with the MatekF411 target as that is what was originally flashed, and what is recommended on their product page.

I can flash back to 4.3.2 and the gyro/accel come back and work properly.
[DarwinFPV AIO bf 4.3.2 dump.txt](https://github.com/betaflight/betaflight/files/10356413/DarwinFPV.AIO.bf.4.3.2.dump.txt)
[DarwinFPV AIO bf 4.4.0 rc3 dump.txt](https://github.com/betaflight/betaflight/files/10356414/DarwinFPV.AIO.bf.4.4.0.rc3.dump.txt)

I looked at a comparison between the "resource show all" output between 4.3.2 and 4.4.0-RC3 and that may be the problem:
selected lines from BF3.3.2:
A01: GYRO_EXTI
A04: GYRO_CS 1

selected lines from BF4.4.0-RC3:
A01: FREE
A04: PREINIT 1

I've tried the following to try and reassign but it didn't do anything:
resource gyro_exti 1 none
resource gyro_cs 1 none
resource gyro_exti 1 a01
resource gyro_cs 1 a04

Is there a different way to fix this?

Thanks,

### To Reproduce

Flash 4.4.0-RC3 to MatekF411 target
Apply custom defaults

### Expected behavior

Gyro and Accelerometer should show up.

### Diff configuration

```txt
# 

# diff

# version
# Betaflight / STM32F411 (S411) 4.4.0 Jan  5 2023 / 23:46:07 (2c34b3949) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name MATEKF411
manufacturer_id MTKS

# master
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0

# end the command batch
batch end

#
```

### Resources configuration

```txt
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: PREINIT 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: FREE
B04: FREE
B05: FREE
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

DARWINFPV 15A 1-3S F411 ULTRALIGHT/WHOOP AIO (pyrodrone)

### Other components

Happymodel EP2
HDZero Whoop VTX

### How are the different components wired up

HDZero MSP on UART 1
ELRS RX on UART 2

haslinghuis, 2023-01-06:
Please report `status` output to reveal the gyro detected in previous version.

You could try target `DARWINF411` as Darwin needs to update the website as there are no revision updates.

chaserporter, 2023-01-06:
# status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.29V, Core temp=24degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3237, max available: 16384
Devices detected: SPI:0, I2C:0
Gyros detected: locked

OSD: MAX7456 (30 x 13)
BUILD KEY: 4.4.0-RC3/MATEKF411/e796a10b0f2346b25d8e1358d4103258
System Uptime: 20 seconds, Current Time: 2023-01-06T04:19:29.975+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 0, System rate: 9
Voltage: 4 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
Arming disable flags: NOGYRO BOOTGRACE CLI MSP

# 

I will try flashing the Darwinf411 target tomorrow.

haslinghuis, 2023-01-06:
This does not show any GYRO detected?

chaserporter, 2023-01-06:
That's right, on 4.4.0-RC3 there is no gyro or accelerometer detected. When I connect to the FC, no sensors are detected/enabled at the top of the screen. When I flash back to 4.3.2 with the same target, everything works as normal.

chaserporter, 2023-01-06:
Update:

I flashed the DarwinF411 target and the gyro and accelerometer are detected and working. I will need to run through and apply my own configuration to make sure everything else is working.

Not sure if there is an issue with the MatekF411 target or if it is something weird when that target is flashed to this board.

chaserporter, 2023-01-06:
Confirmed everything seems to be working when using the DARWINF411 target on 4.4.0-RC3, even though the product comes with MATEKF411 target. Everything worked correctly when using the MATEKF411 target on 4.3.2 so there may be a bug with that target on 4.4.0RC3. For now though, I am no longer am having an issue with my board.

closed 2023-01-06 by haslinghuis