betaflight/betaflight #12147: SkyStars H7 HD FC does not correctly address camera C1 C2 selection by USER1/2 mode - no cam video to vtx

alccrl, 2023-01-06:
### Describe the bug

BF 4.3.2 
No way to see camera 1 or 2 output video through VTX, when OSD is active.
Camera output is visible only if camera directly connected to VTX.
Tested with two different NTSC/PAL cameras.
Suggestion : pinio incorrect ?
Dos anybody suffer same problem ?

Alfredo

### To Reproduce

Problem is on with Both 2s or 3s batteries used,
either vtx to VBAT ot VTX to FC vbat stabilized BATT (?) power line

### Expected behavior

When a CAM is connected to CAM1 or CAM2 , VTX output should include CAM image, it dows not (blank image with OSD data only).
Also playing with USER MODE 1 and 2 in Modes tab does not change behaviour.

### Diff configuration

```txt
# dump

# version
# Betaflight / STM32H743 (SH74) 4.3.2 Nov 28 2022 / 07:31:53 (60c9521) MSP API: 1.44
# config: manufacturer_id: SKST, board_name: SKYSTARSH7HD, version: a1066f52, date: 2022-11-15T14:13:48Z

# start the command batch
batch start

board_name SKYSTARSH7HD
manufacturer_id SKST

# name: -

# resources
resource BEEPER 1 B03
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 D12
resource MOTOR 4 D13
resource MOTOR 5 A00
resource MOTOR 6 A01
resource MOTOR 7 A02
resource MOTOR 8 A03
resource SERVO 1 E05
resource SERVO 2 E06
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 C07
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 D05
[... 1372 lines cut ...]
DMA2 Stream 7: FREE
```

### Flight controller

SkyStars H7 HD

### Other components

CAM 1 B14 on CAM1 video input (see above problem).
RX serial FlySky on Uart1, 
Gps on UART2, 
VTX control on UART5, 
BARO on board, 
external MAG on I2C
external Buzzer

### How are the different components wired up

Schema is on Skystars site, ibus tm  on UART 7
[image]

### Add any other context about the problem that you think might be relevant here

I cannot exclude some hw failure of my FC, but I do not see any anomalies on board, it seems a configuration issue.

c0b0lt, 2023-01-14:
For me both cams are working fine

alccrl, 2023-01-14:
How do you select ?
Did you test defaults ?

Alfredo

alccrl, 2023-01-14:
I still make trials.
I just added an frsky osd because i cant come out of it.

c0b0lt, 2023-01-14:
Try this aux conf

alccrl, 2023-01-14:
If I well undetstand, you use user2 mode with two different rx channels ?
I checked pinio several times vs betaflight, ardupilot and inav, but my tests wete with one channel and one user mode (mode 1, 2, or 3).
Pinio for 42 (user 3) is pd11 for bf and pd02 for ardupilot but 40 41 are pc05 pc02 for both bf and ardupilot.

c0b0lt, 2023-01-14:
Same channel. different values. Try to set
aux 4 40 0 900 2100 0 0
aux 5 41 4 900 1200 0 0
aux 6 41 4 1700 2100 0 0
and you should figure it out.

SteveCEvans, 2023-01-14:
To check the PINIO functionality you could, assign the PINIO 1 resource to the pin used by LED 1 and then assign the LED 1 to NONE. Do the same for PINIO 2/LED 2 and you can then confirm that your aux assignment is correct independent of the camera switching etc.

SteveCEvans, 2023-01-14:
Looking at https://github.com/betaflight/unified-targets/blob/master/configs/default/SKST-SKYSTARSH7HD.config the PINIO mappings look good.

alccrl, 2023-01-15:
Thanks for Your tips,

I am working on SKYSTARSH7HD cuncurrently compiling an INAV custom build and using released online BF 4.3.2.
Basically I use Ardupilot/BF resource setup while searching to build a "running" INAV version.
At present I soldered a Frsky OSD that works both in INAV and BF, since I did not succed, either Inav or BF, to use internal osd and camera C1 or C2 input.
In order to try Your suggestions for aux channels, I have to change the hw setup again, as soon as I will do it, I will let You know and eventually I will close the issue as not-existent or solved we'll see ...
At the moment I got a more serious problem about rear motors in inav, so I am busy in trying to better understand
 what is going on (https://github.com/iNavFlight/inav/issues/8642).

Alfredo

alccrl, 2023-01-17:
Using a frsky osd I noticed an unusual behaviour:
when I first attach lihv power at the system, googles show a quite unstable gray image with no info from camera (apparently) input.
After moving an hand up and down in front of camera lens, google image returns stable and data overlap camera input as usual.
I am not a video signal expert, but this seems to me to reveal a weak wave synch detection at some point in the chain.
I will test if this behaviour persists also with internal osd. if it does I will close this issue. 

Alfredo

alccrl, 2023-02-15:
I succeded in putting everything ro work, all bit bideo input, using inav 5.1 custom built target.
Simve everything xomes from nf umified conf, i expect also cam to work with inav ...
This does not happen, so I sirely miss some detail somewhere.
(e.g. I found simgle gyro and dual gyro fc versions exiat).

I will spend more time on this as soon as possible.

Alfredo

alccrl, 2023-02-22:
I investigated the hw.

B7AJ switch might be responsible for failure of video in passthrough to osd chip.
Could be a soldering problem.
PC2 pin is correctly addressed to fc from the B7AJ switch.
I detached the switch from the board and
I connected directly video from camera to the capacitor of video in pin of osd chip.

Now it works (I see data and image at the same time on goggles).

Alfredo

closed 2023-02-22 by alccrl

tzarmek, 2023-02-25:
I seem to be having the same issue. can you show a picture of what you did?

alccrl, 2023-02-26:
Hi,

I attach here a zip file with results of investigation and fix.
Please note that I am not exactly an hw expert, I am a sw (avionics) engineer and I discussed the solution with a collegue of mines before implementing it, because hw bypasses can cause shortcuts or change nominal current flows in the board.
First, with a multimeter, try to see if PC2 pin on fc changes its voltage when You act on user2 mode with Your transmitter (they should appear on configurator mode tab).
I did this test AFTER detaching the switch, it would be better to test it also BEFORE detaching the switch.
Voltage changes from about 0 to 3.3 V and viceversa, playing with transmitter mode switch for user2.

I think that You have better to have al look at the attached zip file, first.
I solder the yellow cable from direct camera video to video input OSD capacitor (not directly to OSD, the other side of capacitor) : OSD pin 22.

Feel free to ask me more questions later,
and let me know how it proceeds.

Notes:
1)Fix is used with Inav 5.1, not Betaflight. I created a custom SKYSTARSH7HD target for Inav, but I did not put it into the git flow, because I found hw issues (like this) and I suspect this fc to be not much suitable for flying.

2)I got a problem with compass (COMPASS FAILURE and continuos beep) after a few flights.
I have yet to understand this issue where it comes from, at present I don't think that it is caused by the hw mod I did, because I2C is not related to pins in play.

3)My vtx power is connected to power board, not fc, and I use 2s LIHV batteries for this setup, not (at least) 3s.
This is another not clear topic ..., but not related only to Skystars, many FCs are declared 3s min, just becase they come with stacks that have BECs that work at least with 3s and upper.
[Cam problem analysis.zip](https://github.com/betaflight/betaflight/files/10833222/Cam.problem.analysis.zip)

Alfredo