betaflight/betaflight #12148: DarwinFPVF411 Elrs SPI still dont by bind with firmware 4.4.0-rc3 (01/01/2023)

HVieiraFlyaround, 2023-01-06:
### Describe the bug

Hello,

I need some help to FC AIO DarwinFPVF411, because Elrs SPI still dont by bind with firmware 4.4.0-rc3 (01/01/2023). I try with 2 radios (1 built in elrs, another external module) that already have version 3.0.1.

I try the firmware darwinF411, if I put the Betafpvf4sx1280 (old version that has in FC 4.1.3) I dont have gyro configurate.

Thanks

H. Vieira

### To Reproduce

Try binding manual, and phrase process.

### Expected behavior

That bind.

### Cloud build log

```txt
none
```

### Diff configuration

```txt
# version
# Betaflight / STM32F411 (S411) 4.4.0 Jan  6 2023 / 18:31:01 (2c34b3949) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name DARWINF411
manufacturer_id DAFP
mcu_id 004800103535511236323431
signature 

# name: Mirror HV

# resources
resource BEEPER 1 A14
resource MOTOR 1 B08
resource MOTOR 2 A00
resource MOTOR 3 B10
resource MOTOR 4 B07
resource MOTOR 5 A08
resource MOTOR 6 A10
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
[... 1628 lines cut ...]
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

Darwin ELRS2.0 F411 15A Bluejay AIO DarwinFPV-ELRS AIO- F415 Flight control FPV 4-in-4 ESC

haslinghuis, 2023-01-06:
Try target `DARWINF4SX1280HD`

bigjetty, 2023-01-19:
Hello guys any luck on this build yet?? I'm still grounded and have tried all kinds of fixes??  Let me know, thanks

bigjetty, 2023-01-28:
# # Betaflight / STM32F411 (S411) 4.4.0 Jan 17 2023 / 22:20:05 (a7cf8f0c6) MSP API: 1.45

# # config: YES

# # start the command batch

# batch start
Command batch started

# # reset configuration to default settings

# defaults nosave

# resetting to defaults

# board_name DARWINF4SX1280HD
###ERROR IN board_name: BOARD_NAME CANNOT BE CHANGED. CURRENT VALUE: 'BETAFPVF4SX1280'###

# manufacturer_id DAFP

# Set manufacturer_id.
manufacturer_id DAFP

# mcu_id 003100254247501120303634
mcu_id 003100254247501120303634

# signature
signature 

# # feature

# feature -RX_SERIAL
Disabled RX_SERIAL

# feature LED_STRIP
Enabled LED_STRIP

# # serial

# serial 30 2048 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# # led

# led 0 10,6::CB:3
led 0 10,6::CB:3

# # aux

# aux 0 0 0 1300 2100 0 0
aux 0 0 0 1300 2100 0 0

# aux 1 1 1 900 1325 0 0
aux 1 1 1 900 1325 0 0

# aux 2 2 1 1300 1700 0 0
aux 2 2 1 1300 1700 0 0

# aux 3 13 2 1700 2100 0 0
aux 3 13 2 1700 2100 0 0

# aux 4 30 1 1800 2100 0 0
aux 4 30 1 1800 2100 0 0

# aux 5 35 3 1700 2100 0 0
aux 5 35 3 1700 2100 0 0

# # vtxtable

# vtxtable bands 5
[... 3950 lines cut ...]
set quickrates_rc_expo = OFF
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# end the command batch
batch end

#

bigjetty, 2023-01-28:
https://github.com/betaflight/betaflight/issues/12148#top

Still no luck hopefully someone will see something in the diff all dump I posted

bigjetty, 2023-01-28:
here's the darwinf411 / no channels in receiver tab?

# diff all

# version
# Betaflight / STM32F411 (S411) 4.3.2 Nov 28 2022 / 07:27:10 (60c9521) MSP API: 1.44

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name DARWINF411
manufacturer_id DAFP
mcu_id 003100254247501120303634
signature 

# resources
resource BEEPER 1 A14
resource MOTOR 1 B08
resource MOTOR 2 A00
resource MOTOR 3 B10
resource MOTOR 5 A08
resource MOTOR 6 A10
resource PPM 1 NONE
resource LED_STRIP 1 B01
resource SERIAL_TX 11 A08
resource SERIAL_RX 11 A08
resource LED 1 C14
resource LED 2 NONE
resource SPI_SCK 3 B03
resource SPI_MISO 3 B04
resource SPI_MOSI 3 B05
resource ADC_BATT 1 A01
resource ADC_CURR 1 B00
resource RX_SPI_CS 1 A15
resource RX_SPI_EXTI 1 C13
resource RX_SPI_BIND 1 B02
resource RX_SPI_LED 1 C15
resource GYRO_EXTI 1 B06

# timer
timer A00 AF1
# pin A00: TIM2 CH1 (AF1)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B08 AF2
# pin B08: TIM4 CH3 (AF2)
timer A10 AF1
# pin A10: TIM1 CH3 (AF1)

# dma
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0
dma pin A00 0
# pin A00: DMA1 Stream 5 Channel 3
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin B08 0
# pin B08: DMA1 Stream 7 Channel 2
dma pin A10 0
# pin A10: DMA2 Stream 6 Channel 0

# feature
feature -RX_SERIAL
feature -RX_PARALLEL_PWM
feature LED_STRIP
[... 1291 lines cut ...]
set quickrates_rc_expo = OFF
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# end the command batch
batch end

#

haslinghuis, 2023-01-28:
The `DARWINF4SX1280` config is basically the same as `BETAFPVF4SX1280` - only difference is the `ibata_scale` and `feature LED_STRIP` and different driver defines.

```
set expresslrs_uid = 0
save
```

- Power up the board three times to trigger binding mode on the receiver.
- Press bind on the transmitter.

closed 2023-02-25 by haslinghuis