betaflight/betaflight #12149: Betafpvf411rx missing blackbox flash

invalidflaw, 2023-01-06:
### Describe the bug

Betafpv added a flash blackbox to their latest revision of the aio v2 flight controller, however the target is not detecting the chip.

### To Reproduce

Load the correct target on to a betafpv aio v2 flight controller

### Expected behavior

Black box detects the black box flash chip.

### Diff configuration

```txt
# version
# Betaflight / STM32F411 (S411) 4.3.2 Nov 28 2022 / 07:27:10 (60c9521) MSP API: 1.44

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name BETAFPVF411RX
manufacturer_id BEFH
mcu_id 0066003b3535510936323431
signature 

# feature
feature -RX_PARALLEL_PWM

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

### Resources configuration

```txt
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 2
A01: ADC_BATT
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: SERIAL_TX 11
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: RX_SPI_CC2500_LNA_EN
A14: RX_SPI_CC2500_ANT_SEL
A15: RX_SPI_CS
B00: ADC_CURR
B01: FREE
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: GYRO_EXTI
B07: MOTOR 4
B08: MOTOR 1
B09: RX_SPI_CC2500_TX_EN
B10: MOTOR 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: RX_SPI_EXTI
C14: LED 1
C15: LED
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : SERIAL_TX 11
TIM2:
    CH1 : MOTOR 2
    CH3 : MOTOR 3
TIM3: FREE
TIM4:
    CH2 : MOTOR 4
    CH3 : MOTOR 1
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

F4 1S 5A AIO Brushless Flight Controller 

### Add any other context about the problem that you think might be relevant here

The target from betafpv indicates the CS pin for the onboard flash is A08 and the flash_spi_bus is 2.

haslinghuis, 2023-01-07:
Add one of these in custom defines and check. If you find the correct one please report back so we can add the right define for other users.

    FLASH_W25Q128FV
    FLASH_M25P16
    FLASH_W25M
    FLASH_W25M512
    FLASH_W25N01G
    FLASH_W25M02G

invalidflaw, 2023-01-12:
Upon testing, ```FLASH_W25Q128FV``` is the correct define. In order to make the blackbox work, I added the following to the diff.
```
feature -SOFTSERIAL
resource SERIAL_TX 11 NONE
resource SERIAL_RX 11 NONE
resource FLASH_CS 1 A08
set flash_spi_bus = 2
```

The previous define for A08 was used for a softserial in the previous target, which is not present in the current FC. I also needed to include the ```ACCGYRO_BMI270``` in the cloud build due to the gyro change.

haslinghuis, 2023-01-12:
Thank you @invalidflaw for helping out.

closed 2023-01-12 by haslinghuis