betaflight/betaflight #12163: DEBUG_DSHOT_RPM_TELEMETRY incorrect values in Blackbox and Spektrum Telemetry

themeadery, 2023-01-09:
### Describe the bug

Upgraded from 4.3.0 RC3 to 4.4.0 RC3 and noticed the RPM telemetry on my Spektrum DX8e was wrong. Did multiple blackbox logs with the Debug option set to `DEBUG_DSHOT_RPM_TELEMETRY` and noticed the whacky values.

At first I thought maybe Blackbox Explorer needed to be updated to read the new format, then I found the 6 month history of commits to rewriting dshot and dshot telemetry and figured something might be incorrect in the actual BF code.

Then I thought my old bluejay ESC firmware may not be communicating properly with the new 4.4 code, so I flashed the latest 0.18.1 written by the new developers (and the same guy who rewrote dshot/dshot telemetry/extended dshot telemetry @damosvil ), but that didn't fix the problem either, so now I'm stuck.

It seems like eRPM and eRPM to RPM calculations, bi-directional DSHOT and all the features that depend on that to do other calculations internally (like RPM filtering, etc) are working properly, it's just mucked up when it comes time to send it to my Spektrum TX or the blackbox logs.

Correct CLI measurements:
[image]
[image]
The new extended telemetry works! Yay!

But hilariously blackbox logs on a flight show inverse readings and weird stairstep patterns
[image]
[image]
[image]

Spektrum shows MAX 1400 RPM, MIN 960. I know Spektrum has a limitation and I looked through the srxl code, but that's not the problem. It's getting a weird value before it coverts it and sends it in a telemetry frame. This worked on 4.3. I could see ~25,000 RPM.

I would cite PR's, commits and other history of where I think this problem occurred, but there are too many! I have at least 20 tabs open tracking the history of the code trying to pinpoint the issue.

Quad flies great, so it's just telemetry. 

### To Reproduce

Flash 4.4.0 RC3
Configure Bidirectional DSHOT 300
Configure Telemetry on Spektrum
Log DEBUG_DSHOT_RPM_TELEMETRY and view logs in Blackbox Explorer

### Expected behavior

Correct RPM values on Spektrum DX8e telemetry screens.
Correct RPM values in Blackbox Explorer.

### Cloud build log

```txt
Cloud build does not work on my FC, I have to use Classic Mode.
```

### Diff configuration

```txt
# diff

# version
# Betaflight / STM32F745 (S745) 4.4.0 Jan  1 2023 / 19:31:30 (2c34b3949) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name IFLIGHT_F745_AIO_V2
manufacturer_id IFRC

# name: PowerPick

# feature
feature -LED_STRIP
feature -OSD
feature AIRMODE

# serial
serial 6 64 115200 57600 0 115200

# map
map TAER1234

# aux
aux 0 0 0 900 1300 0 0
aux 1 26 1 1300 1700 0 0
aux 2 31 1 900 1300 0 0
aux 3 47 3 1700 2100 0 0

# master
set acc_calibration = -21,77,5,1
set baro_hardware = NONE
set min_check = 1010
set serialrx_provider = SRXL2
set blackbox_sample_rate = 1/2
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set align_board_yaw = 225
set bat_capacity = 850
set small_angle = 180
set debug_mode = DSHOT_RPM_TELEMETRY
set craft_name = PowerPick

profile 0

# profile 0
set throttle_boost = 0
set acro_trainer_angle_limit = 40
set f_pitch = 87
set f_roll = 84
set f_yaw = 84
set dyn_idle_min_rpm = 18
set simplified_feedforward_gain = 70
set tpa_mode = PD

rateprofile 0

# rateprofile 0
set roll_rc_rate = 6
set pitch_rc_rate = 6
set yaw_rc_rate = 3
set roll_expo = 70
set pitch_expo = 70
set yaw_expo = 55
set roll_srate = 45
set pitch_srate = 45
set yaw_srate = 23

# end the command batch
batch end
```

### Resources configuration

```txt
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: CAMERA_CONTROL
B04: MOTOR 3
B05: MOTOR 4
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: PREINIT 2
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: LED 1
C14: FREE
C15: FREE
D00: GYRO_EXTI
D01: FREE
D02: BEEPER
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: SPI_SCK 4
E03: FREE
E04: PREINIT 4
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: SERIAL_TX 7
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2 : CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH2 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

iFlight Beast F7 V2 55A AIO

### Other components

ESC is BLHeli_S flashed to Bluejay 0.18.1 (also tested with 0.16)
Spektrum SRXL2 RX SPM4650
Spektrum DX8e TX

### How are the different components wired up

RX on UART7
1000uF cap on vbat

### Add any other context about the problem that you think might be relevant here

 Board: IFRC/IFLIGHT_F745_AIO_V2(STM32F745) target for firmware

damosvil, 2023-01-09:
@themeadery Please, when possible try PR https://github.com/betaflight/betaflight/pull/12166

themeadery, 2023-01-10:
PR #12166 fixed blackbox, but Spektrum Telemetry is still broken. MIN rpm is 960, MAX rpm is 1070 on DX8e when blackbox shows ~3,000 RPM on arming idle and ~30,000 RPM on full throttle rip. Thoughts @damosvil ? To reiterate, this used to work on my Spektrum DX8e under 4.3.0. Thanks.

damosvil, 2023-01-11:
@themeadery Please, when possible try https://github.com/betaflight/betaflight/pull/12178

closed 2023-01-15 by blckmn