betaflight/betaflight #12169: BETAFPV ELRS F4 1 - 2S 12A AIO; BF 4.4 RC3

demonsgalore88, 2023-01-10:


haslinghuis, 2023-01-10:
No problems here with BETAFPVSX1280 with ICM20689 on latest version.

Please use submit support data button in CLI and provide the support ID.

demonsgalore88, 2023-01-10:
> No problems here with BETAFPVSX1280 on latest version.
> 
> Did you calibrate the accelerometer?
> 
> Please use submit support data button in CLI and provide the support ID.

> No problems here with BETAFPVSX1280 on latest version.
> 
> Did you calibrate the accelerometer?
> 
> Please use submit support data button in CLI and provide the support ID.

Calibrated accelerometer multiple times. Will add support data tomorrow fc is in my office.

demonsgalore88, 2023-01-11:
> No problems here with BETAFPVSX1280 on latest version.
> 
> Did you calibrate the accelerometer?
> 
> Please use submit support data button in CLI and provide the support ID.
videos:
BF 4.3.1 - BETAFPV 12A 1-2S AIO
https://youtube.com/shorts/wTirfOEV4Ic?feature=share

BF 4.4 - BETAFPV 12A 1-2S AIO LAG STUTTER
https://youtube.com/shorts/gHnOM4uW9ZY?feature=share

support data:
version
# Betaflight / STM32F411 (S411) 4.4.0 Jan 11 2023 / 02:41:53 (2417d50d8) MSP API: 1.45

# config: YES
# board: manufacturer_id: BEFH, board_name: BETAFPVF4SX1280

# status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.28V, Core temp=40degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3659, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270
OSD: MAX7456 (30 x 13)
BUILD KEY: 4.4.0-RC4/BETAFPVF4SX1280/5ee2a62b71cb19506dd8045159e62adb
System Uptime: 60 seconds, Current Time: 2023-01-11T06:01:23.302+00:00
CPU:19%, cycle time: 310, GYRO rate: 3225, RX rate: 15, System rate: 9
Voltage: 3 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
Arming disable flags: RXLOSS CALIB CLI MSP MOTOR_PROTO

# dump hardware

# version
# Betaflight / STM32F411 (S411) 4.4.0 Jan 11 2023 / 02:41:53 (2417d50d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name BETAFPVF4SX1280
manufacturer_id BEFH

# resources
resource BEEPER 1 A14
resource MOTOR 1 B08
resource MOTOR 2 A00
resource MOTOR 3 B10
resource MOTOR 4 B07
resource MOTOR 5 A08
resource MOTOR 6 A10
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 B01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
[... 215 lines cut ...]
set gyro_2_spibus = 0
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 1
set scheduler_relax_osd = 1
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

#

demonsgalore88, 2023-01-11:
BF 4.4 RC4 REFLASH - BETAFPV 12A 1-2S AIO Stutter
https://youtu.be/22kBJ_JVHB4

demonsgalore88, 2023-01-11:
should be uploaded in 8min:
BF 4.4 RC 4 without GPS LED VTX - BETAFPV 12A 1-2S AIO etreme stutter
https://youtu.be/Dk5a_jElMOQ

haslinghuis, 2023-01-11:
Cannot reproduce on BETAFPVSX1280 with BMI270.

Cloud build with CRSF, DSHOT and OSD_SD, PIN IO, VTX

[image]

demonsgalore88, 2023-01-11:
but for me it is working with bf 4.3.1. do you think the fc has some sort of damage?

> Cannot reproduce on BETAFPVSX1280 with BMI270.
> 
> Cloud build with CRSF, DSHOT and OSD_SD, PIN IO, VTX
> 
> [image]

haslinghuis, 2023-01-11:
Using default settings here. Somehow there is more electrical noise on your board. Also try another PC / cable.

What is the red wire on top of the board?

demonsgalore88, 2023-01-11:
> Using default settings here. Somehow there is more electrical noise on your board. Also try another PC / cable.
> 
> What is the red wire on top of the board?

Elrs antenna. Will try a different cable.

demonsgalore88, 2023-01-11:
just tried another cable now the fc gets super hot just by touching...

demonsgalore88, 2023-01-13:
just tried again after the heating issues are gone like magic was involved.... same issue new video new support data

https://youtube.com/shorts/22kBJ_JVHB4?feature=share

`version
# Betaflight / STM32F411 (S411) 4.4.0 Jan 13 2023 / 10:49:32 (2417d50d8) MSP API: 1.45

# config: YES
# board: manufacturer_id: BEFH, board_name: BETAFPVF4SX1280

# status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.27V, Core temp=29degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 2948, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270
OSD: MAX7456 (30 x 13)
BUILD KEY: 4.4.0-RC4/BETAFPVF4SX1280/d87b04ee4633b727de27fca524f0c749
System Uptime: 9 seconds, Current Time: 2023-01-13T10:59:38.587+00:00
CPU:7%, cycle time: 310, GYRO rate: 3225, RX rate: 15, System rate: 9
Voltage: 3 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
Arming disable flags: RXLOSS CALIB CLI MSP MOTOR_PROTO

# dump hardware

# version
# Betaflight / STM32F411 (S411) 4.4.0 Jan 13 2023 / 10:49:32 (2417d50d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name BETAFPVF4SX1280
manufacturer_id BEFH

# resources
resource BEEPER 1 A14
resource MOTOR 1 B08
resource MOTOR 2 A00
resource MOTOR 3 B10
resource MOTOR 4 B07
resource MOTOR 5 A08
resource MOTOR 6 A10
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A08
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
[... 196 lines cut ...]
set gyro_2_spibus = 0
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 1
set scheduler_relax_osd = 1
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

# `

demonsgalore88, 2023-01-17:
Bf4.4 rc5 still does something to the gyro. Superlaggy barley notices when I move the flight controller by hand. I hope i is just a broken Fc but i can not tell.

closed 2023-02-24 by github-actions[bot]

JulianBissekkou, 2023-05-15:
what was the issue?