betaflight/betaflight #12172: DJO O3 OSD font

Mcrubens, 2023-01-10:
### Describe the bug

When using 4.4 with DJI O3, the font of the OSD is still large compared to the screen. It also seems there is no advantage of using the HD OSD version. This I compared to a rooted vista, which works very well. 

Is DJI O3 just not supposed or is this supposed to work? Selecting MSP from ports tap works fine. 

### To Reproduce

Just applying OSD. 

### Expected behavior

smalle font and more moveable around the DJI screen (not limited to a 4:3 ratio)

### Diff configuration

```txt
# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan  7 2023 / 20:46:56 (2c34b3949) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name AIKONF7
manufacturer_id AIKO

# feature
feature -TELEMETRY
feature GPS

# serial
serial 0 2 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 4 131073 115200 57600 0 115200

# mixer
mixer Y6

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 46 5 1750 2100 0 0
aux 4 13 2 1300 1700 0 0
aux 5 31 6 900 1050 0 0
aux 6 35 2 1700 2100 0 0
aux 7 36 3 1700 2100 0 0
aux 8 52 6 900 1050 0 0

# adjrange
adjrange 0 0 4 900 2100 12 4 0 0

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = 123,24,44,1
set min_check = 1020
set max_check = 1980
set rssi_channel = 12
set serialrx_provider = CRSF
set blackbox_sample_rate = 1/2
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_output_reordering = 1,2,3,0,5,4,6,7
set failsafe_switch_mode = STAGE2
set failsafe_procedure = GPS-RESCUE
set align_board_roll = 180
set current_meter = NONE
set small_angle = 180
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_min_start_dist = 20
set gps_rescue_min_sats = 5
set gps_rescue_allow_arming_without_fix = ON
set deadband = 5
set yaw_deadband = 5
set osd_warn_bitmask = 7167
set osd_cap_alarm = 1300
set osd_vbat_pos = 417
set osd_rssi_pos = 2542
set osd_tim_1_pos = 503
set osd_tim_2_pos = 471
set osd_flymode_pos = 498
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 34
set osd_ah_pos = 200
set osd_current_pos = 495
set osd_mah_drawn_pos = 490
set osd_craft_name_pos = 487
set osd_gps_speed_pos = 2063
set osd_gps_lon_pos = 34848
set osd_gps_lat_pos = 34816
set osd_gps_sats_pos = 2074
set osd_home_dir_pos = 2102
set osd_home_dist_pos = 2070
set osd_compass_bar_pos = 266
set osd_altitude_pos = 2095
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2508
set osd_avg_cell_voltage_pos = 2546
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_displayport_device = MSP
set debug_mode = GYRO_SCALED

profile 0

# profile 0
set i_pitch = 63
set d_pitch = 39
set i_roll = 60
set d_roll = 39
set i_yaw = 60
set d_min_roll = 39
set d_min_pitch = 39
set dyn_idle_min_rpm = 33
set simplified_i_gain = 75
set simplified_d_gain = 130
set simplified_dmax_gain = 0
set simplified_pitch_d_gain = 90
set tpa_mode = PD

rateprofile 0

# rateprofile 0
set rateprofile_name = Rate_4S
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 76
set pitch_srate = 84
set yaw_srate = 60
set throttle_limit_type = SCALE

# end the command batch
batch end

#
```

### Resources configuration

```txt
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: FLASH_CS
B01: FREE
B02: PREINIT 2
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 6
B07: FREE
B08: MOTOR 5
B09: FREE
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: PREINIT 4
B13: SPI_SCK 2
B14: CAMERA_CONTROL
B15: SPI_MOSI 2
C00: ADC_BATT
C01: FREE
C02: SPI_MISO 2
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 4
C07: MOTOR 1
C08: MOTOR 2
C09: MOTOR 3
C10: FREE
C11: FREE
C12: SERIAL_TX 5
C13: LED 1
C14: PINIO 1
C15: BEEPER
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 3
    CH4 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12:
    CH1 : CAMERA_CONTROL
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 2

#
```

### Flight controller

Aikon F7

### Other components

GPS (Rx1/Tx1)
DJI O3 (Rx5/Tx5)
Tracer Rx (Rx2/Tx2)

x4code, 2023-01-10:
i noticed same behaviour yesterday with O3 and 4.4. seems an issue :(

SteveCEvans, 2023-01-10:
You need not only MSP enabled, but also select the MSP (VTX + DisplayPort) option.

x4code, 2023-01-10:
> on.

i tried that already "VTX + DisplayPort", same behaviour as mentioned.

sugaarK, 2023-01-11:
We are working with dji to get them up to speed with the hd osd setup. It will require a software update for the o3 when it’s released

fabi9163, 2023-01-11:
Hi, I've tried to get Canvas working on my new o3 with Betaflight 4.4 RC3. Set correct Uart (voltage is shown in Googles2->should be correct) and "VTX+DisplayPort" for Peripherie. In OSD-Tab I set Format to HD. But no OSD-Elements in Google.

@sugaarK is the canvas-feature itself not working with 4.4 or should it work but with incorrect fonts/sizes like told in this thread?

Maybe someone else can help or have the same issue

sugaarK, 2023-01-11:
The hd osd button in the osd tab doesn’t currently work with the dji goggles, only fpv wtf, walksnail and hdzero can use it. Dji hasn’t finish their implementation yet

fabi9163, 2023-01-11:
Tried it another time with Video Format: Auto and Font Default or Font Large->Nothing...
Anybody installed 4.4 RC3 with working Canvas?

Mcrubens, 2023-01-11:
@fabi9163: Canvas mode works. Maybe try moving around the elements? They might be out of the screen.

RCSchim, 2023-01-18:
same here:   BF4.4.0RC3 and also RC5 on 2 different copters with O3.
OSD works, but it's only in 4:3  (so it doesnt reach into the 16:9 corners or over the black borders of 4:3
ALSO:  it crops about 2/3rds of the canvas (that means only 66% of the screenwidth are visible starting top left!)
So right third is not shown....   
From above this sounds like it needs an DJI fw update.   in the meantime move elements more top left....

sugaarK, 2023-01-18:
It will 100% need a firmware update from DJi we are waiting for the rest of our testing hardware and then we can work with them to alive the issue

Giammi, 2023-02-03:
> same here: BF4.4.0RC3 and also RC5 on 2 different copters with O3. OSD works, but it's only in 4:3 (so it doesnt reach into the 16:9 corners or over the black borders of 4:3 ALSO: it crops about 2/3rds of the canvas (that means only 66% of the screenwidth are visible starting top left!) So right third is not shown.... From above this sounds like it needs an DJI fw update. in the meantime move elements more top left....
> 
> specifics: bottom 4 rows are cut and 24 columns from the right side are cut

I solved specifying PAL instead of Auto in the video mode. I haven’t tried NTSC, but PAL surely works fine and all the 4:3 rows are available again. In 4.3.2 I was able to use the full height even with auto, but since 4.4 I had to specify PAL.

RCSchim, 2023-02-07:
> I solved specifying PAL instead of Auto in the video mode. I haven’t tried NTSC, but PAL surely works fine and all the 4:3 rows are available again. In 4.3.2 I was able to use the full height even with auto, but since 4.4 I had to specify PAL.

Thanks man, that helped. At least the Disarm stats are there with values again, centered.
Would be nice to have the OSD elements in the black regions again later - but for now I'm happy :)

closed 2023-03-18 by github-actions[bot]