betaflight/betaflight #12173: BETAFPV F4 1S 5A AIO ELRS v 2.0 - on BF 3.1-3.2 MSP calibration never ends and cause gyro stuck, on BF 3.0 work as expected.

porlock, 2023-01-10:
### Describe the bug

After flashing to BF 3.1 or 3.2 MSP calibration takes very long, after some time finishes but there is no readings from GYRO.  NOGYRO error is not present and BF configurator on top bar show gyro highlighted, but there is no movement registered, what is relay interesting power cycle FC don't help, gyro remains in stuck state not sending any data till until FC is re flashed again.  On BF 3.0 everything work as expected. 

### To Reproduce

Flash BETAFPV F4 1S 5A AIO ELRS v 2.0 to BF 3.2, connect to beta flight configurator, wait few minutes till gyro stop send any data.
Power cycle FC notice that gyro is still in stuck state, re flash to  BF 3.1 observe similar behawior, re flash to BF 3.0 notice that gyro get unstuck and everything works again.

### Expected behavior

Correct gyro readings on BF 3.2

### Diff configuration

```txt
defaults from BF, no changes, fresh flashed FC
```

### Resources configuration

```txt
defaults from BF, no changes, fresh flashed FC
```

### Flight controller

BETAFPV F4 1S 5A AIO  ELRS v 2.0

### Other components

None, only AIO connected over USB

### How are the different components wired up

 only AIO connected over USB

haslinghuis, 2023-01-10:
We only support BF 4.0 and up for a few years now.

porlock, 2023-01-10:
> We only support BF 4.0 and up for a few years now.

Sorry i mean 4.3.0 not 3.0, already edited the initial post

Baggsz, 2023-02-01:
Yeah I'm having similar issues. Mine will say no gyro. So I go into cli then cluck out of cli the aio reboots then I connect and my gyro is reading correctly again.

closed 2023-03-12 by github-actions[bot]

porlock, 2023-04-07:
i tested it on another betafpv F4 AIO ELRS board and it's the same, so its BF fault.