betaflight/betaflight #12174: led_strip on cloud_build not working

TheIsotopes, 2023-01-10:
### Describe the bug

Since the led_strip was removed from targets < 512kb, it now has to be added via cloud_build.
Unfortunately there is a problem with the cloud_build that the resource for the led_strip is not set.
Everything else works without problems if I insert the resource myself.

### To Reproduce

make a cloud_build for target MATEKF722MINI with defaults Build Configuration

### Expected behavior

working led_strip

### Diff configuration

```txt
# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 10 2023 / 21:20:35 (2417d50d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name MATEKF722MINI
manufacturer_id MTKS

# name: F7-Ghst

# led
led 0 6,2::AW:0
led 1 6,3::AW:0
led 2 6,4::AW:0
led 3 6,5::AW:0
led 4 9,5::AW:0
led 5 9,4::AW:0
led 6 9,3::AW:0
led 7 9,2::AW:0

# mode_color
mode_color 6 1 3

# aux
aux 0 0 0 1900 2100 0 0
aux 1 13 1 900 1100 0 0
aux 2 36 1 1900 2100 0 0

# master
set acc_hardware = NONE
set rc_smoothing_auto_factor = 25
set rc_smoothing_auto_factor_throttle = 25
set serialrx_provider = GHST
set blackbox_sample_rate = 1/8
set motor_pwm_protocol = DSHOT600
set craft_name = F7-Ghst

profile 0

# profile 0
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 35
set feedforward_jitter_factor = 4
set feedforward_boost = 18

rateprofile 0

# rateprofile 0
set roll_rc_rate = 20
set pitch_rc_rate = 20
set yaw_rc_rate = 20
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50
set roll_srate = 99
set pitch_srate = 99
set yaw_srate = 72

# end the command batch
batch end
```

### Resources configuration

```txt
# resources
resource BEEPER 1 C13
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 A15
resource MOTOR 6 B03
resource MOTOR 7 B06
resource MOTOR 8 B07
resource PWM 1 A02
resource PWM 2 A01
resource PWM 3 A00
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 B08
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 B09
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 A14
resource LED 2 A13
resource LED 3 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_SCK 4 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MISO 4 NONE
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource SPI_MOSI 4 NONE
resource CAMERA_CONTROL 1 NONE
resource ADC_BATT 1 C02
resource ADC_RSSI 1 C00
resource ADC_CURR 1 C01
resource ADC_EXT 1 A04
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 D02
resource GYRO_EXTI 1 C04
resource GYRO_EXTI 2 C03
resource GYRO_CS 1 B02
resource GYRO_CS 2 C15
resource USB_DETECT 1 C14
```

### Flight controller

MATEKF722MINI

haslinghuis, 2023-01-10:
@TheIsotopes please add `LED_STRIP_STATUS_MODE` to custom defines and report back

blckmn, 2023-01-10:
Also `resource LED_STRIP 1 NONE` seems the LED_STRIP is not assigned a pin

haslinghuis, 2023-01-10:
This in the target: `resource LED_STRIP 1 A08`

blckmn, 2023-01-10:
> This in the target: `resource LED_STRIP 1 A08`

@TheIsotopes can you please run the above in CLI and ensure you have led strip active on pin A08

TheIsotopes, 2023-01-10:
> can you please run the above in CLI and ensure you have led strip active on pin A08. The DIFF above implies it is not set.

@blckmn Yes, its working. 

That's what I meant with `...if I insert the resource myself.` above.

blckmn, 2023-01-10:
This should now be sorted. If you load your cloud build (using the load online button) and reflash @TheIsotopes then it should load the correct defaults.

blckmn, 2023-01-10:
Confirmed working for me. @TheIsotopes if it is not for you let me know and we will re-open.

closed 2023-01-10 by blckmn

TheIsotopes, 2023-01-10:
@blckmn I can confirm, works now.