betaflight/betaflight #12182: Problem on DarwinFPV with BF 4.4

Denethor85, 2023-01-12:
### Describe the bug

With the DarwinFPV Baby Ape Pro drone, if you flash BF 4.4 RC4 with the target recommended by the parent company MATEKF411 the gyroscope no longer works, if instead you flash with the DARWINF411 target there are errors and problems in the default configuration at example the programmable LED doesn't work and I don't know if anything else

### To Reproduce

Flash BF 4.4 RC4 on Baby Ape

### Expected behavior

The Gyro working on MATEKF411 or everything works fine on DATWINF411

### Support ID

```txt
.......
```

### Diff configuration

```txt
........
```

### Resources configuration

```txt
.......
```

### Flight controller

Darwinfpv 15A 1-3S F411 Ultralight/Whoop AIO

Denethor85, 2023-01-12:
First of all I ask kindly should it be flashed with target MATEKF411 as in BF 4.3 or with the new target DarwinF411?  so I can also understand if the problem is the gyro of the MATEKF411 target or the incorrect configuration of the DARWINF411 target... consequently which dump to insert here in the issie report

haslinghuis, 2023-01-12:
We should workout how to fix it on the DARWINF411 target. RIght now I'm cooking late dinner :)

haslinghuis, 2023-01-12:
Resource for LEDSTRIP is PA08 on both, did you add LED_STRIP as build option?

Denethor85, 2023-01-12:
> Resource for LEDSTRIP is PA08 on both, did you add LED_STRIP as build option?

In the Cloud Build I had already added the LED option...If I remember correctly but I was able to make the LED management work only by replacing this line with this line: "resource LED_STRIP 1 A08".
I took this value from the original DarwinFPV dump

I don't remember if there are other errors in the configuration presets... I'm at work as soon as I can, I'll check and report it here...

haslinghuis, 2023-01-12:
There was a glitch with LED_STRIP resource. Should be fixed: https://github.com/betaflight/betaflight/issues/12174

But might need development build in stead of RC4

bigjetty, 2023-01-16:
Hey guys great job on everything.. Did you happen to get this issue resolved because I also have the darwinfpv F4 1-3s AIO FC spi Expresslrs with no success binding and then with every work around there's something that prevents binding.. I have 2 of these and same thing on both.. Tried different Firmware and then get no Gyro. Please advise...ty

haslinghuis, 2023-01-16:
Save a `diff` and try `DARWINF4SX1280HD` board.

https://betaflight.com/docs/configurator/firmware-flasher-tab

bigjetty, 2023-01-20:
I've tried it and motors wouldn't work so now
I'm trying my second flight Controller...thanks

bigjetty, 2023-01-20:
So the motors aren't responding... It looks like they are but they aren't
heres video if you wanna see.....
https://linksharing.samsungcloud.com/jZleNYSmYFOA

nightflasher, 2023-01-22:
Shouldn't this be discussed in the "unified target" repo?
Since I've looked at the target, there is the wrong gyro defined. All three boards I have, uses the ICM20689.