betaflight/betaflight #12185: 4.4.0-RC4 Importing saved config from previous RC causes errors on "defaults nosave"

NameOfTheDragon, 2023-01-13:
### Describe the bug

Importing my saved config from a previous release candidate (RC2, I think) produces the following errors:
[image]

### To Reproduce

- Save a config backup while running 4.4.0-RC2 (from the presets screen)
- Flash to firmware 4.4.0-RC4 (I used a cloud build removing LEDSTRIP, OSDHD and all but the CRSF protocol)
- Apply custom defaults
- Go to the CLI screen and import your saved config

The following errors now exist and the config cannot be saved:
[image]

(Note: the MAG option is not included in the cloud build for either RC2 or RC4).

### Expected behavior

I don't really know if this is expected behaviour but I didn't have the issue with RC2.

### Support ID

```txt
I cannot find any "Support ID" button - here is some output that may be relevant.

2023-01-13 @21:57:53 -- Serial port successfully opened with ID: 16

2023-01-13 @21:57:53 -- MultiWii API version: 1.45.0

2023-01-13 @21:57:53 -- Flight controller info, identifier: BTFL, version: 4.4.0

2023-01-13 @21:57:53 -- Running firmware released on: Jan 13 2023 21:30:43

2023-01-13 @21:57:53 -- Board: DIAT/MAMBAF722_I2C(STM32F7X2), version: 0

2023-01-13 @21:57:53 -- Unique device ID: 0x2700373756501220383558

2023-01-13 @21:57:53 -- Craft name: F5 Tornado

2023-01-13 @21:57:53 -- Arming Disabled

2023-01-13 @21:59:10 -- CLI mode detected
```

### Diff configuration

```txt
# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 13 2023 / 21:30:43 (2417d50d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name MAMBAF722_I2C
manufacturer_id DIAT

# name: F5 Tornado

# feature
feature -TELEMETRY
feature GPS
feature ESC_SENSOR

# serial
serial 2 8192 115200 57600 0 115200
serial 4 2 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map AERT1234

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 1 3 1300 2100 0 0
aux 3 1 7 1700 2100 0 0
aux 4 2 1 1300 1700 0 0
aux 5 6 3 1300 2100 0 0
aux 6 27 2 1700 2100 0 0
aux 7 46 7 1700 2100 0 0
aux 8 7 3 1800 2100 0 0
aux 9 13 2 1300 2100 0 0
aux 10 26 0 1700 2100 0 0
aux 11 30 6 1300 1700 0 0
aux 12 31 8 1700 2100 0 0
aux 13 35 6 1700 2100 0 0
aux 14 39 5 900 1100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 200 400 800 1W

# vtx
vtx 0 5 0 0 0 900 1125
vtx 1 5 0 0 1 1125 1300
vtx 2 5 0 0 2 1300 1500
vtx 3 5 0 0 3 1500 1700
vtx 4 5 0 0 4 1700 1900
vtx 5 5 0 0 5 1900 2100

# master
set gyro_lpf1_static_hz = 300
set gyro_lpf2_static_hz = 600
set gyro_lpf1_dyn_min_hz = 300
set gyro_lpf1_dyn_max_hz = 600
set acc_calibration = -56,-27,33,1
set min_check = 1010
set max_check = 1990
set rc_smoothing_auto_factor = 250
set rc_smoothing_auto_factor_throttle = 170
set rc_smoothing_setpoint_cutoff = 12
set rc_smoothing_feedforward_cutoff = 12
set rc_smoothing_throttle_cutoff = 20
set fpv_mix_degrees = 20
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_delay = 8
set failsafe_switch_mode = STAGE2
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1550
set vbat_max_cell_voltage = 435
set vbat_min_cell_voltage = 320
set ibata_scale = 63
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = NMEA
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_min_start_dist = 50
set gps_rescue_ground_speed = 2000
set gps_rescue_pitch_angle_max = 45
set gps_rescue_descent_dist = 150
set gps_rescue_throttle_hover = 1360
set gps_rescue_allow_arming_without_fix = ON
set pid_process_denom = 2
set simplified_gyro_filter_multiplier = 120
set report_cell_voltage = ON
set osd_warn_bitmask = 57343
set osd_rssi_alarm = 40
set osd_rssi_dbm_alarm = -93
set osd_cap_alarm = 1550
set osd_alt_alarm = 120
set osd_tim1 = 0
set osd_tim2 = 1
set osd_vbat_pos = 6545
set osd_link_quality_pos = 2167
set osd_link_tx_power_pos = 2135
set osd_rssi_dbm_pos = 2200
set osd_tim_1_pos = 2423
set osd_tim_2_pos = 2401
set osd_remaining_time_estimate_pos = 2449
set osd_flymode_pos = 2444
set osd_throttle_pos = 2177
set osd_vtx_channel_pos = 6196
set osd_crosshairs_pos = 2285
set osd_ah_pos = 6254
set osd_current_pos = 2454
set osd_mah_drawn_pos = 6552
set osd_motor_diag_pos = 313
set osd_craft_name_pos = 6529
set osd_gps_speed_pos = 6177
set osd_gps_lon_pos = 6529
set osd_gps_lat_pos = 6529
set osd_gps_sats_pos = 14657
set osd_home_dir_pos = 2446
set osd_home_dist_pos = 2113
set osd_flight_dist_pos = 2145
set osd_compass_bar_pos = 6186
set osd_altitude_pos = 2433
set osd_power_pos = 129
set osd_warnings_pos = 14409
set osd_avg_cell_voltage_pos = 6551
set osd_battery_usage_pos = 6544
set osd_disarmed_pos = 6251
set osd_esc_rpm_pos = 88
set osd_log_status_pos = 2234
set osd_profile_name_pos = 266
set osd_camera_frame_pos = 131
set osd_stat_bitmask = 16137852
set osd_canvas_height = 13
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 3
set vtx_freq = 5917
set gyro_1_align_yaw = 1800
set craft_name = F5 Tornado
set pilot_name = NameOfTheDragon

profile 1

# profile 1
set profile_name = STD-6S
set iterm_rotation = ON
set p_pitch = 71
set i_pitch = 95
set d_pitch = 60
set f_pitch = 148
set p_roll = 62
set i_roll = 82
set d_roll = 52
set f_roll = 129
set p_yaw = 62
set i_yaw = 82
set f_yaw = 129
set d_min_roll = 39
set d_min_pitch = 44
set feedforward_smooth_factor = 40
set feedforward_jitter_factor = 16
set simplified_master_multiplier = 120
set simplified_i_gain = 75
set simplified_d_gain = 110
set simplified_pi_gain = 115
set simplified_feedforward_gain = 90
set simplified_pitch_pi_gain = 110
set tpa_mode = PD

rateprofile 0

# rateprofile 0
set rateprofile_name = Sim
set thr_mid = 30
set thr_expo = 50
set roll_rc_rate = 2
set pitch_rc_rate = 2
set yaw_rc_rate = 2
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 72

# end the command batch
batch end

#
```

### Resources configuration

```txt
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: LED 1
B04: FREE
B05: SPI_MOSI 3
B06: SERIAL_TX 1
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: MOTOR 1
C09: MOTOR 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: FREE
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 3
    CH4 : DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 1

#
```

### Flight controller

Diatone Mamba F722 I2C

### Other components

HappyModel EP1 ELRS receiver
Diatone Mamba Ultra 1 Watt VTX

SteveCEvans, 2023-01-13:
Simply because your build doesn’t include Mag support. Given that you probably don’t have a mag then just hit save and then save a new diff.

NameOfTheDragon, 2023-01-17:
> Simply because your build doesn’t include Mag support. Given that you probably don’t have a mag then just hit save and then save a new diff.

If that's correct then we can close the issue, but I thought I had observed a change in behaviour from the previous RC, which was also a cloud build and would have excluded Mag. It's possible I am mistaken. However since this is the default for cloud builds, wouldn't it be better for the CLI to not produce errors for features that have been excluded? I don't know how feasible that is but a lot of people are going to run into this at some point.

NameOfTheDragon, 2023-01-20:
Yup, this is fixed in RC5. Case closed. Thanks guys.

closed 2023-01-20 by NameOfTheDragon